espressif/esptool #873: espefuse.py: No stub support for write commands?

someburner, 2023-04-20:
### Additional context

Due to some engineering / time constraints, our programming fixture doesn't have the ability to hard reset the ESP32 in a single programming "session" (e.g. toggling rts/cts etc has no effect). Once the chip has entered stub mode, you can read efuses, but cannot write them.

For example, using an ESP Prog jumper programmer you can emulate this behavior:

```sh
export ESP_PORT=/dev/ttyUSB0

# put chip into stub mode
./esptool.py --chip esp32 -p $ESP_PORT --after no_reset_stub flash_id

# keep in stub mode - reads OK
espefuse.py --chip esp32 --port $ESP_PORT --before no_reset summary

# keep in stub mode - writes fail
espefuse.py --port $ESP_PORT --do-not-confirm burn_efuse FLASH_CRYPT_CNT 0x7F
```

The issue appears to be due to stub loader expecting different status byte lengths:

```py
class ESP32StubLoader(ESP32ROM):
    """Access class for ESP32 stub loader, runs on top of ROM."""

    FLASH_WRITE_SIZE = 0x4000  # matches MAX_WRITE_BLOCK in stub_loader.c
    STATUS_BYTES_LENGTH = 2  # same as ESP8266, different to ESP32 ROM
    IS_STUB = True
```

but current espefuse.py appears to have no way to load those overrides. I thought perhaps stub loader might not be suited for efuses but I saw this commit: 343bca1a34239b7254fecbfce33024be64c9654d

I'm using a hacked up version of esptool and passing an `esp` object around, and found these changes actually work well:
[esptool.patch.txt](https://github.com/espressif/esptool/files/11278861/esptool.patch.txt)

So not sure if this is a bug/oversight, a needed feature, or if programming efuses with the stub is not recommended.

KonstantinKondrashov, 2023-04-24:
Hi @someburner!
The way that you have tried should work. I will check out this feature to make it workable.
Thanks for the good description.

someburner, 2023-04-24:
Also, you can test this behavior without burning any fuses, it errors out before anything is actually written.

KonstantinKondrashov, 2023-05-01:
Hi @someburner!
I do not see any issue with the following setup (I think it is the same case as yours, correct me if your case is different):
- ESP32 is connected to the USB-SERIAL with only RX, TX, and GND. This COM port is used for the connection (flash/burn).
- ESP32's EN and BOOT are not connected to the USB-SERIAL. (you said that your case does not have the ability to hard reset, e.g. toggling rts/cts etc has no effect.). I manually connect to GND to go to download mode. I do it in the following sequence: 
   - EN ->GND, (keep it)
   - BOOT -> GND, (keep it)
   - Release EN, 
   - Release BOOT.
   - Chip goes into the Download mode.
- When the chip is in the Download mode using any `--after` or `--before` does not make sense because EN and BOOT are not connected to the chip.
- Then you can use any commands from esptool and espefuse.

```
> esptool.py chip_id
esptool.py v4.6-dev
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision v0.0)
Features: WiFi, BT, Dual Core, Coding Scheme None
Crystal is 40MHz
MAC: 00:00:00:00:00:00
Uploading stub...
Running stub...
Stub running...
Warning: ESP32 has no Chip ID. Reading MAC instead.
MAC: 00:00:00:00:00:00
Hard resetting via RTS pin...  <- It does not happen because RTS is not connected, so chip is still in stub

> esptool.py chip_id
esptool.py v4.6-dev
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting...
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision v0.0)
Features: WiFi, BT, Dual Core, Coding Scheme None
Crystal is 40MHz
MAC: 00:00:00:00:00:00
Stub is already running. No upload is necessary.
Warning: ESP32 has no Chip ID. Reading MAC instead.
MAC: 00:00:00:00:00:00
Hard resetting via RTS pin...  <- It does not happen because RTS is not connected, so chip is still in stub

> espefuse.py --port /dev/ttyUSB0  --do-not-confirm burn_bit BLOCK3 246
espefuse.py v4.6-dev
Connecting...
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP32
=== Run "burn_bit" command ===
bit_number:   [255]........................................................[0]
BLOCK3    : 0x0040000000000000000000000000000000000000000000000000000000000000
BLOCK3          (                ) [3 ] regs_to_write: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00400000
Check all blocks for burn...
idx, BLOCK_NAME,          Conclusion
[03] BLOCK3               is not empty
        (written ): 0xfe80000000000000000000000000000000000000000000000000000000000000
        (to write): 0x0040000000000000000000000000000000000000000000000000000000000000
        (coding scheme = NONE)
This is an irreversible operation!
BURN BLOCK3  - OK (all write block bits are set)
Reading updated efuses...
Successful
```

KonstantinKondrashov, 2023-05-01:
@someburner 
Please help to reach your case.

According to your fix. Will this fix work for you as well?

```diff
diff --git a/espefuse/__init__.py b/espefuse/__init__.py
index 681a9ba..6a4ff50 100755
--- a/espefuse/__init__.py
+++ b/espefuse/__init__.py
@@ -178,6 +178,14 @@ def main(custom_commandline=None):
         default="default_reset",
     )
 
+    init_parser.add_argument(
+        "--after",
+        "-a",
+        help="What to do after espefuse.py is finished",
+        choices=["hard_reset", "soft_reset", "no_reset", "no_reset_stub"],
+        default="no_reset",
+    )
+
     init_parser.add_argument(
         "--debug",
         "-d",
@@ -276,6 +284,18 @@ def main(custom_commandline=None):
             if not efuses.burn_all(check_batch_mode=True):
                 raise esptool.FatalError("BURN was not done")
     finally:
+        if args.after == "hard_reset":
+            esp.hard_reset()
+        elif args.after == "soft_reset":
+            print("Soft resetting...")
+            esp.soft_reset(False)
+        elif args.after == "no_reset_stub":
+            print("Staying in flasher stub.")
+        else:  # args.after == 'no_reset'
+            print("Staying in bootloader.")
+            if esp.IS_STUB:
+                esp.soft_reset(True)  # exit stub back to ROM loader
+
         if not common_args.virt and esp._port:
             esp._port.close()
```

What is the reason do not close the port? Do you want to burn some efuses? We have a batch mode for this https://docs.espressif.com/projects/esptool/en/latest/esp32/espefuse/index.html#perform-multiple-operations-in-a-single-espefuse-run

someburner, 2023-05-01:
Did you try my steps with an ESP Prog adapter? I don't believe the issue is specifically tied to our manufacturing setup. In our case I don't think we can release the EN -> Gnd or Boot -> Gnd. In the case of using ESP Prog it seems like the commands I listed for that should work- shouldn't they?

The part about not closing serial is just due to me passing in an existing `esp` object, but if I can get the write command to work while it's in stub mode then that wouldn't be necessary.

Thanks for trying it out though. I'll have to double check on my end.

KonstantinKondrashov, 2023-05-02:
[patch_espefuse_expernal_esp_instance.txt](https://github.com/espressif/esptool/files/11372107/patch_espefuse_expernal_esp_instance.txt)
Please take a look this patch. Is it good for your case?
I did not take the `--noclose` flag into this patch because it looks excessively becuase if an esp instanse was given from above layer then that layer should be responsible for closing and connection to a chip.

closed 2023-05-02 by espressif-bot

someburner, 2023-05-02:
@KonstantinKondrashov yes that's great, thank you

I will try again soon. strange it wasn't working for me but thank you for confirming. I'll tag you in a comment if I get the error again.