espressif/esptool #876: Can't flash/upload after update

neo305, 2023-04-23:
### Operating System

Windows 10

### Esptool Version

v4.5.1

### Python Version

3.9

### Chip Description

ESP32-S2FNR2 (revision v0.0)

### How is Esptool Run

Arduino 1.8.19

### Full Esptool Command Line that Was Run

upload

### Esptool Output

```shell
this is what it looked like before updating to 2.0.8
https://drive.google.com/file/d/1TRKJGF74HBrko-hcenYZMyv8A-7_HloK/view?usp=share_link
```

### More Information

that is what it looked like before I updated to 2.0.8. now all I get it this error

Arduino: 1.8.19 (Windows 10), Board: "ESP32S3 Dev Module, Disabled, QSPI PSRAM, QIO 80MHz, 16MB (128Mb), Core 1, Core 1, Hardware CDC and JTAG, Disabled, Disabled, Disabled, UART0 / Hardware CDC, 16M Flash No OTA (2MB APP/13MB SPIFFS), 240MHz (WiFi), 921600, None, Disabled"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Laz\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Laz\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Laz\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32s3:JTAGAdapter=default,PSRAM=enabled,FlashMode=qio,FlashSize=16M,LoopCore=1,EventsCore=1,USBMode=hwcdc,CDCOnBoot=default,MSCOnBoot=default,DFUOnBoot=default,UploadMode=default,PartitionScheme=app2M_nota_spiffs13M_16MB,CPUFreq=240,UploadSpeed=921600,DebugLevel=none,EraseFlash=none -ide-version=10819 -build-path C:\Users\Laz\AppData\Local\Temp\arduino_build_712283 -warnings=none -build-cache C:\Users\Laz\AppData\Local\Temp\arduino_cache_88209 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mklittlefs.path=C:\Users\Laz\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mklittlefs-3.0.0-gnu12-dc7f933.path=C:\Users\Laz\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mkspiffs.path=C:\Users\Laz\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\Laz\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -verbose C:\Users\Laz\PS4\9.00\ESP32-S3\LilyGo\LAZ305\LAZ305 LilyGo-S3 LCD\ESP32_Server_900u\ESP32_Server_900u.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Laz\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Laz\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Laz\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32s3:JTAGAdapter=default,PSRAM=enabled,FlashMode=qio,FlashSize=16M,LoopCore=1,EventsCore=1,USBMode=hwcdc,CDCOnBoot=default,MSCOnBoot=default,DFUOnBoot=default,UploadMode=default,PartitionScheme=app2M_nota_spiffs13M_16MB,CPUFreq=240,UploadSpeed=921600,DebugLevel=none,EraseFlash=none -ide-version=10819 -build-path C:\Users\Laz\AppData\Local\Temp\arduino_build_712283 -warnings=none -build-cache C:\Users\Laz\AppData\Local\Temp\arduino_cache_88209 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mklittlefs.path=C:\Users\Laz\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mklittlefs-3.0.0-gnu12-dc7f933.path=C:\Users\Laz\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mkspiffs.path=C:\Users\Laz\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\Laz\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -verbose C:\Users\Laz\PS4\9.00\ESP32-S3\LilyGo\LAZ305\LAZ305 LilyGo-S3 LCD\ESP32_Server_900u\ESP32_Server_900u.ino

Using board 'esp32s3' from platform in folder: C:\Users\Laz\AppData\Local\Arduino15\packages\esp32\hardware\esp32

Using core 'esp32' from platform in folder: C:\Users\Laz\AppData\Local\Arduino15\packages\esp32\hardware\esp32

Detecting libraries used...

recipe.preproc.macros pattern is missing

Error compiling for board ESP32S3 Dev Module.

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2023-04-23:
Hi @neo305,
the `Error compiling for board ESP32S3 Dev Module.` is output from Arduino, not esptool.
Your issue is not related to esptool at all. If you want to open an issue, this is the place to do so - https://github.com/espressif/arduino-esp32.

closed 2023-04-23 by radimkarnis

neo305, 2023-04-23:
Thank you. Sorry for wrong place