espressif/esptool #877: Esp tool not flashing on custom board

RaphoufouLeFou, 2023-04-25:
### Operating System

Windows 11 22H2

### Esptool Version

v4.3

### Python Version

Python 3.10.2

### Chip Description

ESP32-s3 WROOM-1U with CH340C serial interface

### Device Description

custom hardware
[image]
[image]
using ESP32-S3-WROOM-1U

### Hardware Configuration

LCD RST    ->  IO9
LCD DC     ->  IO10
LCD MOSI -> IO11
LCD SCK    -> IO12
LCD MISO -> IO13
LCD CS      -> IO38

### How is Esptool Run

i tried with the arduino ide, vs code and cmd

### Full Esptool Command Line that Was Run

esptool.py -p COM5 -b 460800 --trace --before default_reset --after hard_reset --chip esp32 write_flash --flash_mode dio --flash_freq 80m --flash_size detect 0x100000 C:\Users\rapha\Downloads\esp32-doom-master\doom1-cut.wad

### Esptool Output

```shell
esptool.py v2.9-dev
Serial port COM5
Connecting........_____....._
Chip is ESP32-D0WDQ6 (revision 0)
Features: WiFi, BT, Dual Core, Coding Scheme None
WARNING: Detected crystal freq 0.00MHz is quite different to normalized freq 26MHz. Unsupported crystal in use?
Crystal is 26MHz
MAC: 00:00:00:00:00:00
Uploading stub...
Running stub...

A fatal error occurred: Invalid head of packet (0x47)

full version :

esptool.py v2.9-dev
Serial port COM5
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.102 Timed out waiting for packet header
.TRACE +0.065 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
[... 673 lines cut ...]
    75204d6564697461 74696f6e204572   | u Meditation Er
TRACE +0.001 Received full packet: 010604000000000000000000
TRACE +0.004 Read invalid data:
    0106040000000000 00000000c0477572 | .............Gur
    75204d6564697461 74696f6e204572   | u Meditation Er
TRACE +0.001 Remaining data in serial buffer:
    726f723a20436f72 6520302070616e69 | ror: Core 0 pani
    632765642028496c 6c6567616c496e73 | c'ed (IllegalIns

A fatal error occurred: Invalid head of packet (0x47)
```

### More Information

the command never work, and while running, the esp32 spam the console with : "invalid header: 0xffffffff".

i also doesn't work with the USB OTG  (exact same result, bus as COM6 insted of COM5)

### Other Steps to Reproduce

i have succeed to upload something to the esp32 using "flash_download_tool_3.9.4", but the serial output was always "invalid header: 0xffffffff"

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

RaphoufouLeFou, 2023-04-25:
ok, by bad, it work now.
i have replaced the command with : 
python -m esptool -p COM5 -b 460800 --before default_reset --after hard_reset --chip esp32-s3 write_flash --flash_mode dio --flash_freq 80m --flash_size detect 0x100000 C:\Users\rapha\Downloads\esp32-doom-master\doom1-cut.wad

RaphoufouLeFou, 2023-04-25:
but it still output "invalid header: 0xffffffff"

radimkarnis, 2023-04-25:
@RaphoufouLeFou I see you are trying to flash a doom game to the esp32.
The flashing works in your case, but the problem is that you are only flashing the game data. You also need to compile and flash the application (as described [here](https://github.com/espressif/esp32-doom)).

The `invalid header: 0xffffffff` prompt tells you there is not a valid application loaded. The bootloader is trying to start executing at the `0xffffffff` address but there is nothing - the app is missing. Try to follow the instructions in that repository (including the `make flash` part).

Anyway, this is not an esptool issue. I have to close this ticket.

closed 2023-04-25 by radimkarnis

RaphoufouLeFou, 2023-04-25:
No, it was just an example, I've also tried with some examples sketch and other codes, and none of them worked