espressif/esptool #879: Serial Exception PermissionError for CH340/CH341 Driver 3.8

mentesix, 2023-05-01:
### Operating System

Win10

### Esptool Version

v4.5.1

### Python Version

3.9.10

### Full Esptool Command Line that Was Run

esptool --before default_reset --after soft_reset --chip esp8266 --port COM5 --baud 921600 write_flash --flash_mode dout --flash_size 1MB 0x0 .pio\build\esp01_1m\firmware.bin

### Esptool Output

```shell
esptool.py v4.5.1
Serial port COM5

A serial exception error occurred: Cannot configure port, something went wrong. Original message: PermissionError(13, 'Ein an das System angeschlossenes Gerät funktioniert nicht.', None, 31)
Note: This error originates from pySerial. It is likely not a problem with esptool, but with the hardware connection or drivers.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
```

### What is the Expected Behaviour?

Open port COM5 successfully and start firmware upload.
This is from a successfull run with ch340/341 driver version 3.5 and the same command as above:
```
esptool.py v4.5.1
Serial port COM5
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: e8:9f:6d:93:3c:e5
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00042fff...
Flash params set to 0x0320
Compressed 272752 bytes to 200366...
Wrote 272752 bytes (200366 compressed) at 0x00000000 in 3.3 seconds (effective 663.8 kbit/s)...
Hash of data verified.

Leaving...
Soft resetting...
```

### More Information

This was executed in a PlatformIO project.
I had troubles with the generic esp-01_1m project and from my limited knowledge I suspect the `dout` instead of the default `qio` to be the culprit.

In the process of trouble shooting, I updated the CH340/CH341 drivers to the latest **version 3.8 from 2023-03-16**, see: http://www.wch-ic.com/products/CH341.html?
After that, the error occured.

I downloaded an older **version 3.5 from 2019-03-18** with help of archive.org: https://web.archive.org/web/20211129002033/http://www.wch-ic.com/downloads/CH341SER_EXE.html
After installtion, the problem was fixed!

I noticed that the pySerial project has no new releases since mor then 6 months ago. So maybe there is a problem...?
Also, I noticed the technical staff of the manufacturer of CH340 of this bug report.

### Other Steps to Reproduce

- Update driver for CH430/431 to version 3.8 from http://www.wch-ic.com/products/CH341.html?
- start upload procedure with a usb-serial-converter with CH340/341 chip

closed 2023-05-02 by radimkarnis

mentesix, 2023-05-05:
@radimkarnis I see, thank you for your feedback.
I hope, this report helps people out there running into the same issues.

Btw, so far no feedback from the chip's manufacturer Nanjing Qinheng Microelectronics...

roblatour, 2023-07-13:
I ran into this same problem today.
Thanks for posting the above, it was very helpful in resolving my problem.
However, there was more to it than just that for me.
I've detailed the rest at [](https://community.lilygo.cc/topic/481/can-t-upload-using-arduino-2-1-1/4)

Here is the text of that

`[@teastain2](https://www.community.lilygo.cc/uid/612),

Thanks again for your help, and for confirming Arduino 2.1.1 was working fine with the LilyGo T-Display S3 on your system.

After spending virtually the whole day on this, I was able to pin-point the problem and work out a solution.

First, let me say I have also been using the ESP-IDF for some projects.

With that in mind, somewhere along the line I updated the CH430/431 driver to its most current version (3.8).

This turned out to have been at the root of the problem. Finding this post: https://github.com/espressif/esptool/issues/879 I tried reverting the driver to version 3.5, but that alone did not resolve the problem - although it was indeed the key part of the solution.

After downgrading the driver, the sketch would still not upload on either of the two LilyGo T-Display S3 boards which I was working with via the Arduino IDE 2.1.1.

However, I found that I could upload a very basic "hello world" like sketch to the LilyGo T-Display S3 using the ESP-IDF. Also, once I did this, I could then upload the sketch that I needed to the same board using the Arduino IDE.

In other words, it appears not only did the use of the current CH430/431 driver (3.8) prevent uploads to the LilyGo T-Display S3, it also somehow appears to have corrupted the flash in the board such that even with the right driver the Arduino IDE could not upload to it.

Using the ESP-IDF to upload to the board, seems to have reset things such that the Arduino IDE would again work fine with the 3.5 version of the CH430/431 driver.

I repeated the process with my second LilyGo T-Display S3 and got the exact same positive results.

Just posting this here in case it will be of use to others.

Thanks again to [@teastain2](https://www.community.lilygo.cc/uid/612) on this forum, and xamfa, radimkarnis. and mentesix on the Espressif forum for there help and posting toward this resolution.`

elschopi, 2023-08-14:
On my Win11 machine, using the 3.5 version driver from the web archive solved my connection problems with my D1 mini. However, on my Raspberry Pi 4B with Ubuntu 22.04 AARCH64 I still have this issue. How would I downgrade the driver version on that machine?

jorgitsss, 2024-02-21:
El Sketch usa 1509889 bytes (47%) del espacio de almacenamiento de programa. El máximo es 3145728 bytes.
Las variables Globales usan 70388 bytes (21%) de la memoria dinámica, dejando 257292 bytes para las variables locales. El máximo es 327680 bytes.
esptool.py v4.5.1
Serial port COM5

A serial exception error occurred: Cannot configure port, something went wrong. Original message: PermissionError(13, 'Uno de los dispositivos conectados al sistema no funciona.', None, 31)
Note: This error originates from pySerial. It is likely not a problem with esptool, but with the hardware connection or drivers.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Failed uploading: uploading error: exit status 1