espressif/esptool #882: 'A fatal error occurred: This chip is ESP32-S3 not ESP32. Wrong --chip argument?' (When uploading SPIFFS)

conjuspace, 2023-05-09:
### Operating System

mac OSX 13.2 Ventura

### Esptool Version

v4.5.1

### Python Version

Python 3.9

### Chip Description

ESP32-S3 Dev Module (Adafruit Feather ESPS2-S3)

### Device Description

No other hardware, running to an empty board so far.

### How is Esptool Run

Arduino 1.8.19

### Esptool Output

```shell
esptool.py v4.5.1
Serial port /dev/cu.usbmodem14401
Connecting...

A fatal error occurred: This chip is ESP32-S3 not ESP32. Wrong --chip argument?
SPIFFS Upload failed!
```

### More Information

Really appreciate input!

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

dobairoland, 2023-05-10:
You need to look for selecting the right chip in Arduino. And then it will invoke esptool with `--chip esp32s3` instead of `--chip esp32` argument.

This is a project configuration issue, not an esptool one.

closed 2023-05-10 by dobairoland

conjuspace, 2023-05-10:
Thanks a lot for this. ESP32-S3 Dev Module and was selected and every
ESP32-S... available has been attempted.

It has connect connected when going back to the first version of ESP32 in
Board Manager (pot luck guess). Doesn't seem optimum solution though.

dobairoland, 2023-05-10:
I'm sorry, you need to seek help at the Arduino community. I don't use it myself.

VeniceInventors, 2024-09-25:
For anybody else stumbling upon this thread while trying to solve the same problem:
Make sure you have the latest [ESP32FS](https://github.com/me-no-dev/arduino-esp32fs-plugin/releases/) (V1.1 at this time) which adds support for newer esp32 variants like S2 and S3.
If you have ESP32FS V1.0 it won't work because it has "--chip esp32" hard-coded in it.