espressif/esptool #883: flash and erase > 16MB ( 32 MB ) fails - also with --no-stumb (ESP32-S2, ESP32-S3 )

ESP32DE, 2023-05-22:
### Operating System

Ubuntu, Windows, 

### Esptool Version

esptool.py v4.5.1

### Python Version

Python 3.10.6

### Chip Description

ESP32-S2, ESP32-S3

### Device Description

ESP32-S2-DevKitC-1 with WINBOND128MB 
ESP32-S3-DevKitC-1 with WINBOND128MB

### How is Esptool Run

idf console

### Full Esptool Command Line that Was Run

/home/dev23/.espressif/python_env/idf5.0_py3.10_env/bin/python /home/dev23/s3/esp-idf/components/esptool_py/esptool/esptool.py -p /dev/ttyUSB0 -b 460800 --before default_reset --after hard_reset --chip esp32s3 erase_flash  || also second try fails ||   /home/dev23/.espressif/python_env/idf5.0_py3.10_env/bin/python /home/dev23/s3/esp-idf/components/esptool_py/esptool/esptool.py -p /dev/ttyUSB0 -b 460800 --before default_reset --after hard_reset --chip esp32s3 --no-stub erase_flash

### Esptool Output

```shell
esptool.py v4.5.1
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32-S3 (revision v0.0)
Features: WiFi, BLE
Crystal is 40MHz
MAC: (blacked out here)
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Erasing flash (this may take a while)...

A fatal error occurred: Packet content transfer stopped (received 8 bytes)

# a try with --no-stumb

esptool.py v4.5.1
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32-S3 (revision v0.0)
Features: WiFi, BLE
Crystal is 40MHz
MAC: ( blacked out here 
Changing baud rate to 460800
Changed.
Enabling default SPI flash mode...
Erasing flash (this may take a while)...

A fatal error occurred: ESP32-S3 ROM does not support function erase_flash.
```

### More Information

never success with the esptool

i use external spi flash tool for > 16MB ( 32MB ) 
for flash & erase 

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

ESP32DE, 2023-05-22:
[image]

radimkarnis, 2023-05-22:
Hi @ESP32DE,

the stub currently doesn't support quad flash chips larger than 16MB (only octal) - this feature is being worked on.
The ROM never supported the `erase_flash` command, but it should support writing to >16MB quad flash chips.

So in the meantime, you can write a large binary with all `FF` to the flash to essentially "erase" it.

ESP32DE, 2023-05-22:
> Hi @ESP32DE,
> 
> the stub currently doesn't support quad flash chips larger than 16MB (only octal) - this feature is being worked on. 

ok,  what is the upper limit for octal?

> So in the meantime, you can write a large binary with all `FF` to the flash to essentially "erase" it.

this i do here exactly,
txs

closed 2023-09-20 by espressif-bot