espressif/esptool #885: Cannot communicate with ESP32-S

thisandre, 2023-05-29:
### Operating System

Ubuntu 22.04

### Esptool Version

V4.6-dev

### Python Version

3.8.16

### Chip Description

ESP32-S

### Device Description

ESP8266 ESP32-WROVER development Board Test Programmer Sockel Downloader for ESP-01 ESP01S ESP12 ESP32
(https://de.aliexpress.com/item/1005004876091486.html?spm=a2g0o.order_list.order_list_main.21.24bd5c5f1YSAzM&gatewayAdapt=glo2deu)

### Hardware Configuration

nothing connected - all ports free except EN, GPIO0 and Txd and Rxd, connected to USB comms 

### How is Esptool Run

from ubuntu terminal 

### Full Esptool Command Line that Was Run

esptool.py --port /dev/ttyUSB0 --baud 9600 --trace erase_flash

### Esptool Output

```shell
esptool.py v4.6-dev
Serial port /dev/ttyUSB0
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.100 No serial data received.
.TRACE +0.050 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.100 No serial data received.
.TRACE +0.050 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.101 No serial data received.
.TRACE +0.050 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
[... 278 lines cut ...]
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.101 No serial data received.
.

A fatal error occurred: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
```

### More Information

checked with oscilloscope and commands from esptool arrive on RxD of ESP32-S.  No transmit  on TxD of ESP32-S
Put 10microF capacitor between ground and EN of ESP32-S.  Signals arrive during esptool activity on EN and GPIO0. 
Same happens with other ESP32-S chips (4 in total)
Programming an ESP12-F and ESP8266-01 on the same board works without problems. 

### Other Steps to Reproduce

happens every time.

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

dobairoland, 2023-05-29:
Have you tried to connect with a serial program? (https://docs.espressif.com/projects/esptool/en/latest/esp32/troubleshooting.html#early-stage-crash) You should see some boot log.

Or putting the chip into download mode and see if you can read anything with a serial terminal program? (https://docs.espressif.com/projects/esptool/en/latest/esp32/troubleshooting.html#no-serial-data-received)

dobairoland, 2023-05-29:
>  No transmit on TxD of ESP32-S

I guess this answers my question. But in that case we (or esptool) cannot do anything about this. Serial communication must work in order to flash.

thisandre, 2023-05-29:
I tried everything you described but the ESP does not return anything on the TxD line when receiving data from esptool.py.  Could it have anything to do with timings and that EN or GPIO0 is triggered too short for this chip? As I said, I have 4 chips that don't work, at the other hand communicating with ESP12-f and ESP8266-01 on the same development board works without problems.

radimkarnis, 2023-05-29:
> Could it have anything to do with timings and that EN or GPIO0 is triggered too short for this chip?

No, this would only affect the chip automatically resetting into the download mode. There should be data coming out of the chip even when not in download mode. You should at least see the [boot log](https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html#boot-log) being printed on chip reset.

I think the "Downloader module" from your AliExpress link is not compatible with the modules you're using. This is 100% a hardware issue - either there is not a proper Tx/Rx connection or the chip is being held in reset (`EN` pin is pulled to gnd).

dobairoland, 2023-05-29:
Here is a HW design review site of Espressif: https://www.espressif.com/en/contact-us/circuit-schematic-pcb-design-review

I'm sorry but we cannot do anything about this on the esptool level.

closed 2023-05-29 by dobairoland

thisandre, 2023-05-29:
I indeed believe that there is a compatibility issue somewhere.   The hardware connection is OK (Txd, Rxd, GPIO0, EN). I contacted the vendor and we will see what happens.  Thanks for your time.  You can close this issue.