espressif/esptool #886: A fatal error occurred: Timed out waiting for packet header

bfgelectronics, 2023-06-07:
### Operating System

Ubuntu 23

### Esptool Version

__version__ = "3.0"

### Python Version

Python 3.10.8

### Chip Description

Custom esp12f board

### Device Description

Schematic image https://imgur.com/a/vUAVW5R

### Hardware Configuration

Can be seen on the schematic, to mention Pins 12,13,14 Are inputs pulled high

### How is Esptool Run

Platformio

### Esptool Output

```shell
Auto-detected: /dev/ttyUSB0
Uploading .pio/build/esp12e/firmware.bin
HeHe
esptool.py v3.0
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 3c:e9:0e:e2:44:98
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 264720 bytes to 195040...

A fatal error occurred: Timed out waiting for packet header
```

### More Information

Tried to increase the delays for the flash and it didnt worked. Also removed the pin from GPIO 9 and put the EN pin directly to 3.3V

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

bfgelectronics, 2023-06-07:
@dobairoland Unfortunatley thats the latest version installed by platformio also i am not sure i got the version right (just found the variable __version__ in esptool.py )

dobairoland, 2023-06-07:
You can use the binary generated by platformio with a separately installed esptool. Your one is a very old version. There has been a lot of development and fixes over the last three years. Even if we could reproduce the issue, we would be fixing the above mentioned versions (which, as I mentioned, might already contain the fix).

bfgelectronics, 2023-06-07:
@dobairoland new output after upgrade:
```
Auto-detected: /dev/ttyUSB0
Uploading .pio/build/esp12e/firmware.bin
esptool.py v4.5.1
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 3c:e9:0e:e2:44:98
Uploading stub...
Running stub...
Stub running...

A fatal error occurred: Unable to verify flash chip connection (No serial data received.).
```

radimkarnis, 2023-06-08:
@bfgelectronics it seems like the stub flasher crashes when any command involving the SPI flash chip runs.

Can you please try again with the `--no-stub` option?

bfgelectronics, 2023-06-08:
@radimkarnis  Added the flags : 
```
upload_flags = 
  --no-stub 
  --trace
```
And i got this result:
```
Auto-detected: /dev/ttyUSB0
Uploading .pio/build/esp12e/firmware.bin
esptool.py v4.5.1
Serial port /dev/ttyUSB0
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.100 No serial data received.
.TRACE +0.050 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.100 No serial data received.
.TRACE +0.050 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.100 No serial data received.
.TRACE +0.050 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.100 No serial data received.
.TRACE +0.050 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.100 No serial data received.
.TRACE +0.708 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.100 No serial data received.
.TRACE +0.050 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
[... 237 lines cut ...]
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.100 No serial data received.
.TRACE +0.050 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.100 No serial data received.
.TRACE +0.050 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.100 No serial data received.
.

A fatal error occurred: Failed to connect to ESP8266: No serial data received.
```
Also i removed the led from the gpio 9 and tried to pull it down ant it seems its the same

radimkarnis, 2023-06-12:
@bfgelectronics I think there is something wrong with your flash chip or the HW connections. 
In both versions of esptool you've tried it fails the moment the SPI flash chip is used for the first time.

Do you have other hardware to test with? If the issue persists even with other similar boards, we should investigate more. If not, this is definitely a faulty unit.

radimkarnis, 2023-06-13:
There are similar reports - always a result of a faulty HW or wrong pins being used: https://github.com/espressif/esptool/issues/394#issuecomment-1585782153

radimkarnis, 2023-06-26:
I don't think there's more I can do to help you. Closing for inactivity. Feel free to reopen if you believe we should continue looking into this.

closed 2023-06-26 by radimkarnis