espressif/esptool #891: esptool bridged over next MCU Failed to write to target RAM (result was 01070000

mmar22, 2023-06-21:
### Operating System

W10 64

### Esptool Version

v4.4

### Python Version

3.8.7

### Chip Description

ESP32

### Device Description

custom with Wrover module

### Hardware Configuration

USB -> MCU1 -> RX/TX -> ESP

### How is Esptool Run

powershell

### Full Esptool Command Line that Was Run

esptool.py -p com5 -b 921600 --after no_reset --trace read_mac

### Esptool Output

```shell
esptool.py v4.4
Serial port com5
Connecting...
Failed to get PID of a device on com5, using standard reset sequence.
TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.000 Read 3 bytes: 657473
TRACE +0.000 Read invalid data: 657473
TRACE +0.000 Remaining data in serial buffer: 20
.TRACE +0.061 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.107 No serial data received.
.TRACE +0.064 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.004 Read 1 bytes: c0
TRACE +0.000 Read 1 bytes: 01
TRACE +0.000 Read 1 bytes: 08
[... 817 lines cut ...]
TRACE +0.000 Read 1 bytes: 04
TRACE +0.000 Read 1 bytes: 00
TRACE +0.000 Read 1 bytes: 3c
TRACE +0.000 Read 1 bytes: 59
TRACE +0.000 Read 1 bytes: ac
TRACE +0.000 Read 1 bytes: 0c
TRACE +0.000 Read 1 bytes: 01
TRACE +0.000 Read 1 bytes: 07
TRACE +0.000 Read 1 bytes: 00
TRACE +0.000 Read 1 bytes: 00
TRACE +0.000 Read 1 bytes: c0
TRACE +0.000 Received full packet: 010704003c59ac0c01070000

A fatal error occurred: Failed to write to target RAM (result was 01070000: Operation timed out)
```

### More Information

same result on other speeds
trace seems be good
is timeouts setup in esptool OK?

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2023-06-21:
Hi @mmar22, 
can you please explain your setup more? 

In the chain `USB -> MCU1 -> RX/TX -> ESP`, is `MCU1` a MCU configured to act as a USB-to-serial bridge or is it a devkit where a separate USB-to-serial chip is present? If the latter, is the MCU on the devkit disabled? It could be sending some unexpected data over the TX/RX lines.

mmar22, 2023-06-21:
MCU configured to act as a USB-to-serial bridge

radimkarnis, 2023-06-21:
Please try:
1) Updating to the latest esptool (v4.6.2)
2) Setting longer timeouts with a [configuration file](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/configuration-file.html)

If the USB-to-serial implementation running on the secondary MCU is not fast enough, it might be bottlenecking the comms.

mmar22, 2023-06-21:
Please i add trace show me where is timeout and i accept trying , now isnt clean.

radimkarnis, 2023-06-21:
You can create an esptool.cfg file in your current directory with this content:

```
[esptool]
timeout = 10
max_timeout = 240
erase_write_timeout_per_mb = 40
mem_end_rom_timeout = 1
serial_write_timeout = 10
````

If that doesn't help, you can try increasing the numbers.

mmar22, 2023-06-21:
I write nothing changes , because i test as you write with timeouts config file and too with bigger timeouts. Trace log result no change say as is writed.
And sorry in trace is time info then show me operation time with timeout.

mmar22, 2023-06-22:
I download and test now latest exe version with same result
[lastver.log](https://github.com/espressif/esptool/files/11828982/lastver.log)
My tip, USB-to-serial isnt slow, but quick result to timeout calculated 0 and this result to error in esptools.?

radimkarnis, 2023-06-22:
@mmar22 Error `Failed to write to target RAM (result was 01070000: Operation timed out` vcelku jasně říká, že při zápisu do RAMky (když se nahrává flasher stub) timeoutuje operace. Pokud nepomůže zvýšení timeoutů na straně esptoolu, jedná se tedy o timeout v ROMce, který nelze konfigurovat. V tom případě za to pravděpodobně může nedostatečná implementace USB-to-UART bridge na jiném MCU.

Můžeš zkusit použít esptool bez flasher stubu (`esptool.py --no-stub ...`) a vyhnout se tak zápisu do RAMky.

> A ten trace tiez by mohol ukazovat co je problem...

Trace ukazuje data putující po seriové lince, nedokáže odhalit všechny problémy, jako např. vadný hardware, crash v ROMce, flashku chráněnou proti zápisu, ...

mmar22, 2023-06-22:
Ok budeme analyzovat nas MCU ale. 
Maly feature request na tento trace: podme od 
`MAC: 90:38:0c:ac:59:3c
Uploading stub...
TRACE +0.002 command op=0x05 data len=16 wait_response=1 timeout=3.000 data=680d0000010000000018000000e00b40
TRACE +0.000 Write 26 bytes: 
    c000051000000000 00680d0000010000 | .........h......
    000018000000e00b 40c0             | ........@.
`
jasne oznacene co sa deje avsak dalsi write vyzera inak
`TRACE +0.000 Received full packet: 010504003c59ac0c00000000
TRACE +0.002 command op=0x07 data len=3448 wait_response=1 timeout=3.000 data=
    680d000000000000 0000000000000000 | h...............
    0800f43f1c00f43f 0000f43f00c0fc3f | ...?...?...?...?
...
`
potom
`    0322a0db1df00000                  | ."......
TRACE +0.002 Write 3548 bytes: 
    c00007780de70000 00680d0000000000 | ...x.....h......
`
a na zaver
`TRACE +0.000 Received full packet: 010704003c59ac0c01070000

A fatal error occurred: Failed to write to target RAM (result was 01070000: Operation timed out)`

ziadny checksum zidne info o prenose a problem urcite nieje timeout. Pravdepodobne su poskodene data alebo ina vec.

radimkarnis, 2023-06-26:
`Command op=0x05` je začátek nahrávání [flasher stubu](https://github.com/espressif/esptool/blob/master/flasher_stub/README.md) do RAM paměti, `command op=0x07` už jsou ta samotná data stubu. To lze vypnout pomocí `--no-stub`, tím se tento upload přeskočí. Při přenosu toho velkého bloku dat ale espčko vyhodí timeout error (`01070000: Operation timed out`) - tento error nepochází z host počítače ale z ROM bootloaderu v espčku. 

Vidím, že při přenosu bloku dat je timeout nastavený na 3s (`timeout=3.000` v logu). Pokud je tedy tento přenos pomalejší než 3s, failne. Timeout se dá prodloužit pomocí configuration file. Pomalejší přenos je způsobený právě třeba i nedostatečnou implementací USB-to-UART bridge na druhém MCU.

> problem urcite nieje timeout. Pravdepodobne su poskodene data alebo ina vec.

Error `Operation timed out` tvrdí opak. 

Tohle není problém esptoolu. Je to program na sériovou komunikaci. Plně tedy závisí na funkčním seriovém spojení.
Zavírám tedy tohle issue.

closed 2023-06-26 by radimkarnis

evgen-sh, 2023-07-04:
Hi! I faced with similar problem on stub uploading (`01070000: Operation timed out`). My problem was in MCU, part of bytes was dropped due to "high" speed serial connection. Incoming data buffer in MCU solved this problem.

skot, 2023-07-13:
FWIW I was getting this same error on a Mac connected to an ESP32 with a CH9102 usbserial chip. It turns out the driver wasn't right. Installing the WCH driver following [these](https://learn.adafruit.com/how-to-install-drivers-for-wch-usb-to-serial-chips-ch9102f-ch9102/mac-driver-installation) instructions fixed the issue.