espressif/esptool #892: idf_monitor doesn't work well for rfc2217

Andrysky, 2023-06-24:
### Operating System

Windows 11

### Esptool Version

v3.3.2

### Python Version

3.8.7

### Full Esptool Command Line that Was Run

 C:\___\software\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:\___\software\esp\esp-idf\tools\idf_monitor.py -p rfc2217://192.168.2.31:2217?ign_set_control -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 c:\git\__\build\___.elf

### Esptool Output

```shell
--- idf_monitor on rfc2217://192.168.2.31:2217?ign_set_control 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
I (526) cpu_start: Compile time:     Jun 10 2023 15:35:21
I (532) cpu_start: ELF file SHA256:  22427f8f630b6c71...
I (538) cpu_start: ESP-IDF:          v4.4.4
I (543) heap_init: Initializing. RAM available for dynamic allocation:
I (550) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (556) heap_init: At 3FFB7C60 len 000283A0 (160 KiB): DRAM
I (562) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (568) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (575) heap_init: At 400963B0 len 00009C50 (39 KiB): IRAM
I (582) spi_flash: detected chip: gd
I (585) spi_flash: flash io: dio
�ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
```

### What is the Expected Behaviour?

port serial output continues outputting data

radimkarnis, 2023-06-26:
Hello @Andrysky,
do I understand correctly that you are using [esp_rfc2217_server.py](https://github.com/espressif/esptool/blob/master/esp_rfc2217_server.py)? 

By default, IDF Monitor will reset the target when connecting to it. It seems like in combination with the rfc2217 implementation it resets the board again into the download mode. To prevent IDF Monitor from automatically resetting the target on connection, please try calling IDF Monitor with the `--no-reset` option (e.g., `idf_monitor.py --no-reset`).

dobairoland, 2023-07-14:
I'm closing this for inactivity.

closed 2023-07-14 by dobairoland

Andrysky, 2023-09-21:
@radimkarnis  I apologize for the delay in response.,
writes - `idf_monitor - a serial output monitor for esp-idf: error: unrecognized arguments: --no-reset`