espressif/esptool #893: A fatal error occurred: Failed to connect to ESP32: No serial data received. I've yet connected a capacitor between the EN pin and GND.

semen07, 2023-06-25:
### Operating System

windows 11

### Esptool Version

4.0.2

### Python Version

Python 3.11.4

### Chip Description

ESP32-WROOM-33

### Device Description

I'm using an ESP32-"WROOM-32" with a a capacitor connected between the EN pin and GND. I think the chip is "SILABS CP2102"
[image]

### How is Esptool Run

Arduino IDE

### Esptool Output

```shell
Arduino: 1.8.19 (Windows Store 1.8.57.0) (Windows 10), Board: "ESP32 Dev Module, FTDI Adapter, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 115200, Core 1, Core 1, None, Disabled"

Sketch uses 227065 bytes (17%) of program storage space. Maximum is 1310720 bytes.

Global variables use 21760 bytes (6%) of dynamic memory, leaving 305920 bytes for local variables. Maximum is 327680 bytes.

esptool.py v4.2.1

Serial port COM4

Connecting.....................................An error occurred while uploading the sketch

.

A fatal error occurred: Failed to connect to ESP32: No serial data received.

For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

peterdragun, 2023-06-26:
Can you try to connect to the chip using [Serial terminal programs](https://docs.espressif.com/projects/esptool/en/latest/esp32/troubleshooting.html#serial-terminal-programs) (e.g. miniterm)? You should see some [boot log](https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html#boot-log). This will verify if the chip is working and you are using the correct serial port.
The soldered capacitor could help just in case of issues in changing to download mode. In this case, it seems that the chip is not able to communicate at all. Can you double-check if there aren't any solder joints where there aren't supposed to be? Seems to be some kind of hardware issue.

semen07, 2023-06-27:
> Can you try to connect to the chip using [Serial terminal programs](https://docs.espressif.com/projects/esptool/en/latest/esp32/troubleshooting.html#serial-terminal-programs) (e.g. miniterm)? You [...]

I connected to ESP32 via a serial terminal program and I don't see any boot log, does this mean that my ESP32 is no longer working?

semen07, 2023-06-27:
mi sono connesso al dispositivo tramite un Serial terminal program e ho ottenuto i seguenti messaggi:
`ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13xets Jun  8 2016 00:22:57
zets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESets Jun  8 2016 00:22:57
ets J}ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0~ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOTuets Jun  8 201ets Juet{ets Jun  8 2016 00:22:57

rst:0x1 (Pets Jun  ~ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13836
load:0x40080400,len:3608
entry 0x400805f0
ets Jun  8 2016 00:22:57

rst:0x1 (Pets Jun  8 2016 00:22:57

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_F}ets Jun  8 2016 00:22:57

rst:0x1 (POWERONets Jun  8 2016 00:22:57

that means the chip is working. the capacitor i soldered does not create any problem and i connected it because otherwise the board is not recognized by the computer. i still can't figure out what is the cause of the problem. thanks anyway for the help

peterdragun, 2023-06-27:
Can you try monitoring the output while switching the chip to [download mode manually](https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html#manual-bootloader) by holding the Boot button and pressing the Reset button (while boot still pressed)?
You should see similar boot log to this one:
```
rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
```
**DOWNLOAD_BOOT**  (or just DOWNLOAD) is the important part.

If the chip stays in download mode without resetting again - you can close the serial terminal program and try running esptool with argument `--before no_reset` to skip the initial reset.
If the chip is resetting repeatedly - I am afraid your chip may be broken.

closed 2023-07-14 by dobairoland