espressif/esptool #894: Executing esptool binary file in MacOS XCode app

tobyloki, 2023-07-02:
### Operating System

macOS

### Esptool Version

v4.6.2

### Python Version

N/A

### What is the Expected Behaviour?

I'm trying to execute the binary file called esptool from https://github.com/espressif/esptool/releases/tag/v4.6.2 inside of a macOS XCode app.

I've added the file to my project and tried to execute it like so.
```swift
let scriptName = "esptool"
guard let scriptURL = Bundle.main.url(forResource: scriptName, withExtension: nil) else {
	return "Script file not found"
}
print(scriptURL.path)
let esptoolPath = scriptURL.path

let task = Process()
let pipe = Pipe()
task.executableURL = URL(fileURLWithPath: esptoolPath)
task.arguments = components
task.standardOutput = pipe
```

However, when I run it it generates some files in the temp directory and gives me this error popup. How can I fix this?
> “libpython3.8.dylib” can’t be opened because Apple cannot check it for malicious software.

[image]

dobairoland, 2023-07-02:
Have you tried to search for the error message? For example, my first hit was: https://support.apple.com/guide/mac-help/apple-cant-check-app-for-malicious-software-mchleab3a043/mac

Also, are you getting the same error when you are executing the same executable directly, and not from that app?

Another option is to use the one which was was installed by `pip`, i.e. invoke `esptool.py` directly available. The error should not occur in this case.

tobyloki, 2023-07-02:
I don't get the error when running the `esptool` binary from a terminal directly, and I also can execute `esptool.py` after a `pip` install. However, I'm trying to package everything inside of the macOS app so that python and pip wouldn't need to be installed. The error itself comes from the .dylib file that is generated in the temp directory once the `esptool` is run by the code, not when the mac app launches, and the file gets deleted automatically after attempting to run.

dobairoland, 2023-07-10:
I'm sorry but this is not an esptool issue and I cannot help more than I did above. Please search the Internet how to solve this Xcode issue.

closed 2023-07-10 by dobairoland