espressif/esptool #899: ESP32 stays in download mode after flash and hard-reset

swiipius, 2023-07-12:
### Operating System

Ubuntu 22.04

### Esptool Version

v4.5.1

### Python Version

3.10.6

### Chip Description

ESP32-WROOM-32D

### Device Description

Custom build. We built our own circuit. This board does several things and use the esp32 for Wi-Fi and Bluetooth communication.

[image]

### How is Esptool Run

Command Line

### Full Esptool Command Line that Was Run

python3 -m esptool --port /dev/%s --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 32MB 0x0 firmware_path.bin

### Esptool Output

```shell
esptool.py v4.5.1
Serial port /dev/ttyUSB1
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.....
Detecting chip type... ESP32
Chip is ESP32-D0WD (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 70:b8:f6:3d:55:5c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
WARNING: Flasher stub doesn't fully support flash size larger than 16MB, in case of failure use --no-stub.
Flash will be erased from 0x00000000 to 0x003fffff...
Compressed 4194304 bytes to 916930...
Wrote 4194304 bytes (916930 compressed) at 0x00000000 in 27.4 seconds (effective 1224.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

### More Information

The commands always succeed. In order to understand the problem, here is some context. We are using two microcontroler a stm32 to control other element on our board and an esp32 to communicate over wifi and bluetooth. They communicate together with a serial connection. 
The problem is that : when the flash is done, the esp prog still communicate with the esp prog, so the stm32 con't communicate. We've tried to cut the power line between the esp prog and esp32, reset the card, reset the esp, but with no success.

### Other Steps to Reproduce

The goal of this is to be able to flash the esp and then use the chip even if it is always plugged to ESP-PROG. We are planning to create automatic actions including a flash part and then after download sending commands from STM32. But if the esp prog is plugged to the card, we cannot use our card because it's waiting a response from the esp32 which is still in download mode.

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

dobairoland, 2023-07-12:
If I understood you correctly then you have two issues.
1. ESP32 is not reseted properly. You need auto-reset circuitry for this to work. You can read about this at: https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html#automatic-bootloader
2. ESP-Prog is connected to ESP32. Again, you should make possible to switch between ESP-Prog and STM32 to be connected to ESP32.

Both of these are hardware design tasks and related to you board. I'm sorry but I don't get it why do you think that this is an esptool issue or esptool should be do anything about it?

swiipius, 2023-07-12:
I may have forgotten to precise, but, as the image show, I'm using IO pin in order to communicate with the stm32 and the classic RX/TX are for the flash. I posted this issue here because I thought the esptool community would be best able to help me on this subject. I believed that there would be an argument I can add to my command to make the switch.

swiipius, 2023-07-12:
The documentation you sent me, apparently solve my problem. I use the command `sudo stty -F /dev/ttyUSB0 -hupcl`. Thanks for your answer !

closed 2023-07-12 by swiipius