espressif/esptool #901: --compress and --no-compress now missing

dhalbert, 2023-07-27:
### Operating System

Ubuntu 22.04

### Esptool Version

v4.6.2

### Python Version

3.10.6

radimkarnis, 2023-07-27:
Hello @dhalbert,
`--compress` and `--no-compress` are options for the `write_flash` command - they have to come **after** the `write_flash` part when constructing the cmd.

This is reflected when you run `esptool.py write_flash -h`:

```
...
  --ignore-flash-encryption-efuse-setting
                        Ignore flash encryption efuse settings
  --force               Force write, skip security and compatibility checks. Use with caution!
  --compress, -z        Compress data in transfer (default unless --no-stub is specified)
  --no-compress, -u     Disable data compression during transfer (default if --no-stub is specified)
  ```

dhalbert, 2023-07-27:
Aha, thank you! Sorry!

closed 2023-07-27 by dhalbert