espressif/esptool #902: esptool won't flash after failed upload

i-e-b, 2023-07-28:
### Operating System

Xubuntu 22.04.2 LTS

### Esptool Version

v5.2-dev-1890-g28167ea5a3

### Python Version

Python 3.10.6

### Chip Description

ESP32-WROVER-E

### Device Description

Lilygo T-SIM A7670E development board

### Hardware Configuration

Serial RX/TX connected only via USB/RS232 bridge.
No on-board USB used.
Power is from a bench-top power supply.

### How is Esptool Run

via Terminal (bash) and CLion

### Full Esptool Command Line that Was Run

esptool.py -c esp32 -p /dev/ttyUSB0 -b 115200 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 4MB 0x1000 build/bootloader/bootloader.bin 0x10000 build/LilygoEspIdf.bin 0x8000 build/partition_table/partition-table.bin

### Esptool Output

```shell
Variants of `A fatal error occurred: Invalid head of packet (0x01): Possible serial noise or corruption.`
```

### More Information

The setup has been working for a few days.
I had one upload fail (possibly causing the issue, possibly triggered by underlying issue)

Now, if I connect via serial monitor, I get a very clean signal.
The module repeats this on a normal boot:

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
ets Jul 29 2019 12:21:46
```

If I boot with `GPIO0`, I get

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x2 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_FEO_V2))
waiting for download
ets Jul 29 2019 12:21:46
```

All serial very clean.

Running an upload with trace:
```
TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=00100040
TRACE +0.000 Write 14 bytes: c0000a04000000000000100040c0
TRACE +0.000 Received full packet: 010804000712205501050000
Chip is ESP32 in Secure Download Mode
WARNING: Stub loader is not supported in Secure Download Mode, setting --no-stub
Enabling default SPI flash mode...
TRACE +0.000 command op=0x0d data len=8 wait_response=1 timeout=3.000 data=0000000000000000
TRACE +0.000 Write 18 bytes: 
    c0000d0800000000 0000000000000000 | ................
    00c0                              | ..
TRACE +0.000 Read 36 bytes: 
    01050000c0c00108 0400071220550105 | ............ U..
    0000c0c001080400 0712205501050000 | .......... U....
    c0c00118                          | ....
TRACE +0.000 Read invalid data: 
    01050000c0c00108 0400071220550105 | ............ U..
    0000c0c001080400 0712205501050000 | .......... U....
    c0c00118                          | ....
TRACE +0.000 Remaining data in serial buffer: 

A fatal error occurred: Invalid head of packet (0x01): Possible serial noise or corruption.
```

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

i-e-b, 2023-07-28:
Full trace of one of the `Invalid (unsupported) command 0x5` runs:

```
$ esptool.py --trace -c esp32 -p /dev/ttyUSB0 -b 115200 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 4MB 0x1000 build/bootloader/bootloader.bin 0x10000 build/LilygoEspIdf.bin 0x8000 build/partition_table/partition-table.bin
esptool.py v4.6.2
Serial port /dev/ttyUSB0
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.100 No serial data received.
.TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.100 No serial data received.
.TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.100 No serial data received.
.TRACE +0.050 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.025 Read 1 bytes: c0
TRACE +0.000 Read 111 bytes: 
    0108040007122055 00000000c0c00108 | ...... U........
    0400071220550000 0000c0c001080400 | .... U..........
    0712205500000000 c0c0010804000712 | .. U............
    205500000000c0c0 0108040007122055 |  U............ U
    00000000c0c00108 0400071220550000 | ............ U..
    0000c0c001080400 0712205500000000 | .......... U....
    c0c0010804000712 205500000000c0   | ........ U.....
TRACE +0.000 Received full packet: 010804000712205500000000
TRACE +0.000 Received full packet: 010804000712205500000000
TRACE +0.000 Received full packet: 010804000712205500000000
TRACE +0.000 Received full packet: 010804000712205500000000
TRACE +0.000 Received full packet: 010804000712205500000000
TRACE +0.000 Received full packet: 010804000712205500000000
TRACE +0.000 Received full packet: 010804000712205500000000
TRACE +0.000 Received full packet: 010804000712205500000000

TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=00100040
TRACE +0.000 Write 14 bytes: c0000a04000000000000100040c0
TRACE +0.015 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010a0400831df00000000000c0
TRACE +0.000 Received full packet: 010a0400831df00000000000
TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=0ca0f53f
TRACE +0.000 Write 14 bytes: c0000a0400000000000ca0f53fc0
TRACE +0.015 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010a040000a2000000000000c0
TRACE +0.000 Received full packet: 010a040000a2000000000000
TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=0ca0f53f
[... 515 lines cut ...]
    48a3a68601050000 c0c00107040048a3 | H.............H.
    a68601050000c0                    | .......
TRACE +0.000 Received full packet: 0107040048a3a68600000000
TRACE +0.000 command op=0x05 data len=16 wait_response=1 timeout=3.000 data=9c0000000100000000180000a8ebfd3f
TRACE +0.000 Write 26 bytes: 
    c000051000000000 009c000000010000 | ................
    0000180000a8ebfd 3fc0             | ........?.
TRACE +0.000 Received full packet: 0107040048a3a68601050000
Traceback (most recent call last):
  File "/home/iain/.espressif/python_env/idf5.2_py3.10_env/bin/esptool.py", line 37, in <module>
    esptool._main()
  File "/home/iain/.espressif/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages/esptool/__init__.py", line 1073, in _main
    main()
  File "/home/iain/.espressif/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages/esptool/__init__.py", line 717, in main
    esp = esp.run_stub()
  File "/home/iain/.espressif/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages/esptool/loader.py", line 969, in run_stub
    self.mem_begin(length, blocks, self.ESP_RAM_BLOCK, offs)
  File "/home/iain/.espressif/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages/esptool/loader.py", line 767, in mem_begin
    return self.check_command(
  File "/home/iain/.espressif/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages/esptool/loader.py", line 436, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "/home/iain/.espressif/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages/esptool/loader.py", line 419, in command
    raise UnsupportedCommandError(self, op)
esptool.util.UnsupportedCommandError: Invalid (unsupported) command 0x5
```

i-e-b, 2023-07-30:
Powering off the board for 10 min at a time did not help.
I left the board powered down for 30 hours, and it now uploads ok. No change to any other part of the setup.

i-e-b, 2023-07-31:
It seems stable at the moment.

No, I haven't set any eFuses.
Have ordered a few more units to test with.

I will report back if it happens again. Will see if there are any RF noise sources in that case too.

Happy for this issue to be close for now.
Thanks.

radimkarnis, 2023-08-07:
Thank you!

Please do report back if issues continue. I am closing this ticket.

closed 2023-08-07 by radimkarnis