espressif/esptool #903: DETECTED_FLASH_SIZE also for DDR (octal) in esptool as a function with extended capability

ESP32DE, 2023-08-01:
### Describe the solution you'd like

We have the [usually flash sizes](https://github.com/espressif/esptool/blob/4b249cb2bd586d1d3c4820406cb99a230db9888e/esptool/cmds.py#L43-L67) for usually SPI flashes. For testing newer flash we can set the manufacturer and size there.

How will it be handled for DDR / octal Flashes, since there are more and more manufacturer on DDR octal side also HYPERBUS?
I can't find the DDR octal flash sizes "table" for the used MX25UM in the S3-WROOM-2  or i am going been old.

[there is a header](https://github.com/espressif/esp-idf/blob/02f5b9a8982db2d59b593fab4e01ecffd35cfdff/components/esp_rom/include/esp32s3/rom/opi_flash.h#L68-L117) yes, but where we can add more manufacturer we did in the esptool "DETECTED_FLASH_SIZE"

example:
ESP32-S3-WROOM-2 N32R8v
it is 32MB Flash

```
Stub running...
Manufacturer: c2
Device: 8039
Detected flash size: 32MB
Flash type set in eFuse: octal (8 data lines)

```

Just a question and an idea note what happens next: 
i noted that the outside flash package is "eFused" octal ( 8 data lines ) in the DevKit-C and WROOM-2 ( N32R8v ) and burned - 
why extended Flash efused and eMbedded PSRAM ( also 8 data lines ) not? 
this [give me an experimental idea to try](https://twitter.com/eMbeddedHome/status/1680839940060467201) to append more as 8MB PSRAM ( octal ) on the exposed shared flash lines next days.. great. but i think - it will fail cause it is not supported by the IDF "bootloader"  and toolchain ..yet .. but i will try to run it outher this limitations customized.

radimkarnis, 2023-08-02:
Hi @ESP32DE,
I am sorry, but I am not sure what your feature request is about.

Your example shows the current behavior of esptool. Do you think more information should be displayed?
Or are you asking for support for new flash chips?

Please describe your feature request a little more. Thank you!

ESP32DE, 2023-08-02:
Hi @radimkarnis,  

asking for support for new (octal) flash chips. I can't see/found the octal flash list/DETECTED_FLASH_SIZE entry for the used MX25UM25645G in ESP32-S3-WROOM-2 / ESP32-S3-N32R8v-DevKitC-1. 

[image]
[image]

Is this 32MB octal and the 16MB octal  the only one which is supported by espressif?
Where we entry/add newer manufacturer octal flash chips, with manufacturer code and chip size like we do/did in DETECTED_FLASH_SIZE?

esptool info example from ESP32-S3-WROOM-2  which is based on ESP32-S3R8v :
in this case 32MB variation: 

```
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: XX:XX:XX:XX:XX:XX
Uploading stub...
Running stub...
Stub running...
Manufacturer: c2 ..............<<< where we add this for newer octal chips ---(dedected flash size ) 
Device: 8039  .................<<< device ID which was read from device
Detected flash size: 32MB .....<<<  that we know chip size or from where comes this info ?
Flash type set in eFuse: octal (8 data lines)
Hard resetting via RTS pin... 

```

txs

dobairoland, 2023-08-03:
I'm afraid that adding support for new flash chips is not very easy and straightforward at this moment. Yes, it is possible to add a new size to lists like you mention. It might work if one of the drivers can be used automatically. However, especially for larger chips new drivers are needed and the general ones could not be used (different ROM calls & SPI commands are needed supporting larger address space, or sometimes just different SPI command is needed because the manufacturer is different).

I renamed the title to better sum up what is needed here.

We are planning some big changes regarding the flasher stub. So I'm afraid this won't be addressed soon.

ESP32DE, 2023-08-03:
> I'm afraid that adding support for new flash chips is not very easy and straightforward at this moment. Yes, it is possible to add a new size to lists like you mention. It might work if one of the [...]

hi @dobairoland ,

exactly this. thank you.

i noted the drivers/[header](https://github.com/espressif/esp-idf/blob/53ff7d43dbff642d831a937b066ea0735a6aca24/components/esp_rom/include/esp32s3/rom/opi_flash.h#L68-L117) in esp_rom  this the way i would add the new manufacturer;  i was wounder where the 32MB info comes - so i thinked it is in the flasher stub binary embedded ... which we can not change / add, expand "so easy".

thanks for rename the title - it is very better and clear what is mean, sry @radimkarnis for not to be clearer what i was request and thanks for you effort to try figure this out and help me.

JFYI: [ESP32-S3-WoWZa-N256R256](https://twitter.com/eMbeddedHome/status/1683608614689185792) (compatible wroom pin and size ) makes the start and is now in a test phase (without castelatted holes ) (release day 13. Oct 2023 castellated holes )

i use the same flash which is on WROOM-2 ( MX25UM256..) from this side the test goes successful
now bigger and more to the limits and different manufacturer's   ( there is also a 4Gbit DDR Octal p(s)(D)RAM sampler also to test )
  
[image]
( has just been delivered )

best wishes
;-)