espressif/esptool #904: STDOUT redirection in util.py not working sufficiently.

gotfredsen, 2023-08-02:
### Operating System

MacOS

### Esptool Version

v4.6.2

### Python Version

Python 3.11.4

### Full Esptool Command Line that Was Run

['-b', '1000000', 'write_flash', '0x1000', '/var/folders/19/sw8b32492278my8jjky7sx9h0000gn/T/tmpzpne6xiy/CWV_PPU_fw.bin']

### Esptool Output

```shell
File "/opt/homebrew/lib/python3.11/site-packages/esptool/util.py", line 69, in print_overwrite
    if sys.stdout.isatty():
       ^^^^^^^^^^^^^^^^^
AttributeError: 'Multiline' object has no attribute 'isatty'
```

### What is the Expected Behaviour?

Redirection of STDOUT working... see below

### More Information

I called esptool from another Python program:
```Python
                    cmd = [ "-b", "1000000", "write_flash", "0x1000", firmware ]
                    try:
                        esptool.main(cmd)
```
The other Python program tries to redirect the STDOUT so it can be displayed in a window, when the above error occurs.

### Other Steps to Reproduce

I managed to get it to work with this - temporary - fix: 
Canging `util.py`
from:
```Python
    if sys.stdout.isatty():
        print("\r%s" % message, end="\n" if last_line else "")
    else:
        print(message)
```
to:
```Python
    try:
        if sys.stdout.isatty():
            print("\r%s" % message, end="\n" if last_line else "")
        else:
            print(message)
    except:
        print(message)
```

radimkarnis, 2023-08-07:
Hello @gotfredsen,

thanks for the report and sorry for not responding sooner.

I will try to reproduce the issue and issue a proper fix if possible.

dobairoland, 2023-08-09:
Hi @gotfredsen. Please report back if the fix haven't solved your issue in full.

gotfredsen, 2023-08-09:
Thank you @dobairoland, that solved the problem ... in full!