espressif/esptool #906: Unable to erase_flash for ESP32-S3-WROOM-1-N4R2 bare module

whatdtech, 2023-08-06:
### Operating System

windows 10

### Esptool Version

3.0

### Python Version

3.10

### Chip Description

ESP32-S3-WROOM-1-N4R2

### Device Description

Bare module. product link is [https://www.digikey.in/en/products/detail/espressif-systems/ESP32-S3-WROOM-1-N4R2/16162643](url)

### Hardware Configuration

GPIO 0 & GPIO 46 are grounded as per documentation. Nothing else connected.

### How is Esptool Run

command prompt

### Full Esptool Command Line that Was Run

esptool.py --port COM7 erase_flash, esptool.py --chip esp32s3 --port COM7 erase_flash,  esptool.py --chip esp32s3beta2 --port COM7 erase_flash

### Esptool Output

```shell
esptool.py v3.0
Serial port COM7
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header
```

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

whatdtech, 2023-08-06:
upgraded esptool to 4.6.2 and python to 3.10.11 and still same issue

radimkarnis, 2023-08-07:
Hello @whatdtech,
can you please describe the other connections? 
- How are you powering the module?
- Are you connecting with a USB-to-UART bridge (to TX and RX pins) or to the integrated USB-Serial/JTAG unit (pins 19 and 20)?

whatdtech, 2023-08-07:
Powering it through 12 volts 10 amps SMPS stepping down using LM2596. Using TX /RX pins not GPIO 19/20

radimkarnis, 2023-08-07:
Thanks for the info.

> Before flash serial output is:
rst:0x1 (POWERON),boot:0x3 (DOWNLOAD(USB/UART0))

Are you also seeing the `waiting for download` line?

Let's try to analyze the problem. The chip is manually pulled into the download mode (because GPIO0 is pulled low), that means esptool doesn't see the `rst:0x1 (POWERON)...` boot log. Because of that, esptool cannot give more info about what's happening. If auto-reset circuitry was implemented, esptool would see the boot log and you might get an error like `Download mode successfully detected, but getting no sync reply: The serial TX path seems to be down.` or `Wrong boot mode detected (0xXX)! The chip needs to be in download mode.`.

But that's just an explanation why we are not seeing any info about what's going on.

The ESP chip successfully sends data to the host computer, but doesn't receive any sync commands sent by ``esptool``. This might imply a problem with the TX line running from the host to the ESP device (the `The serial TX path seems to be down.` case). Please check that connection and let me know if that helped.

whatdtech, 2023-08-07:
```
Build⸮ESP-ROM:esp32s3-20210327
17:46:01.774 -> Build:Mar 27 2021
17:46:01.774 -> rst:0x1 (POWERON),boot:0x3 (DOWNLOAD(USB/UART0))
17:46:01.774 -> waiting for download
```
. this is the output in serial monitor. I will try to implement auto reset circuit. What is the difference between TX/RX & GPIO 19/20 usage. Now im using TX/RX with ch340g chip. For GPIO 19/20 can I connect them directly to micro usb?

radimkarnis, 2023-08-07:
So you can see the boot log, therefore, the ESP->PC comms work. Do you have a way to verify the PC->ESP line (maybe with a logic analyzer)?

> What is the difference between TX/RX & GPIO 19/20 usage. Now im using TX/RX with ch340g chip. For GPIO 19/20 can I connect them directly to micro usb?

Exactly, the 19/20 pins can be connected directly to the D+/D- USB lines. The USB-to-UART chip and auto-reset circuitry are then not needed. You can read more information about this feature here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/usb-serial-jtag-console.html

whatdtech, 2023-08-07:
@radimkarnis  thank you. I shall implement usb route and let you know the output.

whatdtech, 2023-08-07:
@radimkarnis . Tried with usb and getting error as **Unknown USB Device(Device descriptor request failed)**. What kind of a driver needs to be installed. And how to specify port for using with esptool?

radimkarnis, 2023-08-08:
It should enumerate and show itself like

```
/dev/cu.usbmodem111201
    desc: USB JTAG/serial debug unit
    hwid: USB VID:PID=303A:1001 SER=7C:DF:A1:86:00:44 LOCATION=0-1.1.1.2
```

The necessary drivers can be installed with the [IDF installer fo Windows](https://dl.espressif.com/dl/esp-idf/).

>  And how to specify port for using with esptool?

you can use the `--port`/`-p` option like `esptool.py -p COMX <command>`.

radimkarnis, 2023-08-21:
Closing for inactivity. 

I do believe this is a HW issue with your connections. Please try some more debugging and feel free to reopen this issue if needed! Thank you.

closed 2023-08-21 by radimkarnis