espressif/esptool #908: A fatal error occurred: Failed to write to target RAM (result was 01070000: Operation timed out)

Kymasola, 2023-08-18:
### Operating System

Windows 10

### Esptool Version

v4.5.1

### Python Version

Python 3.11.3

### Chip Description

ESP-WROOM-32

### Device Description

Dev Module on a breadboard.

### How is Esptool Run

Arduino IDE, esptool.py, Arduino Cloud

### Full Esptool Command Line that Was Run

esptool.py --port COM3 read_flash 0x00000 0x100000 image1M.bin     

### Esptool Output

```shell
esptool.py v4.6.2
Serial port COM3
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.......
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 54:43:b2:c5:0d:64
Uploading stub...

A fatal error occurred: Failed to write to target RAM (result was 01070000: Operation timed out)
```

### More Information

I first tried it with Arduino IoT Cloud Sketch and I get this
`esptool.py v4.2.1
Serial port COM3
Connecting....
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 54:43:b2:c5:0d:64
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00120fff...
Compressed 18912 bytes to 13025...
Writing at 0x00001000... (100 %)
Wrote 18912 bytes (13025 compressed) at 0x00001000 in 0.2 seconds (effective 844.7 kbit/s)...
File md5: e7d5057d627598659b7aa7b697f6cff6
Flash md5: 4ca6a95d1789c3d6317811ab47dfa684
MD5 of 0xFF is 4ca6a95d1789c3d6317811ab47dfa684
A fatal error occurred: MD5 of file does not match data in flash`

I open the Full Editor and get the same output. So I tried using an example code in Arduino IDE which is "WiFiScan"
'esptool.py v4.5.1
Serial port COM3
Connecting....
Chip is ESP32-D0WD-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 54:43:b2:c5:0d:64
Uploading stub...

A fatal error occurred: Failed to write to target RAM (result was 01070000: Operation timed out)
Failed uploading: uploading error: exit status 2'

I've tried changing baud rates and doing --no-stub but it doesnt work.

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2023-08-22:
Hello @Kymasola,

this issue is usually caused by delays in the UART communication, which can be a result of faulty drivers or the USB-to-UART bridge.

See this similar issue: https://github.com/espressif/esptool/issues/280#issuecomment-907622126

A possibility is to use a [configuration file](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/configuration-file.html) and increase the timeouts used by esptool.

Please try to do this and let me know if that helped. Thanks!

closed 2023-09-08 by dobairoland

hhartzer, 2024-02-16:
I'm getting this error as well with 4.7.0. I tried setting `mem_end_rom_timeout = 10`, but it still seems to timeout rapidly.

lawrencebensaid, 2024-07-05:
Any update on this? I have the same issue. Feels like I've tried everything..

kiler129, 2024-07-30:
@lawrencebensaid  @Stan-T889:

If you're on macOS this is most likely issue with WCH CH3xx UART chipset. Either use a different one (e.g. legitimate FTDI, RPi or similar), or if you trust it you can install https://github.com/WCHSoftGroup/ch34xser_macos
This isn't a new problem and nothing to do with `esptool` - https://meshtastic.discourse.group/t/failed-to-write-to-target-ram-result-was-01070000/3874/25

Jumbo125, 2024-12-21:
same on windows 10

lennartkoopmann, 2024-12-22:
FYI, this happened to me because I forgot to switch the ESP32 power supply from an external power supply to the `+3.3V` and `GND` pins of my `CP2102` USB UART adapter. Common ground is critical.

Interestingly, it was able to pull system information like chip and MAC before failing:

```
Connecting...
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 88:13:bf:31:27:60
Uploading stub...

A fatal error occurred: Failed to write to target RAM (result was 01070000: Operation timed out)
```

edujo3, 2025-01-15:
> FYI, this happened to me because I forgot to switch the ESP32 power supply from an external power supply to the `+3.3V` and `GND` pins of my `CP2102` USB UART adapter. Common ground is critical. [...]

Can you elaborate a little more? I have the same problem and I haven't been able to solve it. I'm loading the sketch from a Macintosh.

dosipod, 2025-01-29:
I was having the same issue on macOS , fixed as @kiler129  listed by using https://github.com/WCHSoftGroup/ch34xser_macos

[image]

[image]

rolark, 2025-02-20:
New drivers fixed it for me as well as @dosipod  and @kiler129 noted (thanks!) . Default Mac OS drivers partially worked for me (uploading 1 sketch just once and subsequently throwing errors above)

bingcheng1998, 2025-03-15:
first install https://www.wch.cn/downloads/CH34XSER_MAC_ZIP.html 
[image]
remenber to open this extend in "general" config pad, and then reboot your mac.

then you can check:

ls /dev/cu.*
/dev/cu.BLTH                    /dev/cu.Bluetooth-Incoming-Port /dev/cu.wchusbserial5A4B0277261

you should realize that " /dev/cu.usbmodem5A4B0277261" turns into "/dev/cu.wchusbserial5A4B0277261"

it must start with "wchusbserial" or it will not work.

$ esptool.py -p /dev/cu.wchusbserial5A4B0277261 -b 115200 write_flash -fs 16MB -fm dio -e 0x0 ~/Downloads/merged-binary.bin
[image]

iseebadpixels, 2025-07-22:
Installing the driver mentioned earlier solved issue for me