espressif/esptool #910: hardreset without flashing

ligantx, 2023-08-21:
### Is your feature request related to a problem?

For testing purposes I would like to hard reset (emulate reset button pressing) **without** flashing (I have already flashed in a previous step). 
It looks like `esptool.py --after hard_reset -p /dev/ttyUSBX` doesnt work without flashing and I dont find "reset only" command [here](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/advanced-options.html#reset-after-operation).

### Describe the solution you'd like

A command like:
`esptool.py --reset`

radimkarnis, 2023-08-21:
Hi @ligantx,
esptool doesn't support resetting the chip without running any command. This can be achieved by writing a simple script replicating the [`HardReset` behavior](https://github.com/espressif/esptool/blob/master/esptool/reset.py#L114).

On the other hand, you can use any commands that don't manipulate the flash memory to do this. e.g. running `esptool.py --no-stub flash_id` resets the chip (and prints some info about the flash chip - you can discard that).

radimkarnis, 2023-08-21:
Great! Thanks for the future reference info.

Closing the issue as it has been solved.

closed 2023-08-21 by radimkarnis