espressif/esptool #912: Return code 1 for the " can not exit the download mode over USB" warning confuses external processes

EternityForest, 2023-08-23:
### Is your feature request related to a problem?

When a manual reset is required, the flasher process exits with return code 1, causing scripts that call it to think there was a failure.

### Describe the solution you'd like

I think it would be helpful to either change it to return 0 unless there is a true error, or better yet reserve and document a specific error code just for indicating that a manual reset is required, so that automated tools can correctly interpret things.

I'm not that familiar with the tool in general, but I think having some well defined way to access the status would be helpful.

radimkarnis, 2023-08-29:
Hello @EternityForest,

the reset behavior can be changed with the `--after` option. If `--after no_reset` is used, then the error message is not printed and esptool exits with return code 0.

`--after hard_reset` is set by default, resulting in the error if USB-OTG mode is used (other modes don't require manual reset).

I believe the correct fix is for the automated tools to correctly configure esptool and set the `--after` option accordingly if USB-OTG is used to prevent the failure.

EternityForest, 2023-08-29:
It would be nice to keep the message so the user knows what to do, and still try to reset if it's possible to do so.  I suppose you could keep a list on the tool side of which boards should get which option, but that seems not very DRY.

Maybe an `--after auto_reset` option could be added to do it?