espressif/esptool #916: Custom SPI Pin Configuration support for ESP32-S3

alanmosely, 2023-08-31:
### Is your feature request related to a problem?

I would like to be able to save data directly to an external flash on a custom PCB with an ESP32-S3 as part of my production line 

### Describe the solution you'd like

I would like support for Custom SPI Pin Configuration in esptool.py for ESP32-S3 devices, I am trying to target SPI2. I would like the following command to work: `esptool.py --chip esp32s3  write_flash --spi-connection 19,20,9,35,47 0x0 storage.bin`

### Describe alternatives you've considered

I've tried to run the command above but it complains the pins are above 31

### Additional context

Is this something I could contribute myself if it is not planned?

alanmosely, 2023-08-31:
Somewhat related issues: https://github.com/espressif/esptool/issues/243 & https://github.com/espressif/esptool/issues/903

radimkarnis, 2023-09-01:
Hello @alanmosely,
thank you for the feature request!

this feature is currently implemented only on the classic ESP32.

We will look into the possibility of expanding this to other chips.

alanmosely, 2023-09-01:
Thank you, is this something you plan on doing soon? Is it something that I could help to build?
If no to both, how would you advise addressing my issue of needing to flash an external chip as part of my production line?

radimkarnis, 2023-09-04:
> is this something you plan on doing soon?

This is currently not planned to be done in this quarter. We will re-evaluate the priority of this task and see if this can be done during the next quarter.

> Is it something that I could help to build?

You can definitely try helping build that, PRs are more than welcome!

Pointers:
- Attaching SPI flash in the stub flasher: 
https://github.com/espressif/esptool/blob/master/flasher_stub/stub_commands.c#L182-L197
https://github.com/espressif/esptool/blob/master/flasher_stub/stub_flasher.c#L472-L491
- Attaching SPI flash in esptool:
https://github.com/espressif/esptool/blob/master/esptool/__init__.py#L739-L750

closed 2023-11-10 by espressif-bot