espressif/esptool #918: Port doesn't exist

Massimiliano-solutiontech, 2023-09-07:
### Operating System

Ubuntu 22.04 LTS

### Esptool Version

v4.6.2

### Python Version

Python 3.10

### Full Esptool Command Line that Was Run

esptool flash_id

### Esptool Output

```shell
esptool.py v4.6.2
Found 3 serial ports
Serial port /dev/ttyUSB1
/dev/ttyUSB1 failed to connect: Could not open /dev/ttyUSB1, the port doesn't exist
Serial port /dev/ttyUSB0
/dev/ttyUSB0 failed to connect: Could not open /dev/ttyUSB0, the port doesn't exist
Serial port /dev/ttyS4
/dev/ttyS4 failed to connect: Could not open /dev/ttyS4, the port doesn't exist

A fatal error occurred: Could not connect to an Espressif device on any of the 3 available serial ports.
```

### What is the Expected Behaviour?

Connect and read the `flash_id`

Massimiliano-solutiontech, 2023-09-07:
Solved, the user wasn't in the `dialout` group. Maybe the error could be more descriptive:

```python
try:
    self._port = serial.serial_for_url(port)
except serial.serialutil.SerialException:
    raise FatalError(f"Could not open {port}, {sys.exc_info()[1]}")
```

instead of the actual

```python
try:
    self._port = serial.serial_for_url(port)
except serial.serialutil.SerialException:
    raise FatalError(f"Could not open {port}, the port doesn't exist")
```

jakub-kocka, 2023-09-11:
Thank you @Massimiliano-solutiontech for the good poin, we'll do that change.

closed 2023-09-14 by Massimiliano-solutiontech