espressif/esptool #920: Cannot flash ESP32 DevkitV1 on Mac M1

LuizTokuhara, 2023-09-19:
### Operating System

macOS Ventura 13.5.2

### Esptool Version

esptool.py v4.6.2

### Python Version

Python 3.9.6

### Chip Description

ESP32-WROOM32

### Device Description

DevkitV1

### How is Esptool Run

Mac Terminal

### Full Esptool Command Line that Was Run

python3 -m esptool   --chip esp32  --port /dev/cu.usbserial-0001   --baud 115200   --after hard_reset write_flash   -z   --flash_mode dio    --flash_freq 40m   --flash_size detect   0x1000 "binaries/bootloader.bin"   0x8000 "binaries/partitions_espruino.bin"    0x10000 "binaries/espruino_esp32.bin"

### Esptool Output

```shell
esptool.py v4.6.2
Serial port /dev/cu.usbserial-0001
Traceback (most recent call last):
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/Users/user/Library/Python/3.9/lib/python/site-packages/esptool/__main__.py", line 9, in <module>
    esptool._main()
  File "/Users/user/Library/Python/3.9/lib/python/site-packages/esptool/__init__.py", line 1073, in _main
    main()
  File "/Users/user/Library/Python/3.9/lib/python/site-packages/esptool/__init__.py", line 678, in main
    esp = esp or get_default_connected_device(
  File "/Users/user/Library/Python/3.9/lib/python/site-packages/esptool/__init__.py", line 961, in get_default_connected_device
    _esp = chip_class(each_port, initial_baud, trace)
  File "/Users/user/Library/Python/3.9/lib/python/site-packages/esptool/loader.py", line 303, in __init__
    self._port = serial.serial_for_url(port)
  File "/Users/user/Library/Python/3.9/lib/python/site-packages/serial/__init__.py", line 90, in serial_for_url
    instance.open()
  File "/Users/user/Library/Python/3.9/lib/python/site-packages/serial/serialposix.py", line 332, in open
    self._reconfigure_port(force_update=True)
  File "/Users/user/Library/Python/3.9/lib/python/site-packages/serial/serialposix.py", line 517, in _reconfigure_port
    termios.tcsetattr(
termios.error: (22, 'Invalid argument')
```

### More Information

If I try multiple times the macbook turns off the usb ports and shows a message saying that a usb device is using too much power.

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

dobairoland, 2023-09-20:
Are you able to establish serial communication with any terminal program?

BTW, I'm not sure it related but just yesterday we had an issue with the same built-in Python of macOS being broken (https://github.com/espressif/esp-idf/issues/12265).

dobairoland, 2023-09-20:
I did mean that please try a serial terminal program instead of esptool and see if you can connect with that one. I'm sorry for not being clear.

Here is a short guide: https://jasonmurray.org/posts/2020/serialconsolemacos/

LuizTokuhara, 2023-09-20:
This is the output:
```
sudo cu -s 115200 -l /dev/tty.usbserial-0001
cu: open (/dev/tty.usbserial-0001): Device not configured
cu: /dev/tty.usbserial-0001: Line in use
```

LuizTokuhara, 2023-09-20:
@dobairoland got it, will try to restart or use another PC and see what happens.
Thanks for your help.

LuizTokuhara, 2023-09-21:
@dobairoland Just as an update, I've run on a Mac Intel and it just works.
Must be something with Macs with M1 processors.

radimkarnis, 2023-09-21:
Hello @LuizTokuhara,
I am running Ventura 13.5.2 on Mac M1 and everything works here. 
Are using a USB hub? Have you tried another devkit?

LuizTokuhara, 2023-09-21:
Hi @radimkarnis interesting, on the Mac M1 (only have USB-C) I'm using a USB hub but not on the Mac Intel.
I didn't try another kit, just have this one 🥲

radimkarnis, 2023-09-22:
@LuizTokuhara the USB hub can easily be the source of your issues, please try directly with your Mac's USB-C port.

dobairoland, 2023-09-25:
I'm closing this because it doesn't look like an esptool issue.

closed 2023-09-25 by dobairoland

LuizTokuhara, 2023-09-25:
@radimkarnis I'll see if I find a usb-c to usb cable to test without the usb hub