espressif/esptool #921: Encrypted flash write on ESP32 - Invalid (unsupported command 0xd4

DCSBL, 2023-09-20:
### Operating System

macOS 13.4.1

### Esptool Version

4.7.dev1 (and tested with 4.6.2)

### Python Version

3.9.4

### Chip Description

ESP32-WEROOM-32E

### Device Description

Module on custom PCB, connected to FTDI programmer. Normal reads/writes are working as expected.

### Hardware Configuration

Bare minimum, no external peripherals except for power supply or flashing

### How is Esptool Run

Terminal in VSCode

### Full Esptool Command Line that Was Run

esptool.py -b 460800 --before default_reset --after no_reset --chip esp32 --no-stub write_flash --flash_mode dio --flash_size 4MB --flash_freq 40m 0x121000 cusotm-nvs/bin/custom-1.bin --encrypt-files 0x120000 cusotm-nvs/keys/keys-custom-1.bin

### Esptool Output

```shell
esptool.py v4.7.dev1
Found 3 serial ports
Serial port /dev/cu.usbserial-02BFRZ28
Connecting....
Chip is ESP32-D0WD-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 94:3c:c6:e2:b3:c8
Changing baud rate to 460800
Changed.
Enabling default SPI flash mode...
Configuring flash size...
Flash will be erased from 0x00121000 to 0x00124fff...
Erasing flash...
Took 0.05s to erase flash block
Writing at 0x00120000... (25 %)Traceback (most recent call last):
  File "~/.espressif/python_env/idf5.1_py3.9_env/bin/esptool.py", line 37, in <module>
    esptool._main()
  File "~/.espressif/python_env/idf5.1_py3.9_env/lib/python3.9/site-packages/esptool/__init__.py", line 1072, in _main
    main()
  File "~/.espressif/python_env/idf5.1_py3.9_env/lib/python3.9/site-packages/esptool/__init__.py", line 867, in main
    operation_func(esp, args)
  File "~/.espressif/python_env/idf5.1_py3.9_env/lib/python3.9/site-packages/esptool/cmds.py", line 604, in write_flash
    esp.flash_encrypt_block(block, seq)
  File "~/.espressif/python_env/idf5.1_py3.9_env/lib/python3.9/site-packages/esptool/loader.py", line 858, in flash_encrypt_block
    self.check_command(
  File "~/.espressif/python_env/idf5.1_py3.9_env/lib/python3.9/site-packages/esptool/loader.py", line 436, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "~/.espressif/python_env/idf5.1_py3.9_env/lib/python3.9/site-packages/esptool/loader.py", line 419, in command
    raise UnsupportedCommandError(self, op)
esptool.util.UnsupportedCommandError: Invalid (unsupported) command 0xd4
```

### More Information

[... 575 lines cut ...]
Reading with esp_flash_read:
I (371) example: 0x3ffb4c90   fb 78 93 6a ef e7 c8 2c  91 70 7d ea cd 77 d4 60  |.x.j...,.p}..w.`|
I (381) example: 0x3ffb4ca0   ca 60 d8 aa 84 b3 eb 75  1d bf 12 0f f4 da 15 de  |.`.....u........|
I (391) nvs: NVS key partition empty, generating keys
I (451) nvs: NVS partition "nvs" is encrypted.
I (471) example: NVS partition "custom_nvs" is encrypted.
I (471) main_task: Returned from app_main()
```

7. Tested rewrites `idf.py encrypted-flash`. Passes (same log output)
8. Run `esptool.py -b 460800 --before default_reset --after no_reset --chip esp32 --no-stub write_flash --flash_mode dio --flash_size 4MB --flash_freq 40m 0x121000 cusotm-nvs/bin/custom-1.bin --encrypt-files 0x120000 cusotm-nvs/keys/keys-custom-1.bin`. See esptool output above.

Also, the ESP prints this after a failed write:

```
...
I (391) nvs: NVS key partition empty, generating keys
I (451) nvs: NVS partition "nvs" is encrypted.
I (471) example: NVS partition "custom_nvs" is encrypted.
...
```

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

DCSBL, 2023-09-25:
We currently have an workaround by flashing all partitions first, including the custom NVS partition, and then let the bootloader encrypt all required partitions.

We suspect this is not an option for mass production.

radimkarnis, 2023-09-25:
Hi @DCSBL,
thanks for the report! 

could you please also try older releases of esptool (e.g `4.2.1` or even `3.3.3`). This would help to rule out any possible regressions.

I will try to reproduce the issue and investigate more.

DCSBL, 2023-09-25:
Yes, will do this now.

DCSBL, 2023-09-25:
Using esptool==4.2.1 and esptool==3.3.3 I get the following: (added `-p /dev/cu.usbserial-02BFRZ28` to the full command.)

```
esptool.py v3.3.3
Serial port /dev/cu.usbserial-02BFRZ28
Connecting......................................

A fatal error occurred: Failed to connect to ESP32: Wrong boot mode detected (0x13)! The chip needs to be in download mode.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
```

<br><details><summary>Full log with --trace</summary>
<pre>
esptool.py v3.3.3
Serial port /dev/cu.usbserial-02BFRZ28
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.000 Read 12 bytes: 643a30783430303830343034
TRACE +0.000 Read invalid data: 643a30783430303830343034
TRACE +0.000 Remaining data in serial buffer: 
.TRACE +0.103 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.100 No serial data received.
.TRACE +0.107 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.091 Read 1 bytes: 1b
TRACE +0.000 Read invalid data: 1b
TRACE +0.000 Remaining data in serial buffer: 
.TRACE +0.102 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.000 Read 2 bytes: 3530
TRACE +0.000 Read invalid data: 3530
TRACE +0.000 Remaining data in serial buffer: 
.TRACE +0.105 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.000 Read 4 bytes: 8dbdc995
TRACE +0.000 Read invalid data: 8dbdc995
TRACE +0.000 Remaining data in serial buffer: 
.TRACE +0.719 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
[... 295 lines cut ...]
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.000 Read 13 bytes: 3429206370755f7374617274a2
TRACE +0.000 Read invalid data: 3429206370755f7374617274a2
TRACE +0.000 Remaining data in serial buffer: 025aa509
.TRACE +0.102 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.000 Read 2 bytes: 6d70
TRACE +0.000 Read invalid data: 6d70
TRACE +0.000 Remaining data in serial buffer: 
.

A fatal error occurred: Failed to connect to ESP32: Wrong boot mode detected (0x13)! The chip needs to be in download mode.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
</pre>
</details>

radimkarnis, 2023-09-25:
Hi @DCSBL,
that is another unrelated issue - your host cannot automatically reset the ESP into download mode with older esptool versions. You can try manually bridging IO0 to GND or pressing and holding the BOOT button while flashing to do this manually. Then the connection should work and the real output will be visible.

DCSBL, 2023-09-25:
Hmm, tried that but didn't work. Let me do that again..

radimkarnis, 2023-09-25:
Please see instructions here: https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html#manual-bootloader

DCSBL, 2023-09-25:
Ah, that works. But sadly got the same response..

```
esptool.py v3.3.3
Serial port /dev/cu.usbserial-02BFRZ28
Connecting.........
Chip is ESP32-D0WD-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 94:3c:c6:e2:ae:6c
Changing baud rate to 460800
Changed.
Enabling default SPI flash mode...
Configuring flash size...
Flash will be erased from 0x003f9000 to 0x003fcfff...
Erasing flash...
Took 0.17s to erase flash block
Wrote 16384 bytes at 0x003f9000 in 0.7 seconds (189.0 kbit/s)...
Hash of data verified.
Erasing flash...
Took 0.05s to erase flash block
Writing at 0x003ff000... (25 %)Traceback (most recent call last):
  File "~/.espressif/python_env/idf5.1_py3.9_env/bin/esptool.py", line 5651, in <module>
    _main()
  File "~/.espressif/python_env/idf5.1_py3.9_env/bin/esptool.py", line 5644, in _main
    main()
  File "~/.espressif/python_env/idf5.1_py3.9_env/bin/esptool.py", line 5076, in main
    operation_func(esp, args)
  File "~/.espressif/python_env/idf5.1_py3.9_env/bin/esptool.py", line 4231, in write_flash
    esp.flash_encrypt_block(block, seq)
  File "~/.espressif/python_env/idf5.1_py3.9_env/bin/esptool.py", line 854, in flash_encrypt_block
    self.check_command(
  File "~/.espressif/python_env/idf5.1_py3.9_env/bin/esptool.py", line 520, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "~/.espressif/python_env/idf5.1_py3.9_env/bin/esptool.py", line 505, in command
    raise UnsupportedCommandError(self, op)
__main__.UnsupportedCommandError: Invalid (unsupported) command 0xd4
```

DCSBL, 2023-10-05:
Sorry to bother you, but did you have time to look into this for me? We need to order modules soon.. 🙂

Harshal5, 2023-10-05:
Hello @DCSBL ,
Thanks for the report!

The ESP32 ROM does not support the encrypted-writes command. Thus, the need for the esptool's flasher-stub is necessary, so could you retry by just removing the `--no-stub` flag from the above command in the case of ESP32?

```
esptool.py -b 460800 --before default_reset --after no_reset --chip esp32 write_flash --flash_mode dio --flash_size 4MB --flash_freq 40m 0x121000 cusotm-nvs/bin/custom-1.bin --encrypt-files 0x120000 cusotm-nvs/keys/keys-custom-1.bin
```

DCSBL, 2023-10-06:
Yes, that's it! So easy when you know what to do. Thanks!

Closing this as my problem is fixed. But besides that, did I miss any documentation about this behaviour or is there an opportunity to improve something?

closed 2023-10-06 by DCSBL

dobairoland, 2023-10-06:
@Harshal5 Thank you for your help!

Harshal5, 2023-10-06:
> is there an opportunity to improve something?

Yes actually this needs to be documented, shall update the [Encrypted Flash Protection](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/basic-commands.html#encrypted-flash-protection) docs page.