espressif/esptool #922: Failed to connect to ESP32-C6: Wrong boot mode detected (0xc)

Massimiliano-solutiontech, 2023-09-26:
### Operating System

macOS 13.4.1 (c)

### Esptool Version

v4.6.2 - v4.7-dev

### Python Version

Python 3.11.5

### Chip Description

ESP32-C6

### Device Description

Custom hardware

### How is Esptool Run

No IDE

### Full Esptool Command Line that Was Run

esptool.py --port /dev/cu.usbserial-0005 --chip esp32-c6 read_mac

### Esptool Output

```shell
esptool.py --port /dev/cu.usbserial-0005 --chip esp32-c6 --trace read_mac
esptool.py v4.7-dev
Serial port /dev/cu.usbserial-0005
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.105 No serial data received.
.TRACE +0.052 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.105 No serial data received.
.TRACE +0.053 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.105 No serial data received.
.TRACE +0.057 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.104 No serial data received.
[... 276 lines cut ...]
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.105 No serial data received.
.

A fatal error occurred: Failed to connect to ESP32-C6: Wrong boot mode detected (0xc)! The chip needs to be in download mode.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
```

### More Information

Using [`IO Ninja`](https://ioninja.com) I can correctly reading the serial output and even put the ESP32-C6 in download mode by manually trigger the `DTR` and `RTS` signals

`DTR on`, `RTS on`

```sh
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0x4 (DOWNLOAD(USB/UART0/SDIO_FEI_FEO))
waiting for download
```

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2023-09-26:
Hi @Massimiliano-solutiontech,
is your custom hardware correctly implementing the auto-reset circuitry?  It is explained in detail [here](https://docs.espressif.com/projects/esptool/en/latest/esp32c6/advanced-topics/boot-mode-selection.html#automatic-bootloader) or in [this webinar](https://www.youtube.com/live/zh-Y_s4X6zs?si=HGB4SW57w4F9nv6L&t=1272).

> DTR on, RTS on 

This shouldn't be possible, the auto-reset circuit should prevent both of these lines from being asserted at the same time according to this truth table (notice the `IO0` and `EN` pins not changing their states if both `DTR` and `RTS` are on at the same time):
[image]

This leads me to believe maybe you haven't implemented the transistor network like this (changing `IO0` to `IO9`): 
[image]

You can see the esptool reset sequence [here](https://github.com/espressif/esptool/blob/master/esptool/reset.py#L61-L90), please notice how it's not doing `DTR on, RTS on` at any time.

radimkarnis, 2023-10-09:
I am closing this ticket for inactivity. Also, this doesn't seem like an esptool issue, but a wrong HW design.
Feel free to reopen this if you believe otherwise.

closed 2023-10-09 by radimkarnis