espressif/esptool #923: Building merged binary command issue (IDFGH-11152)

krupis, 2023-10-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.0.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

I am trying to build a one-file binary that contains the bootloader, partition, app and all other binaries in a single file.

I have read the official documentation
https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/basic-commands.html#merge-binaries-for-flashing-merge-bin
He uses a simple command:
```
esptool.py --chip esp32 merge_bin -o merged-flash.bin "@flash_args"
```

In my case, instead of using esptool.py, I must use full esptool and python paths (I think this has something to do with the way I have installed esp-idf). I have installed it via the vscode esp-idf extension setup.

So I cd in to my project build dir and run the following command with full python and esptool paths:

### What is the actual behavior?

After executing the following command in my project build directory:
```
C:\Users\kapeika.lu\Desktop\WORK\myproject\myproject2\myproject-fw\build> C:\Users\kapeika.lu\.espressif5\python_env\idf5.0_py3.8_env\Scripts\python.exe C:\Users\kapeika.lu\esp5\esp-idf\components\esptool_py\esptool\esptool.py --chip esp32 merge_bin -o flash_image.bin @flash_args
```
I am getting an error message that I am missing some arguments (logs attached to **Build or Installation Logs** section).

### Steps to reproduce.

1. cd into build directory
2. Run a command ```esptool.py --chip ESP32 merge_bin -o merged-flash.bin @flash_args``` 

Keep in mind that you may need to use full paths for python and esptool.py

### Build or installation Logs.

```plain
usage: esptool merge_bin [-h] --output OUTPUT [--format {r,a,w}] [--flash_freq {keep,80m,60m,48m,40m,30m,26m,24m,20m,16m,15m,12m}] [--flash_mode {keep,qio,qout,dio,dout}]
                         [--flash_size {keep,256KB,512KB,1MB,2MB,2MB-c1,4MB,4MB-c1,8MB,16MB,32MB,64MB,128MB}] [--spi-connection SPI_CONNECTION] [--target-offset TARGET_OFFSET]
                         [--fill-flash-size {256KB,512KB,1MB,2MB,4MB,8MB,16MB,32MB,64MB,128MB}]
                         <address> <filename> [<address> <filename> ...]
esptool merge_bin: error: the following arguments are required: <address> <filename>
```

### More Information.

I would appreciate any clarification regarding this issue.

jakub-kocka, 2023-10-04:
Hi @krupis, I have been able to reproduce your issue on Windows machine.

I think the problem will be with the command, there are quotes (") missing around the `@flash_args`.
In command prompt there is no problem with it but for example in PowerShell the `@` is used as array operator and this will probably be the VS Code terminal.

So please try to adjust your command like this:

    C:\Users\kapeika.lu\.espressif5\python_env\idf5.0_py3.8_env\Scripts\python.exe C:\Users\kapeika.lu\esp5\esp-idf\components\esptool_py\esptool\esptool.py --chip esp32 merge_bin -o flash_image.bin "@flash_args"

krupis, 2023-10-05:
@jakub-kocka 
Thanks for the reply. I can confirm that what you have suggested made some progress. Now I am getting different kind of error:

```
esptool esp32 merge_bin -o flash_image.bin --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 bootloader/bootloader.bin 0x10000 myproject.bin 0x8000 partition_table/partition-table.bin 0xd000 ota_data_initial.bin 0xf000 phy_init_data.bin
usage: esptool merge_bin [-h] --output OUTPUT [--format {r,a,w}] [--flash_freq {keep,80m,60m,48m,40m,30m,26m,24m,20m,16m,15m,12m}] [--flash_mode {keep,qio,qout,dio,dout}]
                         [--flash_size {keep,256KB,512KB,1MB,2MB,2MB-c1,4MB,4MB-c1,8MB,16MB,32MB,64MB,128MB}] [--spi-connection SPI_CONNECTION]
                         [--target-offset TARGET_OFFSET] [--fill-flash-size {256KB,512KB,1MB,2MB,4MB,8MB,16MB,32MB,64MB,128MB}]
                         <address> <filename> [<address> <filename> ...]
esptool merge_bin: error: argument --flash_size/-fs: invalid choice: 'detect' (choose from 'keep', '256KB', '512KB', '1MB', '2MB', '2MB-c1', '4MB', '4MB-c1', '8MB', '16MB', '32MB', '64MB', '128MB')
```

As you can see from the log above, it no longer complains about the flash_args, but it does not like something about ```detect``` option for flash size. Is there any particular reason?

jakub-kocka, 2023-10-05:
Hi @krupis,
It looks like the `detect` option is not fully supported with other arguments in the actual release version. And also it is not needed when only using `merge_bin` and not flashing.

Could you try to edit `flash_args` file for example like that:

    --flash_mode dio --flash_freq 40m --flash_size keep
    0x1000 bootloader/bootloader.bin
    0x10000 myproject.bin
    0x8000 partition_table/partition-table.bin
    0xd000 ota_data_initial.bin
    0xf000 phy_init_data.bin

or get rid of the whole first row.

krupis, 2023-10-05:
@jakub-kocka 
Works like  charm when ```detect``` is replaced with ```keep```

Thanks.

jakub-kocka, 2023-10-05:
Glad to hear that.
You are welcome.