espressif/esptool #924: Protocol error when trying to flash ESP32-C3 Mini

ma3oun, 2023-10-09:
### Operating System

Ubuntu 22.04 LTS

### Esptool Version

4.5.1

### Python Version

Python 3.9.13

### Chip Description

ESP32-C3-MINI-1

### Device Description

I'm using a custom board, using a design similar to [this](https://hackaday.io/project/187504-esp32-3-channel-power-logger)
The ESP32 connects to the PC directly by USB (IO18 and IO19)

### Hardware Configuration

I'm using a USB-C port on the board to connect to USB 2.0 or 3.0 standard ports on my PC.
The ESP32 interacts with a INA3221 current sensor by I2C.

### How is Esptool Run

Terminal (python esptool.py ...) and latest Arduino IDE (lolin C3 board selected)

### Full Esptool Command Line that Was Run

python esptool.py -t --before usb_reset --chip esp32c3 write_flash --no-compress 0x00 ~/Arduino/BlinkWithoutDelay/build/esp32.esp32.lolin_c3_mini/BlinkWithoutDelay.ino.bin

### Esptool Output

```shell
esptool.py v4.5.1
Found 1 serial ports
Serial port /dev/ttyACM0
Connecting...
/dev/ttyACM0 failed to connect: [Errno 71] Protocol error

A fatal error occurred: Could not connect to an Espressif device on any of the 1 available serial ports.
```

### More Information

The PC seems to recognize the ESP32 when the USB cable is plugged in.
dmesg output:
`[ 8006.453801] usb 1-1: new full-speed USB device number 15 using xhci_hcd
[ 8006.627939] usb 1-1: New USB device found, idVendor=303a, idProduct=1001, bcdDevice= 1.01
[ 8006.627951] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 8006.627955] usb 1-1: Product: USB JTAG/serial debug unit
[ 8006.627958] usb 1-1: Manufacturer: Espressif
[ 8006.627961] usb 1-1: SerialNumber: 64:E8:33:E2:9F:5C
[ 8006.635779] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
`

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

ma3oun, 2023-10-09:
I have also tried the "no_reset" and "no_reset_no_sync" options without much success. I get "Failed to connect to ESP32-C3: No serial data received"

dobairoland, 2023-10-09:
Could you please also share the output of `dmesg` after attaching the device and trying to flash?

ma3oun, 2023-10-09:
I have already given the output of `dmesg` after attaching the device. When I try to flash it, `dmesg` outputs nothing more.

radimkarnis, 2023-10-09:
Hi @ma3oun,
this seems to be an issue with the drivers. Esptool fails due to the host system failing to set the RTS line. Inspecting your backtrace, you can actually see the error originates in `pySerial`, not `esptool`. I am not sure how to solve the drivers issue

> I have also tried the "no_reset" and "no_reset_no_sync" options without much success. I get "Failed to connect to ESP32-C3: No serial data received"

Using the `no_reset` option requires you to put the ESP into the download mode **manually** (by pressing the `RESET` button while holding down the `BOOT` button - if your board has these). You can read about the process [here](https://docs.espressif.com/projects/esptool/en/latest/esp32c3/advanced-topics/boot-mode-selection.html#manual-bootloader). 

This could be the workaround to your issue - place the devkit manually into the download mode and then run esptool with the `--before no_reset` option.

You can see a similar issue being discussed [here](https://esp32.com/viewtopic.php?t=10741).

ma3oun, 2023-10-09:
Thanks @radimkarnis . Luckily, I can pull-down GPIO9 on my custom board and I was able to flash the board using the `before no_reset` option, as explained in the link you provided.

radimkarnis, 2023-10-09:
> Now, I can again flash the board directly from Arduino IDE without any custom option.

In that case, the previously flashed code interfered with the USB console in pins `IO18` and `IO19`. 

I will update the documentation to mention this, I will leave this ticket open until then. Thanks for the report!

closed 2023-10-19 by espressif-bot

icaro-bezerra-taak, 2025-07-21:
> Hi [@ma3oun](https://github.com/ma3oun), this seems to be an issue with the drivers. Esptool fails due to the host system failing to set the RTS line. Inspecting your backtrace, you can actually see [...]

Thank you sir this worked perfectly