espressif/esptool #925: Wrong boot mode detected (0x12)

jakobofner, 2023-10-13:
### Operating System

UBUNTU 20.04

### Esptool Version

esp-idf-v5.1.1

### Python Version

Python 3.8.10

### Chip Description

ESP32-WROVER-E

### Device Description

Custum Hardware
[mainboard.pdf](https://github.com/espressif/esptool/files/12899252/mainboard.pdf)

### How is Esptool Run

Eclipse

### Full Esptool Command Line that Was Run

esptool.py --chip esp32 -p /dev/ttyUSB5 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x1000 bootloader/bootloader.bin 0x10000 app-template.bin 0x8000 partition_table/partition-table.bin

### Esptool Output

```shell
esptool.py --chip esp32 -p /dev/ttyUSB5 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x1000 bootloader/bootloader.bin 0x10000 app-template.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.7.dev1
Serial port /dev/ttyUSB5
Connecting......................................

A fatal error occurred: Failed to connect to ESP32: Wrong boot mode detected (0x12)! The chip needs to be in download mode.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
CMake Error at run_serial_tool.cmake:66 (message):
  
  /home/jakob/.espressif/python_env/idf5.1_py3.8_env/bin/python;;/programs/esp/esp-idf-v5.1.1/components/esptool_py/esptool/esptool.py;--chip;esp32
  failed.

FAILED: CMakeFiles/flash /home/jakob/eclipse-workspace/test/build/CMakeFiles/flash 
cd /programs/esp/esp-idf-v5.1.1/components/esptool_py && /home/jakob/.espressif/tools/cmake/3.24.0/bin/cmake -D IDF_PATH=/programs/esp/esp-idf-v5.1.1 -D "SERIAL_TOOL=/home/jakob/.espressif/python_env/idf5.1_py3.8_env/bin/python;;/programs/esp/esp-idf-v5.1.1/components/esptool_py/esptool/esptool.py;--chip;esp32" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args" -D WORKING_DIRECTORY=/home/jakob/eclipse-workspace/test/build -P /programs/esp/esp-idf-v5.1.1/components/esptool_py/run_serial_tool.cmake
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/jakob/eclipse-workspace/test/build/log/idf_py_stderr_output_46926 and /home/jakob/eclipse-workspace/test/build/log/idf_py_stdout_output_46926
```

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

dobairoland, 2023-10-14:
Hi @jakobofner. So how did the troubleshooting went according to the linked guide? Why do you think this is an esptool issue instead of your hardware issue?

jakobofner, 2023-10-14:
> Hi @jakobofner. So how did the troubleshooting went according to the linked guide? Why do you think this is an esptool issue instead of your hardware issue?

I do not know which error it is, that is the reason why I posted it here.

dobairoland, 2023-10-15:
> A fatal error occurred: Failed to connect to ESP32: Wrong boot mode detected (0x12)! The chip needs to be in download mode.

https://docs.espressif.com/projects/esptool/en/latest/esp32/troubleshooting.html#wrong-boot-mode-detected-0xxx-the-chip-needs-to-be-in-download-mode

dobairoland, 2023-10-20:
I'm sorry but we don't have the resources to follow through step-by-step every case. That is why we maintain the troubleshooting guide. If there are no more information provided then I'm closing this issue.