espressif/esptool #927: ESP32-S3 "Hash of data verified." not reported after writing flash block

frenchf1, 2023-10-24:
### Operating System

Windows 10

### Esptool Version

v4.5.1

### Python Version

3.10.10

### Chip Description

ESP32-S3

### Device Description

ESP-PROG

### How is Esptool Run

Windows Terminal

### Full Esptool Command Line that Was Run

esptool --chip esp32s3 -p COM11 -b 1000000 --before=default_reset --after=hard_reset --no-stub write_flash --flash_mode dio --flash_freq 80M --flash_size 4MB --force 0x0 bootloader.bin 0x20000 firmware.bin 0xc000  partition-table.bin 0x1d000 ota_data_initial.bin

### Esptool Output

```shell
esptool.py v4.6.2
Serial port COM11
Connecting....
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:5a:4b:a4
Changing baud rate to 1000000
Changed.
Enabling default SPI flash mode...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00008fff...
Flash will be erased from 0x00020000 to 0x00190fff...
Flash will be erased from 0x0000c000 to 0x0000cfff...
Flash will be erased from 0x0001d000 to 0x0001efff...
Erasing flash...
Took 0.34s to erase flash block
Writing at 0x00000000... (2 %)
Writing at 0x00000400... (5 %)
Writing at 0x00000800... (8 %)
Writing at 0x00000c00... (11 %)
Writing at 0x00001000... (13 %)
Writing at 0x00001400... (16 %)
Writing at 0x00001800... (19 %)
Writing at 0x00001c00... (22 %)
Writing at 0x00002000... (25 %)
Writing at 0x00002400... (27 %)
Writing at 0x00002800... (30 %)
Writing at 0x00002c00... (33 %)
Writing at 0x00003000... (36 %)
Writing at 0x00003400... (38 %)
Writing at 0x00003800... (41 %)
Writing at 0x00003c00... (44 %)
Writing at 0x00004000... (47 %)
Writing at 0x00004400... (50 %)
Writing at 0x00004800... (52 %)
Writing at 0x00004c00... (55 %)
Writing at 0x00005000... (58 %)
Writing at 0x00005400... (61 %)
Writing at 0x00005800... (63 %)
Writing at 0x00005c00... (66 %)
Writing at 0x00006000... (69 %)
Writing at 0x00006400... (72 %)
Writing at 0x00006800... (75 %)
Writing at 0x00006c00... (77 %)
Writing at 0x00007000... (80 %)
Writing at 0x00007400... (83 %)
Writing at 0x00007800... (86 %)
Writing at 0x00007c00... (88 %)
Writing at 0x00008000... (91 %)
Writing at 0x00008400... (94 %)
Writing at 0x00008800... (97 %)
Writing at 0x00008c00... (100 %)
Wrote 36864 bytes at 0x00000000 in 1.1 seconds (256.4 kbit/s)...
Hash of data verified.
Erasing flash...
Took 6.84s to erase flash block
...
Sometimes for each flash block the "Hash of data verified." text is missing ? What does it mean ? Why ?
```

### More Information

Sometimes for each flash block the "Hash of data verified." text is missing ? What does it mean ? Why ?

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2023-10-25:
Hi @frenchf1,
the `Hash of data verified.` text is displayed after every file that is written to the flash memory, if the write is not encrypted or if the ESP is not in secure download mode.

In your case, you are flashing 4 files:
- 0x0 bootloader.bin 
- 0x20000 firmware.bin 
- 0xc000 partition-table.bin 
- 0x1d000 ota_data_initial.bin

After each of these, you should see these two lines:

```
Wrote 36864 bytes at 0x00000000 in 1.1 seconds (256.4 kbit/s)...
Hash of data verified.
```

Does this not happen in your case? Your log clearly shows the lines.

frenchf1, 2023-10-25:
Hi @radimkarnis ,

In my case the line "Hash of data verified." is generated most of time but sometimes this line is missing after each writing and I have no idea about the root cause ?

Thank you for your help.

radimkarnis, 2023-10-25:
@frenchf1 please send a log where the line is missing.

frenchf1, 2023-10-25:
@radimkarnis 
[log.txt](https://github.com/espressif/esptool/files/13163234/log.txt)

radimkarnis, 2023-10-26:
@frenchf1 thanks for the log. I am not sure why these are not shown in your case. Let's analyze.

This is the code for showing the `Hash of data verified.` text:

```py
    if not encrypted and not esp.secure_download_mode:
            try:
                res = esp.flash_md5sum(address, uncsize)
                if res != calcmd5:
                    print("File  md5: %s" % calcmd5)
                    print("Flash md5: %s" % res)
                    print(
                        "MD5 of 0xFF is %s"
                        % (hashlib.md5(b"\xFF" * uncsize).hexdigest())
                    )
                    raise FatalError("MD5 of file does not match data in flash!")
                else:
                    print("Hash of data verified.")
            except NotImplementedInROMError:
                pass
```

What this means is that it is shown only if the write is not encrypted, the esp is not in secure download mode and the `esp.flash_md5sum` function does not result in `NotImplementedInROMError` for every one of the flashed files.

Does this still happen if you don't use the `--no-stub` option? Does the flashed application work? Maybe the ROM is just refusing to calculate the MD5 hash for some reason.

radimkarnis, 2023-11-09:
I will be closing this issue.

Please feel free to open another one if the flashing process fails (or if you suspect flashing to fail silently and the app doesn't work).

closed 2023-11-09 by radimkarnis