espressif/esptool #928: ESP32-C6 documentation is unclear

baylf2000, 2023-10-25:
### Operating System

Any

### Esptool Version

Any

### Python Version

Any

### Chip Description

ESP32-C6

### Device Description

Any

### Hardware Configuration

Any

### How is Esptool Run

Any

### Full Esptool Command Line that Was Run

Any

### Esptool Output

```shell
Any
```

### More Information

The documentation page for esptool for the ESP32-C6 is contradictory. It states that GPIO8 must be unconnected or low AND must be high in order to enter the serial boot loader. It cannot be both at the same time.

```GPIO8[](https://docs.espressif.com/projects/esptool/en/latest/esp32c6/advanced-topics/boot-mode-selection.html#gpio8)

GPIO8 must also be either left unconnected/floating, or driven Low, in order to enter the serial bootloader.

GPIO8 must also be driven High, in order to enter the serial bootloader reliably. The strapping combination of GPIO8 = 0 and GPIO9 = 0 is invalid and will trigger unexpected behavior.

In normal boot mode (GPIO9 high), GPIO8 is ignored.```

An urgent correction/clarification would be much appreciated as this is holding up the final sign-off on our project.

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2023-10-25:
Hi @baylf2000,
please see the ground truth information for the final sign-off:

[image]

The `GPIO8 must also be either left unconnected/floating, or driven Low, in order to enter the serial bootloader.` sentence is a leftover from documentation of older chips. Sorry for that, I will correct the docs ASAP.

baylf2000, 2023-10-25:
@radimkarnis Much appreciated :) Thank you!

closed 2023-10-26 by espressif-bot

radimkarnis, 2023-10-26:
@baylf2000 thanks for noticing this!