espressif/esptool #931: ESP32-S3 blinking yellow after flashing

hydnum-repandum, 2023-11-01:
### Operating System

Kubuntu 23.04

### Esptool Version

esptool.py v4.7-dev

### Python Version

3.11.4

### Chip Description

ESP32-S3-WROOM

### Device Description

I am not expert. I have just started with my first micro-controller .   I bought the development board

"Freenove Basic Starter Kit for ESP32-S3-WROOM (Included) (Compatible with Arduino IDE)" 

to learn microphyton.   I have followed instructions to burn the firmware on the ESP S3. I didn't get any errors but the serial monitor shows that the device is rebooting continuously while blinking yellow. 

### Hardware Configuration

Nothing attached just the board connected by USB

### How is Esptool Run

Running from Konsole (kubuntu 23.04)

### Full Esptool Command Line that Was Run

python3 esptool.py --chip esp32s3 --port /dev/ttyCH343USB0 --baud 2000000 write_flash -z 0 ESP32_GENERIC_S3-20230426-v1.20.0.bin

### Esptool Output

```shell
1) Write the latest firmware "GENERIC_S3-20220618-v1.19.1.bin" to the drive
-------------------------------------------------------------------------
hydnum@fontaines:~/Freenove/Basic/Python/Python_Firmware$ python3 esptool/esptool.py --chip esp32s3 --port /dev/ttyCH343USB0 --baud 2000000 write_flash -z 0 ESP32_GENERIC_S3-20230426-v1.20.0.bin
esptool.py v4.7-dev
Serial port /dev/ttyCH343USB0
Connecting...
Failed to get PID of a device on /dev/ttyCH343USB0, using standard reset sequence.
.
Chip is ESP32-S3 (QFN56) (revision v0.1)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: 34:85:18:94:28:fc
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 2000000
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00160fff...
Compressed 1445232 bytes to 957441...
Wrote 1445232 bytes (957441 compressed) at 0x00000000 in 11.0 seconds (effective 1053.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
---------------------------------------------------------------------------------

2) Check the new flash id in the drive
--------------------------------------------------------------
hydnum@fontaines:~/Downloads/esptool-master$ python3 esptool.py -p /dev/ttyCH343USB0 flash_id
esptool.py v4.7-dev
Serial port /dev/ttyCH343USB0
Connecting...
Failed to get PID of a device on /dev/ttyCH343USB0, using standard reset sequence.
.
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (QFN56) (revision v0.1)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: 34:85:18:94:28:fc
Uploading stub...
Running stub...
Stub running...
Manufacturer: c8
Device: 4017
Detected flash size: 8MB
Flash type set in eFuse: quad (4 data lines)
Hard resetting via RTS pin...
--------------------------------------------------------------

And now the blinking yellow starts and never ends . . .
```

### More Information

the serial monitor is displaying in a loop as follows:

------------------------------------------------------------
ELF file SHA256: 1fc5db5094440301

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x420ff696
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0xf34
load:0x403c9700,len:0xc44
load:0x403cc700,len:0x2c78
entry 0x403c98f4
[0;31mE (253) psram: PSRAM ID read error: 0x00ffffff[0m
[0;31mE (254) spiram: SPI RAM enabled but initialization failed. Bailing out.[0m
[0;31mE (10) spiram: SPI RAM not initialized[0m

abort() was called at PC 0x420069e8 on core 1

Backtrace: 0x40375c5e:0x3fcee2a0 0x40383919:0x3fcee2c0 0x40389ea2:0x3fcee2e0 0x420069e8:0x3fcee350 0x4200720f:0x3fcee370

ELF file SHA256: 1fc5db5094440301

Rebooting...
...
Rebooting...

### Other Steps to Reproduce

I use my Kubuntu 23.04 for flashing.  I have not tried on any other O/S

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

dobairoland, 2023-11-01:
The flashing done by esptool went fine. So I'm sorry but we cannot do anything about this here.

The issue is with your application. So please try to check your micropython project setup. We cannot help you with this. The registers were dumped because `abort()` was called by the application. You can read about this for example here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/fatal-errors.html#register-dump-and-backtrace. This is a link to ESP-IDF documentation but I suspect that the issue is with your micropython project.

dobairoland, 2023-11-01:
>  All I have done is flash the firmware and the device is abort looping. 

> The flashing is causing the abort

The flashing was successful and the firmware got flashed. The abort is caused by running the firmware.

hydnum-repandum, 2023-11-01:
Thank you for your help Roland and sorry for taking your time.  I will try to investigate what is SPI RAM and the bootloader.   I have no idea what that is or how it works.

What would be your advice on how to proceed?  Shall I buy another microprocessor?  Which one would you recommend for a beginner?  I just want to learn micropython lol

 If you think of anything else please let me know any help is greatly appreciated.

radimkarnis, 2023-11-02:
> What would be your advice on how to proceed?

@hydnum-repandum [this repository](https://github.com/Freenove/Freenove_ESP32_S3_WROOM_Board) contains everything you need to know to start using your devkit. Open the `Start Here.pdf` file, there are step-by-step descriptions of how to use your board with Arduino or MicroPython. This should be a good place to start!

> Shall I buy another microprocessor?

Not needed. The ESP32-S3 is now widely supported in Arduino, ESP-IDF, MicrOpython and other projects from the community.

hydnum-repandum, 2023-11-03:
Hi Roland

Thank you again for your precious advice.  Yes that is the tutorial that I am following.  Mind you I didn't know they have a repository in github.  They ask me to download the tutorial from their website.  The tutorial I downloaded is not as comprehensive as the one here and as far as I can tell not the exactly the same.  

The problem with the freenove tutorial is that is mainly windows based.  For example the chapter about installing the CH343 driver I had so many problems.  There are 10 pages with screenshots for Windows and Mac there there is two line paragraph for Linux.  After days and days of failing finally I got a replay from them.  Just two words "Try this" and they pointed to a repository in github.  What can I say they don't make you feel welcome and I am NOT going to install Windows lol.

Anyway I am a bit lost a the moment not sure what I am going to do next.  Still want to learn micrOpython hehe

Have a great weekend!!

hydnum-repandum, 2023-11-03:
Sorry I assumed it was Roland . . .  Thank you Radim Karnis the reply is for you!  8)