espressif/esptool #932: [ESP32-C2] Flash size reported incorrectly (auto detect is correct)

TD-er, 2023-11-10:
### Operating System

Windows 11

### Esptool Version

v4.7-dev

### Python Version

Python 3.11.4

### Chip Description

ESP8684-MINI-1  (MAH4)

### Device Description

Espressif ESP8684-DevKitM-1

### How is Esptool Run

From PlatformIO/VS code

### Esptool Output

```shell
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-builtin, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM11
Uploading .pio\build\custom_ESP32c2_4M316k_LittleFS\ESP_Easy_mega_20231110_custom_ESP32c2_4M316k_LittleFS.bin
esptool.py v4.7-dev
Serial port COM11
Connecting....
Chip is ESP32-C2 (revision v1.0)
Features: WiFi, BLE, Embedded Flash 2MB (XMC)
Crystal is 26MHz
MAC: 08:3a:8d:41:1f:b0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
```

### More Information

In my project (ESPEasy), I also try to collect as much info from burnt fuses as possible and this C2 board is actually the first I know of that has set the fuses to indicate it has embedded flash.
So maybe esptool.py also looks at these fuses and draws incorrect conclusions from it when reporting the detected features?

<html>
<body>

Label | Value
-- | --
ESP Chip ID: | 4267952 (0x411FB0)
ESP Chip Frequency: | 120 MHz
ESP Crystal Frequency: | 26 MHz
ESP APB Frequency: | 40 MHz
ESP Chip Model: | ESP32-C2
ESP Chip Features: | Wi-Fi bgn / BLE / Emb. Flash
ESP Chip Revision: | 1.0
ESP Chip Cores: | 1
ESP Board Name: | Espressif Generic ESP32-C2 = 4M Flash, ESPEasy 1810k Code/OTA 316k FS
Flash Chip ID: | Vendor: 0xC8 Device: 0x4016 (Embedded)
Flash Chip Real Size: | 4096 kB
Flash IDE Size: | 4096 kB
Flash Chip Speed: | 40 MHz
Flash IDE Speed: | 80 MHz
Flash IDE Mode: | QIO
Flash Writes: | 0 daily / 0 boot
Sketch Size: | 1215 kB (1856 kB free)
Max. OTA Sketch Size: | 1856 kB (1900544 bytes)
Little FS Size: | 316 kB (156 kB free)
Page size: | 256
Block size: | 8192
Number of blocks: | 39

</body>
</html>

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2023-11-10:
Hello @TD-er,
thanks for the report and sorry for the inconvenience.

It seems like the ESP32-C2 class in esptool inherits the `get_chip_features` from the C3 class, which results in a wrongly reported feature `Embedded Flash 2MB`. We will fix this!

TD-er, 2023-11-10:
It is not really an inconvenience... anymore :)
I was taken a bit by surprise when first connecting the C2 module as I really didn't expect it to be a 2M module.
Anyway, thanks already for the upcoming fix :)

TD-er, 2023-11-10:
I'm still looking at the flash stuff on my C2.
As you can see in the table above, the "IDE speed" was 80 MHz and it is now apparently running at 40 MHz.
So it looks like esptool might have changed it based on its detected speed.

Is it possible the detected speed is also incorrect, when it is inheriting from the C3 class?

TD-er, 2023-11-10:
OK, the reported speed is a bug in the Arduino code:
https://github.com/espressif/arduino-esp32/blob/6b287db119dab39561572d75599f8171b942352b/cores/esp32/Esp.cpp#L410C1-L424C2
Since the flash speeds of the C2 are quite different.

closed 2023-12-13 by espressif-bot