espressif/esptool #934: KeyboardInterrupt during the write_flash process returns ERRORLEVEL 0, but ERRORLEVEL 1 or more is required.

IhorNehrutsa, 2023-11-15:
### Operating System

Windows 10

### Esptool Version

v4.6.2

### Python Version

3.12.0

### Chip Description

ESP32

### How is Esptool Run

batch file

### Full Esptool Command Line that Was Run

python -m esptool --chip %CHIP% --baud %BAUD% --port %PORT% write_flash -z %ADDRESS% %FILENAME_BIN% 

### Esptool Output

```shell
python -m esptool --chip esp32 --baud 1500000 --port  write_flash -z 0x1000 firmware.bin  )
esptool.py v4.6.2
Found 2 serial ports
Serial port COM33
Connecting....
Chip is ESP32-D0WD-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 90:38:0c:55:aa:a8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 1500000
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x001b9fff...
Compressed 1804096 bytes to 1179198...
Writing at 0x00020b42... (5 %)Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\negrutsa_ii\AppData\Local\Programs\Python\Python312\Lib\site-packages\esptool\__main__.py", line 9, in <module>
    esptool._main()
  File "C:\Users\negrutsa_ii\AppData\Local\Programs\Python\Python312\Lib\site-packages\esptool\__init__.py", line 1073, in _main
    main()
  File "C:\Users\negrutsa_ii\AppData\Local\Programs\Python\Python312\Lib\site-packages\esptool\__init__.py", line 868, in main
    operation_func(esp, args)
  File "C:\Users\negrutsa_ii\AppData\Local\Programs\Python\Python312\Lib\site-packages\esptool\cmds.py", line 595, in write_flash
    esp.flash_defl_block(block, seq, timeout=timeout)
  File "C:\Users\negrutsa_ii\AppData\Local\Programs\Python\Python312\Lib\site-packages\esptool\loader.py", line 131, in inner
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\negrutsa_ii\AppData\Local\Programs\Python\Python312\Lib\site-packages\esptool\loader.py", line 1037, in flash_defl_block
    self.check_command(
  File "C:\Users\negrutsa_ii\AppData\Local\Programs\Python\Python312\Lib\site-packages\esptool\loader.py", line 436, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\negrutsa_ii\AppData\Local\Programs\Python\Python312\Lib\site-packages\esptool\loader.py", line 395, in command
    self.write(pkt)
  File "C:\Users\negrutsa_ii\AppData\Local\Programs\Python\Python312\Lib\site-packages\esptool\loader.py", line 348, in write
    self._port.write(buf)
  File "C:\Users\negrutsa_ii\AppData\Local\Programs\Python\Python312\Lib\site-packages\serial\serialwin32.py", line 321, in write
    win32.GetOverlappedResult(self._port_handle, self._overlapped_write, ctypes.byref(n), True)
KeyboardInterrupt
^CTerminate batch job (Y/N)? n
 Write flash is success!
Press any key to continue . . .
```

### More Information

I press Ctrl-C during write_flash process and I expect ERRORLEVEL 1, but ERRORLEVEL 0 is returned

### Other Steps to Reproduce

python -m esptool --chip %CHIP% --baud %BAUD% --port %PORT% write_flash -z %ADDRESS% %FILENAME_BIN% 
@if ERRORLEVEL 1 (
@echo [91m Error: Write flash is not success! [0m
) else (
@echo [92m Write flash is success! [0m
)

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

dobairoland, 2023-11-15:
This is slightly related to https://github.com/espressif/esptool/issues/888. 

This behavior is present for a long time. The fix will change the behavior. Therefore, we plan to fix this in the next major release of esptool.

IhorNehrutsa, 2023-11-15:
I found some hook. 
When Ctrl-C is pressed during the write_flash process, the ERRORLEVEL is negative value.
So possible to use `if %ERRORLEVEL% LSS 0`
```
python -m esptool --chip %CHIP% --baud %BAUD% --port %PORT% write_flash -z %ADDRESS% %FILENAME_BIN% 
@echo ERRORLEVEL=%ERRORLEVEL%
@if ERRORLEVEL 1 (
@echo [91m Error: Write flash is not success! [0m
) else (
@if %ERRORLEVEL% LSS 0 (
@echo [91m Error: KeyboardInterrupt: Write flash interrupted! [0m
@goto EXIT
)
@echo [92m Write flash is success! [0m
)
:EXIT
```

closed 2025-01-29 by espressif-bot