espressif/esptool #937: [ESP32-C2] Are reported flash frequencies correct for ESP32-C2?

TD-er, 2023-12-05:
### Operating System

Windows 11

### Esptool Version

latest

### Python Version

Python 3.11.4

### What is the Expected Behaviour?

Recently I added [this PR](https://github.com/espressif/arduino-esp32/pull/8864) to Arduino/ESP32 repo, which uses the frequencies in esptool.py for the Arduino function `magicFlashChipSpeed()`

However on further inspection, I notice the SPI bus frequency of the ESP32-C2 is running at 40 MHz, even though I have set the flash frequency to 60 MHz in the board definition and use esptool.py to flash.

See [this issue](https://github.com/espressif/arduino-esp32/issues/8960) about my concerns regarding the APB/SPI bus frequency.

This makes me wonder if the frequencies mentioned in esptool for the ESP32-C2 are correct.

radimkarnis, 2023-12-07:
Hello @TD-er,
in your recent discoveries, you mention the actual [measured frequency is indeed 60 mHz](https://github.com/espressif/arduino-esp32/issues/8960#issuecomment-1842593632). 

Is this issue ticket still relevant? If it is, I will need to check with more developers to be sure.

TD-er, 2023-12-07:
Well from an esptool perspective, this no longer seems relevant.
But if you know something to help me find the correct way to read/detect in software the actual registers where the flash freq. can be read or used to compute the actual flash freq. then I would be very interested.

closed 2023-12-07 by TD-er