espressif/esptool #938: Invalid head of packet (0xXX): Possible serial noise or corruption.

AlanEEng, 2023-12-09:
### Operating System

windows

### Esptool Version

v4.6.2

### Python Version

Python 3.11.2

### Chip Description

ESP32-C3-MINI-1-N4

### Device Description

I am using a Custom made pcb with the connections as shown in the pictures, im trying to program the ESP32-C3-MINI-1-N4 via a FTDI232  All though no matter what i am doing i cannot get it to flash. The attachments show my hardware connections. I have tried flashing using VScode, Arduino and also in the commando prompt using esptool.py. I have run Putty to make sure my serial connection is correct and to configure my flashing settings, attached is also a screenshot of putty showing different bootloader modes, but with a normal reset booting into SPI_FAST_FLASH_BOOT and also with the EN and GPIO9 logic, here as shown in the putty terminal screenshot i am indeed entering into the download boot mode for flashing.

when trying to --trace my flashingi see the following in the terminal sometimes it spams 
TRACE +0.000 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.104 No serial data received.

and then spam a bunch of 

TRACE +0.000 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.000 Read 1 bytes: 45
TRACE +0.000 Read invalid data: 45
TRACE +0.000 Remaining data in serial buffer:

Full trace is also attached

i have tried reinstalling the drivers, i have tried another one of my PCB's, couple different USB cables, lowering the baud rate to 9600, flashing in VScode,arduino and command prompt. reinstalling esptools, different PC, checked setting in my device manager under port to make sure the USB to serial is the same baud rate, i am not using breadboard for connections only for the reset and boot circuitry with transistors. i have tried specifying the chip type when flashing.

i am powering the esp using a bench power supply with stable 3.3v and not via the FTDI own LDO

[image]
[image]

### How is Esptool Run

VScode, Arduino and Command promt

### Full Esptool Command Line that Was Run

esptool.py --chip esp32c3 -p COM6 -b 115200 write_flash --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 C:\Users\pathToBin\bootloader.bin 0x8000 C:\Users\pathToTable\table.bin

### Esptool Output

```shell
esptool.py v4.6.2
Serial port COM6
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.007 Read 1 bytes: 45
TRACE +0.000 Read invalid data: 45
TRACE +0.000 Remaining data in serial buffer:
[... 988 lines cut ...]
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.110 No serial data received.
.TRACE +0.054 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.104 No serial data received.
.TRACE +0.054 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.105 No serial data received.
```

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2023-12-11:
Hello @AlanEEng,
your module is not entering the download mode correctly:
1) You can see it is entering the spi flash boot mode first (`Build:Feb  7 2021..rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)`)
2) Then you are entering USB download mode, which is wrong (`(USB_BOOT) wait usb download`)
You need to enter the UART download mode  (`(DOWNLOAD(USB/UART0/1)) waiting for download`)

I suspect your strapping pins are wrong. Please check the [esptool ESP32-C3 Boot Mode selection page](https://docs.espressif.com/projects/esptool/en/latest/esp32c3/advanced-topics/boot-mode-selection.html): It states: 

```
GPIO8 must also be driven High, in order to enter the serial bootloader reliably. The strapping combination of GPIO8 = 0 and GPIO9 = 0 is invalid and will trigger unexpected behavior.
```

In your schematic, `GPIO8` is left floating. Try pulling it high and see if you get different results!

AlanEEng, 2023-12-11:
@radimkarnis 

Thank you so much for your in depth reply and for taking the time to understand my issue. 
After some research and testing on the ESP32-C3-DevKitM-1. I figured it would be easier to just bypass my external ftdi bridge and to use the build in serial to USB d+ & d- on my esp32c3 mini module for programming and this time also keeping the GPIO 8 pin high at the same time. This seemed to do the trick when programming the dev board via its D+ and D- pins

Once again I truly appreciate your time and expertise I hope you have a great day and will help many others like me.

Sincerely 
two electrical engineers from Denmark

radimkarnis, 2023-12-11:
@AlanEEng I am glad you guys managed to make this work! Using the internal USB-Serial/JTAG is a good call, you can make the whole thing cheaper and simpler.

Good luck with your future efforts!

closed 2023-12-11 by radimkarnis

MahadiHasantauhid, 2024-06-15:
@radimkarnis My module entering into the above mention download mode` ((DOWNLOAD(USB/UART0/1)) waiting for download)` But still I'm unable to flash. The error is similar- `Invalid head of packet (0x45): Possible serial noise or corruption.
[2024-06-04 15:56:17,242][ESP8266Loader_spi[1]][espDownloader.py][line:782][ERROR]: ESP32C3 Chip stub error esp_stub_and_set_baud.` I'm trying to flash firmware to my module. I changed to baud rate 9600 to 115200 still no luck.  the troubleshooting guide indicate it could be the USB Cable connected from PC to FTDI Serial to USB adapter. I will change it soon. Unfortunate, I cannot check the GPIO8 pin status as I'm working with custom board having this ESP32-C3-MINI module without exposing GPIO8.