espressif/esptool #939: program failed with esp-usb-bridge under Windows

wuxx, 2023-12-14:
### Operating System

Win10

### Esptool Version

esptool.py v4.7.dev3

### Python Version

Python 3.11.2

### Chip Description

ESP32-S2FH4

### Device Description

just a normal ESP32-S2 dev board

### How is Esptool Run

Windows Terminal

### Full Esptool Command Line that Was Run

esptool.py  --chip esp32s2 -p COM12 -b 115200 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x1000 build/bootloader/bootloader.bin 0x10000 build/hello_world.bin 0x8000 build/partition_table/partition-table.bin

### Esptool Output

```shell
D:\Espressif\frameworks\esp-idf-v5.1.2\examples\get-started\hello_world>esptool.py --chip esp32s2 -p COM12 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x1000 build/bootloader/bootloader.bin 0x10000 build/hello_world.bin 0x8000 build/partition_table/partition-table.bin
esptool.py v4.7.dev3
Serial port COM12
Connecting....
Chip is ESP32-S2FH4 (revision v0.0)
Features: WiFi, Embedded Flash 4MB, No Embedded PSRAM, ADC and temperature sensor calibration in BLK2 of efuse V1
Crystal is 40MHz
MAC: 84:f7:03:e7:62:82
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.

A fatal error occurred: Unable to verify flash chip connection (No serial data received.).
```

### More Information

changed baudrate failed, it still use default 115200 baudrate to communicate with the target chip
Issues related to esp-usb-bridge
https://github.com/espressif/esp-usb-bridge/issues/19

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2023-12-15:
Hello @wuxx,
thanks for the report. Could you please describe your setup a little more?
I understand that:
1) This fails on Windows, works on Linux
2) The target MCU that's being flashed is ESP32-S2F4

But what MCU is acting as the esp-usb-bridge? Is it the S2 or S3? Also, what version of esp-usb-bridge are you running? Thank you.

wuxx, 2023-12-15:
hello, @radimkarnis the esp-usb-bridge is also based on ESP32-S2FH4, I try the latest repo (commit 3ad34a7), also failure.

radimkarnis, 2023-12-18:
Thanks for the clarification, we will try to reproduce this and see if something can be done.

closed 2024-01-24 by espressif-bot