espressif/esptool #941: Cannot flash ESP8266EX in Smart Powerstrip

crapforge, 2023-12-16:
### Operating System

Fedora 39 and Windows 10

### Esptool Version

v4.7.0 (Windows) / v4.6.2 (Linux)

### Python Version

Python 3.10 (Windows) / ? Linux (device not available atm)

### Chip Description

ESP8266EX

### Device Description

Yagala ‎YFTIT037 Smart Powerstrip, identical to SWB2 3AC + 2USB Power Strip as described on https://templates.blakadder.com/SWB2.html

### Hardware Configuration

Relays specific to function as smart powerstrip

### How is Esptool Run

Windows terminal / Bash on Linux

### Full Esptool Command Line that Was Run

python -m esptool -b 9600 --port COM14 erase_flash AND python -m esptool --port COM14 -b 9600 write_flash -fm dout 0x0 tasmota-DE.bin

### Esptool Output

```shell
C:\Users\Benjamin\Desktop>python -m esptool --port COM14 -b 9600 write_flash -fm dout 0x0 tasmota-DE.bin
esptool.py v4.7.0
Serial port COM14
Connecting...
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: d8:bf:c0:ee:f0:c5
Stub is already running. No upload is necessary.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x0009ffff...
Compressed 653856 bytes to 467105...
Writing at 0x0000b839... (10 %)Traceback (most recent call last):
  File "C:\Users\Benjamin\AppData\Roaming\Python\Python310\site-packages\esptool\__init__.py", line 1139, in _main
    main()
  File "C:\Users\Benjamin\AppData\Roaming\Python\Python310\site-packages\esptool\__init__.py", line 923, in main
    operation_func(esp, args)
  File "C:\Users\Benjamin\AppData\Roaming\Python\Python310\site-packages\esptool\cmds.py", line 598, in write_flash
    esp.flash_defl_block(block, seq, timeout=timeout)
  File "C:\Users\Benjamin\AppData\Roaming\Python\Python310\site-packages\esptool\loader.py", line 131, in inner
    return func(*args, **kwargs)
  File "C:\Users\Benjamin\AppData\Roaming\Python\Python310\site-packages\esptool\loader.py", line 1077, in flash_defl_block
    self.check_command(
  File "C:\Users\Benjamin\AppData\Roaming\Python\Python310\site-packages\esptool\loader.py", line 467, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "C:\Users\Benjamin\AppData\Roaming\Python\Python310\site-packages\esptool\loader.py", line 436, in command
    p = self.read()
  File "C:\Users\Benjamin\AppData\Roaming\Python\Python310\site-packages\esptool\loader.py", line 369, in read
    return next(self._slip_reader)
StopIteration

A fatal error occurred: The chip stopped responding.

OR

python -m esptool --port COM14 -b 9600 write_flash --no-compress -fm dout 0x0 tasmota-DE.bin
esptool.py v4.7.0
Serial port COM14
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: d8:bf:c0:ee:f0:c5
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x0009ffff...
Writing at 0x00004000... (5 %)Traceback (most recent call last):
  File "C:\Users\Benjamin\AppData\Roaming\Python\Python310\site-packages\esptool\__init__.py", line 1139, in _main
    main()
  File "C:\Users\Benjamin\AppData\Roaming\Python\Python310\site-packages\esptool\__init__.py", line 923, in main
    operation_func(esp, args)
  File "C:\Users\Benjamin\AppData\Roaming\Python\Python310\site-packages\esptool\cmds.py", line 609, in write_flash
    esp.flash_block(block, seq)
  File "C:\Users\Benjamin\AppData\Roaming\Python\Python310\site-packages\esptool\loader.py", line 872, in flash_block
    self.check_command(
  File "C:\Users\Benjamin\AppData\Roaming\Python\Python310\site-packages\esptool\loader.py", line 467, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "C:\Users\Benjamin\AppData\Roaming\Python\Python310\site-packages\esptool\loader.py", line 436, in command
    p = self.read()
  File "C:\Users\Benjamin\AppData\Roaming\Python\Python310\site-packages\esptool\loader.py", line 369, in read
    return next(self._slip_reader)
StopIteration

A fatal error occurred: The chip stopped responding.

Corresponding output on Linux is similar:

sudo esptool --port /dev/ttyUSB3 write_flash --no-compress -fm dout 0x0 Downloads/tasmota-DE.bin 
esptool.py v4.6.2
Serial port /dev/ttyUSB3
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: d8:bf:c0:ee:f0:c5
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x0009ffff...
Writing at 0x00004000... (5 %)Traceback (most recent call last):
  File "/usr/lib/python3.12/site-packages/esptool/__init__.py", line 1073, in _main
    main()
  File "/usr/lib/python3.12/site-packages/esptool/__init__.py", line 868, in main
    operation_func(esp, args)
  File "/usr/lib/python3.12/site-packages/esptool/cmds.py", line 606, in write_flash
    esp.flash_block(block, seq)
  File "/usr/lib/python3.12/site-packages/esptool/loader.py", line 832, in flash_block
    self.check_command(
  File "/usr/lib/python3.12/site-packages/esptool/loader.py", line 436, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/esptool/loader.py", line 405, in command
    p = self.read()
        ^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/esptool/loader.py", line 338, in read
    return next(self._slip_reader)
           ^^^^^^^^^^^^^^^^^^^^^^^
StopIteration

A fatal error occurred: The chip stopped responding.
```

### More Information

I first tried using tasmotizer 1.2 which reproducibly stopped at 10% while flashing showing a "Timed out waiting for packet header" error message. Erasing flash stopped consistently at 50% and timed out after a while. 

After several attempts I switched over to using `esptool` directly to have more control over baud rate and be probably able to narrow down the problem to transfer speed issues or something like this, but to no avail. 

Then I found https://github.com/espressif/esptool/issues/890 and tried flashing with the `--no-compress` option, but unfortunately this did not help either.

My `esptool.cfg` with the following settings did not improve anything:

```
[esptool]
timeout = 30
max_timeout = 240
erase_write_timeout_per_mb = 40
mem_end_rom_timeout = 0.2
serial_write_timeout = 10
```

`flash_id` shows: 

```
C:\Users\Benjamin\Desktop>python -m esptool --port COM14 -b 9600 flash_id
esptool.py v4.7.0
Serial port COM14
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: d8:bf:c0:ee:f0:c5
Uploading stub...
Running stub...
Stub running...
Manufacturer: 68
Device: 4014
Detected flash size: 1MB
Hard resetting via RTS pin...
```

After that I tried flashing an ESP8266 on a NodeMCU v3 by connecting my CH340G serial-to-usb adapter directly to the RX/TX pins, which worked absolutely perfect, so I am not thinking that my adapter is to blame. This working chip is a 

```
Manufacturer: 68
Device: 4016
Detected flash size: 4MB
```

I connected my CH430G adapter directly to an usb port of my computer to rule out any cable or hub problems, the problem persisted.

Can this be an issue with the flash chip itself or a bug in `esptool`?

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

Jason2866, 2023-12-16:
There is probably something connected to the rx / tx gpios disturbing the flash process.
Disconnect anything from this GPIOs.

crapforge, 2023-12-16:
I'm afraid that's impossible as it would mean desoldering the module carrying the ESP which is in immediate vincinity of the transformer that is powering the ESP and (presumably) the USB ports. This is a device which is carrying mains voltage in large parts of it and tinkering around more than trying to flash a new firmware is a no-go for me. 

At the moment I'm pinning my hopes on a different serial-to-usb adapter which I have just ordered and that should arrive on Monday. I'll wait and see.

radimkarnis, 2023-12-18:
Hi @crapforge,
thanks for doing your research and poking around a lot before opening an issue, that's appreciated!

I am not sure about what's causing the failures, but they seem to happen whenever the ESP starts writing to the flash chip. Maybe it is write-protected (or just a region). You can try inspecting and setting the flash register bits with the [read_flash_status and write_flash_status](https://docs.espressif.com/projects/esptool/en/latest/esp8266/esptool/advanced-commands.html#read-flash-chip-registers-read-flash-status) commands.

Please let us know if this or a new USB-to-UART bridge helps!

crapforge, 2023-12-18:
Using a CP2102-based serial-to-usb adapter I was finally able to flash Tasmota successfully.

Thanks to all contributors for their suggestions!

radimkarnis, 2023-12-19:
I am happy you could resolve the issue. Thanks for your report!

closed 2023-12-19 by radimkarnis