espressif/esptool #944: esptool-v4.7.0-win64.zip is detected as Trojan:AndroidOS/ZkarletFlash by Windows Defender

rlowens, 2024-01-02:
### Operating System

Windows 10

### Esptool Version

v4.7.0

### Python Version

N/A

### Esptool Output

```shell
Windows Defender (Security intelligence version: 1.403.1516.0 Version created on: 1/1/2024 9:21 PM) blocks access to esptool-v4.7.0-win64.zip as Trojan:AndroidOS/ZkarletFlash.

I've submitted the file for review at https://www.microsoft.com/en-us/wdsi/submission/67f5ca74-b772-4821-9730-ec0226f3d84c

If I manually Allow the detection in Windows Defender I can unzip and the resulting files are not detected as infected, only the zip file is.
```

### What is the Expected Behaviour?

File should download and be usable without Windows Defender blocking it.

### More Information

Detection by Windows Defender:
[image]

Current online Windows Defender false-detection case result:
[image]

I've also scanned esptool-v4.7.0-win64.zip with VirusTotal.com and it found 16 detections https://www.virustotal.com/gui/file/2bf76ac51f537f5e409fdc38e502f0ca4217db832af4572a80f97f2c6e1cb63e

[image]

Running the same Windows Defender and VirusTotal.com scans on esptool-v4.6.2-win64.zip result in no detections.

Jason2866, 2024-01-02:
False positive virus alert. Caused from the used PyInstaller.

radimkarnis, 2024-01-03:
Hello @rlowens,
thanks for the report and for submitting the file for review. This is indeed a false positive.

We sign the releases on Windows to prevent these false antivirus detections from happening, but that might not be enough.
I will also send the files for review, hopefully this will be whitelisted soon. 

I don't think there's more that can be done at the moment. I'll keep this open until the Microsoft ticket is resolved.

closed 2024-02-06 by espressif-bot