espressif/esptool #945: ESP32-Wrover-E can't flash enter download mode

joe-fang-iD, 2024-01-09:
### Operating System

Windows 10

### Esptool Version

3.2-dev

### Python Version

3.8.7

### Chip Description

ESP32-WROVER-E-N8R8

### Device Description

ESP32-WROVER-E on custom PCB being programmed by ESP-PROG, plugged into Windows PC.

### Hardware Configuration

IO0 directly connected to boot on ESP-PROG.
EN pulled up by 25KOhm and connected to EN on ESP-PROG
IO2 floating
IO5 floating
IO15 floating

### How is Esptool Run

ESP-IDF-4.4

### Full Esptool Command Line that Was Run

idf.py flash monitor -p COM14

### Esptool Output

```shell
T:\support\esp-idf-v4.4\esp-idf-v4.4\examples\get-started\hello_world>idf.py flash monitor -p COM14
Executing action: flash
Running ninja in directory t:\support\esp-idf-v4.4\esp-idf-v4.4\examples\get-started\hello_world\build
Executing "ninja flash"...
[1/5] cmd.exe /C "cd /D T:\support\esp-idf-v4.4\esp-idf-v4.4\examples\g....4/esp-idf-v4.4/examples/get-started/hello_world/build/hello_world.bin"
hello_world.bin binary size 0x29730 bytes. Smallest app partition is 0x100000 bytes. 0xd68d0 bytes (84%) free.
[2/5] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D T:\support\esp-idf-v4.4\esp-idf-v4.4\examples\get-started\hello_world\build\bootloader\esp-idf\esptool_py && C:\Users\joe.fang\Desktop\esp-idf-44\python_env\idf4.4_py3.8_env\Scripts\python.exe C:/Espressif/frameworks/esp-idf-v4.4/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 T:/support/esp-idf-v4.4/esp-idf-v4.4/examples/get-started/hello_world/build/bootloader/bootloader.bin"
Bootloader binary size 0x62b0 bytes. 0xd50 bytes (12%) free.
[2/3] cmd.exe /C "cd /D C:\Espressif\frameworks\esp-idf-v4.4\components...if/frameworks/esp-idf-v4.4/components/esptool_py/run_serial_tool.cmake"
esptool.py esp32 -p COM14 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x1000 bootloader/bootloader.bin 0x10000 hello_world.bin 0x8000 partition_table/partition-table.bin
esptool.py v3.2-dev
Serial port COM14
Connecting......................................

A fatal error occurred: Failed to connect to ESP32: Invalid head of packet (0x65): Possible serial noise or corruption.
For troubleshooting steps visit: https://github.com/espressif/esptool#troubleshooting
CMake Error at run_serial_tool.cmake:56 (message):

  C:/Users/joe.fang/Desktop/esp-idf-44/python_env/idf4.4_py3.8_env/Scripts/python.exe
  C:/Espressif/frameworks/esp-idf-v4.4/components/esptool_py/esptool/esptool.py
  --chip esp32 failed

Results from Monitor:

rst:0x10 (RTCWDT_RTC_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fffeba4,len:4
load:0x4009f000,len:3248
entry 0x4009f574
�OHAI�ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fffeba4,len:4
load:0x4009f000,len:3248
entry 0x4009f574
�OHAI�ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fffeba4,len:4
load:0x4009f000,len:3248
entry 0x4009f574
�OHAI�
```

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2024-01-09:
Hello @joe-fang-iD,

- could you please run esptool with the `--trace` option and post the resulting log?
- does your ESP-PROG have the IO0 jumper installed? (It is possible to deactivate the the IO0 line with a jumper)
- could you please try with a more recent esptool version, just so we can be sure this is not caused by old software?

Thank you!

dobairoland, 2024-01-09:
Related https://esp32.com/viewtopic.php?f=2&t=37692

joe-fang-iD, 2024-01-10:
Weird. I ran it again it today, and it worked. I would LOVE to hear if there's any suggestions of how this happened, or recommendations of how to prevent it from happening. 

Trace output attached.

[2401091637progtracelog.txt](https://github.com/espressif/esptool/files/13881557/2401091637progtracelog.txt)

My ESP-PROG does have a IO0 ON/OFF jumper. Should I leave the jumper in there, or pull it out?

I'll hold off trying again with the latest esptool until this issue happens again, although is there anyway to update ESP-IDF v4.4 with later versions of esptool?

joe-fang-iD, 2024-01-11:
Happened again. See attached trace logs.
[01111120tracelogs.txt](https://github.com/espressif/esptool/files/13907476/01111120tracelogs.txt)

radimkarnis, 2024-01-17:
> is there anyway to update ESP-IDF v4.4 with later versions of esptool?

ESP-IDF v4.4 uses esptool v3, it can't be updated to esptool v4 because of breaking changes. But it is possible to install the latest esptool outside of the IDF environment (perhaps in a virtual environment) and use that directly.

Your reset sequence issues might be caused by the drivers on Windows 10, we had reports of this happening on W10, unfortunately, it can't be fixed in esptool.

joe-fang-iD, 2024-01-17:
I really appreciate you guys looking into it, and I think I learnt a bit about esptool.

It looks like the problem is actually a defect in the esp-prog cable. I  was able to replicate it and fix it temporarily by bending the cable. 

Closing this issue.

closed 2024-01-17 by joe-fang-iD

radimkarnis, 2024-01-17:
Wow, it's often the most unexpected things... I am glad you resolved your issue!