espressif/esptool #946: Use PySerial locking so the serial port isn't used by other programs

redfast00, 2024-01-11:
### Is your feature request related to a problem?

I'm always frustrated when I try to flash a chip using esptool, and I accidentally left a serial console open somewhere else, causing esptool to behave weirdly (since multiple programs are using the same serial port,  some of the data on the serial port ends up in esptool and the other part end up in the serial console).

### Describe the solution you'd like

This can be avoided by advisory locking the serial port; in pyserial, this can be done with https://github.com/pyserial/pyserial/pull/206. A lot of programs already do this, but since this is **advisory** locking, it only helps if all programs you're using check the lock. See also https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=734086

A one-line fix would be to add under this line https://github.com/espressif/esptool/blob/master/esptool/loader.py#L303 `self._port.exclusive = True`. If you want to do something fancier, you can also catch the error, and maybe even display the program that is using it. For me, just seeing 'this port is already in use by another program' would already be useful.

### Additional context

This is on Linux; I don't think locking in pyserial is implemented in Windows, so it should not cause any issues there.

dobairoland, 2024-01-18:
Thank you for letting us know about this.

closed 2024-01-19 by espressif-bot

redfast00, 2024-01-19:
Thanks for implementing it :)