espressif/esptool #948: Flashing secure boot bootloader with --no-stub will fail, causing invalid header

law-ko, 2024-01-20:
### Operating System

macOS

### Esptool Version

esptool.py v4.7.0

### Python Version

Python 3.9.7

### Chip Description

ESP32-PICO-MINI-02

### Device Description

Custom PCB with only EN pin.

### How is Esptool Run

macOS

### Esptool Output

```shell
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0x50203a73
invalid header: 0x50203a73
invalid header: 0x50203a73
invalid header: 0x50203a73
invalid header: 0x50203a73
invalid header: 0x50203a73
invalid header: 0x50203a73
invalid header: 0x50203a73
```
```

### More Information

<img width="743" alt="Screenshot 2024-01-20 at 11 33 00 AM" src="https://github.com/espressif/esptool/assets/105698389/086ebf97-5b8c-4852-8cbe-923c7800933a">

### Other Steps to Reproduce

1. `idf.py bootloader`
2. `esptool.py --chip esp32 -p /dev/ttyUSB0 -b 460800 --before default_reset --after
no_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 8MB
0x0000 build/bootloader/bootloader.bin` (`--no-stub` option is removed to flash correctly or else would fail)
3. `idf.py build`
4. `idf.py -p /dev/ttyUSB0 flash monitor`

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

dobairoland, 2024-01-22:
Hi @law-ko! Yes, it is possible to damage your chip by running random esptool commands, especially enabling security features and burning eFuses.

Please follow the guide at https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/secure-boot-v2.html in order to avoid issue. We don't support steps different to those presented in the official guide.

radimkarnis, 2024-01-22:
Hi @law-ko,
I can't reproduce this issue, can you please provide more info:
1) Is the failure visible during flashing (e.g. esptool.py failing to verify MD5 of the flashed file), or only after you start monitoring the app? (`invalid header: 0x50203a73` is the only sign of the failure?)
2) You manually remove the `--no-stub` flag. Is your serial flasher configured in menuconfig to not use the flasher stub?
3) Does the same issue happen if the bootloader is not securely signed?
4) What if you skip the second step? What if bootloader gets flashed during the fourth command? (`idf.py -p /dev/ttyUSB0 flash monitor`)

Thank you.

radimkarnis, 2024-02-19:
I believe this is a result of a wrong bootloader offset. Closing this issue due to inactivity.

closed 2024-02-19 by radimkarnis