espressif/esptool #952: Error code 01640000 is not recognized by esptool.

ciniml, 2024-01-29:
### Operating System

Windows

### Esptool Version

esptool.exe v4.5.1 (prebuild executable)

### Python Version

No python

### Chip Description

ESP32-S3

### Device Description

Custom hardware with ESP32-S3-WROOM-1-N8R8

### How is Esptool Run

Running esptool.exe from Custom GUI tool

### Full Esptool Command Line that Was Run

esptool.exe --chip esp32s3 --port COM4 --baud 2000000 --before default_reset --after no_reset --no-stub write_flash --flash_mode dio --flash_freq 40m --flash_size 8MB --force --encrypt 0x0000 bootloader.bin 0x9000 partitions.bin 0xf000 phy_init_data.bin 0x10000 ota_data_initial.bin 0x230000 factory.bin 0x320000 main.bin 0x590000 test-app.bin 0xd000 partdata_0xd000.bin 0x16000 nvskey.bin

### Esptool Output

```shell
esptool.py v4.5.1
Serial port COM4
Connecting...
2024/1/29 上午9:56:33: stdout: 
Chip is ESP32-S3 in Secure Download Mode
Changing baud rate to 2000000
Changed.
Enabling default SPI flash mode...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00008fff...
Flash will be erased from 0x00009000 to 0x00009fff...
Flash will be erased from 0x0000f000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00011fff...
Flash will be erased from 0x00230000 to 0x00300fff...
Flash will be erased from 0x00320000 to 0x00480fff...
Flash will be erased from 0x00590000 to 0x00670fff...
Flash will be erased from 0x0000d000 to 0x0000efff...
Flash will be erased from 0x00016000 to 0x00016fff...
Erasing flash...
Warning: Image file at 0x0 is protected with a hash checksum, so not changing the flash frequency setting. Use the --flash_frequency=keep option instead of --flash_frequency=40m in order to remove this warning, or use the --dont-append-digest option for the elf2image command in order to generate an image file without a hash checksum
A fatal error occurred: Failed to enter Flash download mode (result was 01640000: Unknown result)
```

### More Information

The main problems is that the error code `01640000` is returned by ESP32-S3. 
It seeems that actual error code is 0x0164, which is not listed in `err_defs` variable defined in `util.py`

https://github.com/espressif/esptool/blob/e9489931f10d27d487bebaed3b2500674b4f444d/esptool/util.py#L132

The code 0x0164 seems that an error from ROM downloader, because of its 0x01 prefix.
The device is in Secure Download mode so I assume that the error is related to Secure Download mode.

So, what I want to know is the meaning of error code 01640000. 
It's very helpful if these error codes are listed in `err_defs` and esptool.py shows appropriate error message.

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

ciniml, 2024-01-30:
I've confirmed that the error code `01640000` is returned when the target device is in **Secure download mode (Release)** and `--encrypt` option is specified, which is invalid with Release mode.

I think the correct description message for error code `01640000` should be provided by Espressif and `err_defs` should be updated.

radimkarnis, 2024-01-30:
Hi @ciniml,
thank you for your report, I agree having error definitions in esptool is useful.

I searched through the ROM code and there is no mention of any `164` error code. I found some missing error definitions for flash chip operations, but these are all in a `0x600N` range (where `N` is `1`-`6`). I can add these, but it is not a solution to your specific error.

I wonder where this comes from, can you please send the output of esptool when using the `--trace` flag? Maybe it's just some stray data being interpreted as an error code.

ciniml, 2024-01-30:
Hi @radimkarnis 
Thank you for replying.

Here is a trace log when I reproduce the problem with a device in Release mode. 
The environment is not the same as the initial report, but I've confirmed that the result is the same with this environment.

The current environment is Ubuntu 22.04, esptool.py v4.6.2 and Python 3.10.12.

```
$ dd if=/dev/zero of=dummy.bin bs=1024 count=4
$ esptool.py --port /dev/ttyACM0 --baud 2000000 --chip esp32s3 --trace  write_flash --flash_size 8MB --encrypt 0x0 dummy.bin
```

Below is the trace log from esptool.py output. 
I don't know about the packet format, but the last packet  `010204000000000001640000c0` seems to contain the error code.

```
esptool.py v4.6.2
Serial port /dev/ttyACM0
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.001 Read 1 bytes: c0
TRACE +0.000 Read 27 bytes: 
    0108040007071220 00000000c0c00108 | ....... ........
    0400070712200000 0000c0           | ..... .....
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010804000707122000000000c0
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Read 14 bytes: c0010804000707122000000000c0
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010804000707122000000000c0
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Read 14 bytes: c0010804000707122000000000c0
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010804000707122000000000c0
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Read 14 bytes: c0010804000707122000000000c0
TRACE +0.000 Received full packet: 010804000707122000000000

TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=00100040
TRACE +0.000 Write 14 bytes: c0000a04000000000000100040c0
TRACE +0.000 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010004000000000001050000c0
TRACE +0.000 Received full packet: 010004000000000001050000
TRACE +0.000 command op=0x14 data len=0 wait_response=1 timeout=3.000 data=
TRACE +0.000 Write 10 bytes: c00014000000000000c0
TRACE +0.001 Read 1 bytes: c0
TRACE +0.000 Read 33 bytes: 
    0114180000000000 c50600000709040a | ................
    0b07070c09000000 0000000000050000 | ................
    c0                                | .
TRACE +0.000 Received full packet: 
    0114180000000000 c50600000709040a | ................
    0b07070c09000000 0000000000050000 | ................
Chip is ESP32-S3 in Secure Download Mode
WARNING: Stub loader is not supported in Secure Download Mode, setting --no-stub
Changing baud rate to 2000000
TRACE +0.000 command op=0x0f data len=8 wait_response=1 timeout=3.000 data=80841e0000000000
TRACE +0.000 Write 18 bytes: 
    c0000f0800000000 0080841e00000000 | ................
    00c0                              | ..
TRACE +0.000 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010f04000000000000000000c0
TRACE +0.000 Received full packet: 010f04000000000000000000
Changed.
Enabling default SPI flash mode...
TRACE +0.050 command op=0x0d data len=8 wait_response=1 timeout=3.000 data=0000000000000000
TRACE +0.000 Write 18 bytes: 
    c0000d0800000000 0000000000000000 | ................
    00c0                              | ..
TRACE +0.002 Read 1 bytes: c0
TRACE +0.000 Read 15 bytes: c0c0010d04000000000000000000c0
TRACE +0.000 Received full packet: 
TRACE +0.000 Received full packet: 010d04000000000000000000
Configuring flash size...
TRACE +0.000 command op=0x0b data len=24 wait_response=1 timeout=3.000 data=
    0000000000008000 0000010000100000 | ................
    00010000ffff0000                  | ........
TRACE +0.000 Write 34 bytes: 
    c0000b1800000000 0000000000000080 | ................
    0000000100001000 0000010000ffff00 | ................
    00c0                              | ..
TRACE +0.001 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010b04000000000000000000c0
TRACE +0.000 Received full packet: 010b04000000000000000000
Flash will be erased from 0x00000000 to 0x00000fff...
Erasing flash...
Warning: Image file at 0x0 doesn't look like an image file, so not changing any flash settings.
TRACE +0.000 command op=0x02 data len=20 wait_response=1 timeout=3.000 data=
    0010000004000000 0004000000000000 | ................
    01000000                          | ....
TRACE +0.000 Write 30 bytes: 
    c000021400000000 0000100000040000 | ................
    0000040000000000 0001000000c0     | ..............
TRACE +0.001 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010204000000000001640000c0
TRACE +0.000 Received full packet: 010204000000000001640000

A fatal error occurred: Failed to enter Flash download mode (result was 01640000: Unknown result)
```

radimkarnis, 2024-01-30:
Thank you for the log, unfortunately, this didn't provide any more insight.

> I don't know about the packet format, but the last packet 010204000000000001640000c0 seems to contain the error code.

The packet format is described [here](https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/serial-protocol.html#response-packet):

`010204000000000001640000c0`:
`01` - packet direction 
`02` - command op (this response is a result of the 0x02 = `FLASH_BEGIN` command (`entering Flash download mode`)) 
`0400` - the size of the data (expecting 4 bytes of status response)
`00000000` - value (no value here)
`01640000` - response data (the expected 4 bytes of status response are here)
`C0` - slip packet end limiter

I don't know where this data is coming from, I found no mention of it or any branch that could lead to such data. So, in a way, this really is an `Unknown result`.

I will at least update the mentioned flash chip error definitions.

> Running esptool.exe from Custom GUI tool

Does this also happen if you run the latest esptool from this repository (or pip) on a command line (not GUI)?

ciniml, 2024-01-30:
> Does this also happen if you run the latest esptool from this repository (or pip) on a command line (not GUI)?

The previous trace log is from esptool v4.6.2 running on Ubuntu and invoked from command line by hand.

Here is another trace log from esptool.py in the current HEAD (96a5c21007e28d9949e12c82fe913eebfd52f36d) of master branch.

```
git clone https://github.com/espressif/esptool.git
cd esptool
dd if=/dev/zero of=dummy.bin bs=1024 count=4
./esptool.py --port /dev/ttyACM0 --baud 2000000 --chip esp32s3 --trace  write_flash --flash_size 8MB --encrypt 0x0 dummy.bin
``` 

```
esptool.py v4.7.0
Serial port /dev/ttyACM0
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.002 Read 1 bytes: c0
TRACE +0.000 Read 41 bytes: 
    0108040007071220 00000000c0c00108 | ....... ........
    0400070712200000 0000c0c001080400 | ..... ..........
    0707122000000000 c0               | ... .....
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Read 56 bytes: 
    c001080400070712 2000000000c0c001 | ........ .......
    0804000707122000 000000c0c0010804 | ...... .........
    0007071220000000 00c0c00108040007 | .... ...........
    07122000000000c0                  | .. .....
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Read 14 bytes: c0010804000707122000000000c0
TRACE +0.000 Received full packet: 010804000707122000000000

TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=00100040
TRACE +0.000 Write 14 bytes: c0000a04000000000000100040c0
TRACE +0.000 Read 14 bytes: c0010004000000000001050000c0
TRACE +0.000 Received full packet: 010004000000000001050000
TRACE +0.000 command op=0x14 data len=0 wait_response=1 timeout=3.000 data=
TRACE +0.000 Write 10 bytes: c00014000000000000c0
TRACE +0.001 Read 1 bytes: c0
TRACE +0.000 Read 33 bytes: 
    0114180000000000 c50600000709040a | ................
    0b07070c09000000 0000000000050000 | ................
    c0                                | .
TRACE +0.000 Received full packet: 
    0114180000000000 c50600000709040a | ................
    0b07070c09000000 0000000000050000 | ................
Chip is ESP32-S3 in Secure Download Mode
WARNING: Stub loader is not supported in Secure Download Mode, setting --no-stub
Changing baud rate to 2000000
TRACE +0.000 command op=0x0f data len=8 wait_response=1 timeout=3.000 data=80841e0000000000
TRACE +0.000 Write 18 bytes: 
    c0000f0800000000 0080841e00000000 | ................
    00c0                              | ..
TRACE +0.001 Read 14 bytes: c0010f04000000000000000000c0
TRACE +0.000 Received full packet: 010f04000000000000000000
Changed.
Enabling default SPI flash mode...
TRACE +0.051 command op=0x0d data len=8 wait_response=1 timeout=3.000 data=0000000000000000
TRACE +0.000 Write 18 bytes: 
    c0000d0800000000 0000000000000000 | ................
    00c0                              | ..
TRACE +0.002 Read 1 bytes: c0
TRACE +0.000 Read 15 bytes: c0c0010d04000000000000000000c0
TRACE +0.000 Received full packet: 
TRACE +0.000 Received full packet: 010d04000000000000000000
Configuring flash size...
TRACE +0.000 command op=0x0b data len=24 wait_response=1 timeout=3.000 data=
    0000000000008000 0000010000100000 | ................
    00010000ffff0000                  | ........
TRACE +0.000 Write 34 bytes: 
    c0000b1800000000 0000000000000080 | ................
    0000000100001000 0000010000ffff00 | ................
    00c0                              | ..
TRACE +0.001 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010b04000000000000000000c0
TRACE +0.000 Received full packet: 010b04000000000000000000
Flash will be erased from 0x00000000 to 0x00000fff...
Erasing flash...
Warning: Image file at 0x0 doesn't look like an image file, so not changing any flash settings.
TRACE +0.001 command op=0x02 data len=20 wait_response=1 timeout=3.000 data=
    0010000004000000 0004000000000000 | ................
    01000000                          | ....
TRACE +0.000 Write 30 bytes: 
    c000021400000000 0000100000040000 | ................
    0000040000000000 0001000000c0     | ..............
TRACE +0.001 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010204000000000001640000c0
TRACE +0.000 Received full packet: 010204000000000001640000

A fatal error occurred: Failed to enter Flash download mode (result was 01640000: Unknown result)
```

radimkarnis, 2024-01-30:
I see you are using the USB-Serial/JTAG port of the ESP32-S3, can you please try flashing with a regular USB-to-UART bridge (if your device has one/you have an external one)?

I really don't know what else can be done here. Esptool copies the error definitions from the ROM. This error def doesn't exist and I can't trace it.

ciniml, 2024-01-30:
Here is the trace log via external USB-UART (CP2102) from the same ESP32-S3 device.  (I'm currently using ESP32-S3-DevKitC-1-N8R8 for debugging.)

The command line is the same except the `--port` option to specify the external USB-UART device. 
And the device returns the same result.

```
./esptool.py --port /dev/ttyUSB0 --baud 2000000 --chip esp32s3 --trace  write_flash --flash_size 8MB --encrypt 0x0 dummy.bin
```

```
esptool.py v4.7.0
Serial port /dev/ttyUSB0
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.101 No serial data received.
.TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.012 Read 1 bytes: c0
TRACE +0.000 Read 63 bytes: 
    0108040007071220 00000000c0c00108 | ....... ........
    0400070712200000 0000c0c001080400 | ..... ..........
    0707122000000000 c0c0010804000707 | ... ............
    122000000000c0c0 01080400070712   | . .............
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.005 Read 1 bytes: 20
TRACE +0.000 Read 47 bytes: 
    00000000c0c00108 0400070712200000 | ............. ..
    0000c0c001080400 0707122000000000 | ........... ....
    c0c0010804000707 122000000000c0   | ......... .....
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Received full packet: 010804000707122000000000

TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=00100040
TRACE +0.000 Write 14 bytes: c0000a04000000000000100040c0
TRACE +0.005 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010004000000000001050000c0
TRACE +0.000 Received full packet: 010004000000000001050000
TRACE +0.000 command op=0x14 data len=0 wait_response=1 timeout=3.000 data=
TRACE +0.000 Write 10 bytes: c00014000000000000c0
TRACE +0.007 Read 1 bytes: c0
TRACE +0.000 Read 33 bytes: 
    0114180000000000 c50600000709040a | ................
    0b07070c09000000 0000000000050000 | ................
    c0                                | .
TRACE +0.000 Received full packet: 
    0114180000000000 c50600000709040a | ................
    0b07070c09000000 0000000000050000 | ................
Chip is ESP32-S3 in Secure Download Mode
WARNING: Stub loader is not supported in Secure Download Mode, setting --no-stub
Changing baud rate to 2000000
TRACE +0.000 command op=0x0f data len=8 wait_response=1 timeout=3.000 data=80841e0000000000
TRACE +0.000 Write 18 bytes: 
    c0000f0800000000 0080841e00000000 | ................
[... 31 lines cut ...]
    c000021400000000 0000100000040000 | ................
    0000040000000000 0001000000c0     | ..............
TRACE +0.001 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010204000000000001640000c0
TRACE +0.000 Received full packet: 010204000000000001640000

A fatal error occurred: Failed to enter Flash download mode (result was 01640000: Unknown result)
```

By the way, according to the packet format, the result `01640000` means

* `01` : Status code, `01` means failure.
* `64` : Error type
* `0000`: Reserved

And esptool uses first 2 bytes (combination of `Status code` and `Error type` field) as a key of `err_defs` .

https://github.com/espressif/esptool/blob/e9489931f10d27d487bebaed3b2500674b4f444d/esptool/util.py#L159

Actual `error type` of the result is `0x64`, but you said that there are no such error definition in ROM code. 

> I found some missing error definitions for flash chip operations, but these are all in a 0x600N range (where N is 1-6). 

The `Error type` field is just one byte and cannot hold `0x600N` since it is 2 bytes value.
So, I doubt that the ROM downloader does not return the internal error code but convert it to some other code before setting it to the `Error type` field.

radimkarnis, 2024-01-31:
> The Error type field is just one byte and cannot hold 0x600N since it is 2 bytes value.

It can, this is a snippet from the ROM err defs, these get sent raw over the wire. The `01 : Status code` info is incomplete, the status can be also `60` meaning flash memory issues:

```C
...
#define ESP_ERR_INVALID_CRC         0x109   /*!< CRC or checksum was invalid */
#define ESP_ERR_INVALID_VERSION     0x10A   /*!< Version was invalid */
#define ESP_ERR_INVALID_MAC         0x10B   /*!< MAC address was invalid */

#define ESP_ERR_FLASH_BASE          0x6000  /*!< Starting number of flash error codes */
#define ESP_ERR_FLASH_OP_FAIL           (ESP_ERR_FLASH_BASE+1)
#define ESP_ERR_FLASH_OP_TIMEOUT        (ESP_ERR_FLASH_BASE+2)
#define ESP_ERR_FLASH_NOT_INITIALISED   (ESP_ERR_FLASH_BASE+3) 
...
```

But this is not relevant to your err code. 

> convert it to some other code

I considered that, but no, I found no signs of conversion or I didn't look hard enough. Nevertheless, esptool defines only the error codes that are defined in the ROM code (the snippet above this text). It is not going to be tracing everything the ROM does.

closed 2024-02-01 by espressif-bot

radimkarnis, 2024-02-01:
Thanks for the report and investigation, I have at least added some more error definitions from the ROM.