espressif/esptool #955: esptool v4.7.0 - esp32-h2 - A fatal error occurred: Packet content transfer stopped (looks like #857)

Z80sp, 2024-02-09:
### Operating System

Linux

### Esptool Version

4.7.0

### Python Version

Python 3.9.0

### Chip Description

esp32-h2-mini-1-n4

### Device Description

ESP32-H2-DevKitM-1

### Hardware Configuration

The kit as it is from out of the box

### How is Esptool Run

Espressif IDF v5.1.2 for Linux

### Full Esptool Command Line that Was Run

esptool.py --trace -p /dev/ttyACM0 --baud 115200 --before default_reset --after hard_reset --chip esp32h2  write_flash --flash_mode dio --flash_size 2MB --flash_freq 48m 0x0 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/hello_world.bin

### Esptool Output

```shell
esptool.py v4.7.0
Serial port /dev/ttyACM0
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.005 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010804000707122000000000c0
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.001 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010804000707122000000000c0
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.001 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010804000707122000000000c0
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.001 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010804000707122000000000c0
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.001 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010804000707122000000000c0
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.002 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010804000707122000000000c0
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.001 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010804000707122000000000c0
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.001 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010804000707122000000000c0
TRACE +0.000 Received full packet: 010804000707122000000000

[... 748 lines cut ...]
TRACE +0.002 Read 1 bytes: 00
TRACE +0.001 Read 1 bytes: 00
TRACE +0.001 Read 1 bytes: 00
TRACE +0.002 Read 1 bytes: 00
TRACE +0.001 Read 1 bytes: 00
TRACE +0.001 Read 1 bytes: 00
TRACE +0.001 Read 1 bytes: c0
TRACE +0.000 Received full packet: 01110200000000000000
TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=00100040
TRACE +0.000 Write 14 bytes: c0000a04000000000000100040c0
TRACE +0.212 Read 1 bytes: c0
TRACE +0.017 Read 1 bytes: 01
TRACE +0.002 Read 1 bytes: 0a
TRACE +3.047 Packet content transfer stopped (received 2 bytes)

A fatal error occurred: Packet content transfer stopped (received 2 bytes)
```

### More Information

Never succeeded and it doesn't stop always within the same step

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2024-02-09:
Hello @Z80sp,
#857 was a known bug in the stub flasher for ESP32-S3, you are using ESP32-H2, these are not related.

Z80sp, 2024-02-09:
Ok, sorry I didn't know.

Z80sp, 2024-02-09:
Don't know if that helps, but I just tried to change the target chip to ESP32-C3 and dev board to ESP32-C3-DevKitM-1 and it worked fine:

esptool.py --chip esp32c3 -p /dev/ttyUSB0 -b 115200 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x0 bootloader/bootloader.bin 0x10000 hello_world.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.7.0
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32-C3 (QFN32) (revision v0.3)
Features: WiFi, BLE, Embedded Flash 4MB (XMC)
Crystal is 40MHz
MAC: 7c:df:a1:61:ec:1c
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x00039fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 20576 bytes to 12635...
Writing at 0x00000000... (100 %)
Wrote 20576 bytes (12635 compressed) at 0x00000000 in 1.5 seconds (effective 110.2 kbit/s)...
Hash of data verified.
Compressed 169360 bytes to 89535...
Writing at 0x00010000... (16 %)
Writing at 0x0001a5a9... (33 %)
Writing at 0x00020fe0... (50 %)
Writing at 0x0002846e... (66 %)
Writing at 0x0002edb5... (83 %)
Writing at 0x00035efa... (100 %)
Wrote 169360 bytes (89535 compressed) at 0x00010000 in 8.3 seconds (effective 163.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.1 seconds (effective 338.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
[100%] Built target flash
Done

Z80sp, 2024-02-09:
Changed back to chip ESP32-H2 and board ESP32-H2-DevKitM-1 in case and I'm still getting a failure:

esptool.py --chip esp32h2 -p /dev/ttyACM0 -b 115200 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 48m --flash_size 2MB 0x0 bootloader/bootloader.bin 0x10000 hello_world.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.7.0
Serial port /dev/ttyACM0
Connecting....
Chip is ESP32-H2 (revision v0.1)
Features: BLE, IEEE802.15.4
Crystal is 32MHz
MAC: 74:4d:bd:ff:fe:60:2b:f7
BASE MAC: 74:4d:bd:60:2b:f7
MAC_EXT: ff:fe
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x00035fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 21056 bytes to 12830...
Writing at 0x00000000... (100 %)

A fatal error occurred: Packet content transfer stopped (received 2 bytes)
CMake Error at run_serial_tool.cmake:66 (message):

  /opt/Espressif/IDF-TOOLS/python_env/idf5.1_py3.9_env/bin/python;;/mnt/ESP32/opt/Espressif/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32h2
  failed.

Z80sp, 2024-02-09:
It works using extra option --no-stub 
Do you think it could be something with the stub flasher for ESP32-H2 ?

radimkarnis, 2024-02-12:
> Do you think it could be something with the stub flasher for ESP32-H2 ?

Yes, it is possible. I suspect the compression to be the issue. Can you try adding the `-u` option after `write_flash` and running the command again? This turns off the compression in the stub flasher. Thanks!

Z80sp, 2024-02-13:
With compression:

[root@devel-box hello_world]# esptool.py --chip esp32h2 -p /dev/ttyACM0 -b 115000 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 48m --flash_size 2MB 0x0 build/bootloader/bootloader.bin 0x10000 build/hel
lo_world.bin 0x8000 build/partition_table/partition-table.bin
esptool.py v4.7.0
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-H2 (revision v0.1)
Features: BLE, IEEE802.15.4
Crystal is 32MHz
MAC: 74:4d:bd:ff:fe:60:2b:f7
BASE MAC: 74:4d:bd:60:2b:f7
MAC_EXT: ff:fe
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x00035fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 21056 bytes to 12831...
Wrote 21056 bytes (12831 compressed) at 0x00000000 in 0.4 seconds (effective 480.7 kbit/s)...
Hash of data verified.
Compressed 153552 bytes to 81408...
Writing at 0x00019af8... (40 %)Traceback (most recent call last):
  File "/opt/Espressif/IDF-TOOLS/python_env/idf5.1_py3.9_env/lib/python3.9/site-packages/esptool/__init__.py", line 1139, in _main
    main()
  File "/opt/Espressif/IDF-TOOLS/python_env/idf5.1_py3.9_env/lib/python3.9/site-packages/esptool/__init__.py", line 923, in main
    operation_func(esp, args)
  File "/opt/Espressif/IDF-TOOLS/python_env/idf5.1_py3.9_env/lib/python3.9/site-packages/esptool/cmds.py", line 598, in write_flash
    esp.flash_defl_block(block, seq, timeout=timeout)
  File "/opt/Espressif/IDF-TOOLS/python_env/idf5.1_py3.9_env/lib/python3.9/site-packages/esptool/loader.py", line 131, in inner
    return func(*args, **kwargs)
  File "/opt/Espressif/IDF-TOOLS/python_env/idf5.1_py3.9_env/lib/python3.9/site-packages/esptool/loader.py", line 1077, in flash_defl_block
    self.check_command(
  File "/opt/Espressif/IDF-TOOLS/python_env/idf5.1_py3.9_env/lib/python3.9/site-packages/esptool/loader.py", line 467, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "/opt/Espressif/IDF-TOOLS/python_env/idf5.1_py3.9_env/lib/python3.9/site-packages/esptool/loader.py", line 436, in command
    p = self.read()
  File "/opt/Espressif/IDF-TOOLS/python_env/idf5.1_py3.9_env/lib/python3.9/site-packages/esptool/loader.py", line 369, in read
    return next(self._slip_reader)
StopIteration

A fatal error occurred: The chip stopped responding.

Z80sp, 2024-02-13:
Without compression (option -u):

[root@devel-box hello_world]# esptool.py --chip esp32h2 -p /dev/ttyACM0 -b 115000 --before=default_reset --after=hard_reset write_flash -u --flash_mode dio --flash_freq 48m --flash_size 2MB 0x0 build/bootloader/bootloader.bin 0x10000 build/hello_world.bin 0x8000 build/partition_table/partition-table.bin
esptool.py v4.7.0
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-H2 (revision v0.1)
Features: BLE, IEEE802.15.4
Crystal is 32MHz
MAC: 74:4d:bd:ff:fe:60:2b:f7
BASE MAC: 74:4d:bd:60:2b:f7
MAC_EXT: ff:fe
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x00035fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Writing at 0x00004000... (100 %)Traceback (most recent call last):
  File "/opt/Espressif/IDF-TOOLS/python_env/idf5.1_py3.9_env/lib/python3.9/site-packages/esptool/__init__.py", line 1139, in _main
    main()
  File "/opt/Espressif/IDF-TOOLS/python_env/idf5.1_py3.9_env/lib/python3.9/site-packages/esptool/__init__.py", line 923, in main
    operation_func(esp, args)
  File "/opt/Espressif/IDF-TOOLS/python_env/idf5.1_py3.9_env/lib/python3.9/site-packages/esptool/cmds.py", line 609, in write_flash
    esp.flash_block(block, seq)
  File "/opt/Espressif/IDF-TOOLS/python_env/idf5.1_py3.9_env/lib/python3.9/site-packages/esptool/loader.py", line 872, in flash_block
    self.check_command(
  File "/opt/Espressif/IDF-TOOLS/python_env/idf5.1_py3.9_env/lib/python3.9/site-packages/esptool/loader.py", line 467, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "/opt/Espressif/IDF-TOOLS/python_env/idf5.1_py3.9_env/lib/python3.9/site-packages/esptool/loader.py", line 436, in command
    p = self.read()
  File "/opt/Espressif/IDF-TOOLS/python_env/idf5.1_py3.9_env/lib/python3.9/site-packages/esptool/loader.py", line 369, in read
    return next(self._slip_reader)
StopIteration

A fatal error occurred: The chip stopped responding.

Z80sp, 2024-02-13:
With option --no-stub:

[root@devel-box hello_world]# esptool.py --chip esp32h2 -p /dev/ttyACM0 -b 115000 --no-stub --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 48m --flash_size 2MB 0x0 build/bootloader/bootloader.bin 0x10000 build/hello_world.bin 0x8000 build/partition_table/partition-table.bin
esptool.py v4.7.0
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-H2 (revision v0.1)
Features: BLE, IEEE802.15.4
Crystal is 32MHz
MAC: 74:4d:bd:ff:fe:60:2b:f7
BASE MAC: 74:4d:bd:60:2b:f7
MAC_EXT: ff:fe
Enabling default SPI flash mode...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x00035fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Erasing flash...
Took 0.19s to erase flash block
Wrote 21504 bytes at 0x00000000 in 0.2 seconds (708.8 kbit/s)...
Hash of data verified.
Erasing flash...
Took 0.75s to erase flash block
Wrote 153600 bytes at 0x00010000 in 1.6 seconds (747.1 kbit/s)...
Hash of data verified.
Erasing flash...
Took 0.03s to erase flash block
Wrote 3072 bytes at 0x00008000 in 0.0 seconds (851.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

radimkarnis, 2024-02-13:
Thank you, so it is not the compression.

Could you share your binaries (`build/bootloader/bootloader.bin`, `build/hello_world.bin`, `build/partition_table/partition-table.bin`)? I will try to reproduce this locally. Thank you

Z80sp, 2024-02-13:
Sure!
Here's the archive:

[root@devel-box hello_world]# tar --gzip -tf hello_world.tar.gz
build/hello_world.bin
build/bootloader/bootloader.bin
build/partition_table/partition-table.bin

[root@devel-box hello_world]# md5sum hello_world.tar.gz
b37a718c61d1cd52c8586ff9d6c9dcf0  hello_world.tar.gz

[hello_world.tar.gz](https://github.com/espressif/esptool/files/14266236/hello_world.tar.gz)

radimkarnis, 2024-02-19:
Hi @Z80sp,
thank you for the binaries.

I am trying to reproduce this, but without any luck:

```
esptool.py -p /dev/tty.usbmodem111301 --baud 115200 --before default_reset --after hard_reset --chip esp32h2 write_flash --flash_mode dio --flash_size 2MB --flash_freq 48m 0x0 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/hello_world.bin

esptool.py v4.7.0
Serial port /dev/tty.usbmodem111301
Connecting...
Chip is ESP32-H2 (revision v0.1)
Features: BLE, IEEE802.15.4
Crystal is 32MHz
MAC: 60:55:f9:ff:fe:f7:3e:51
BASE MAC: 60:55:f9:f7:3e:51
MAC_EXT: ff:fe
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00010000 to 0x00035fff...
Compressed 21056 bytes to 12831...
Wrote 21056 bytes (12831 compressed) at 0x00000000 in 0.3 seconds (effective 594.7 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.0 seconds (effective 557.8 kbit/s)...
Hash of data verified.
Compressed 153552 bytes to 81408...
Wrote 153552 bytes (81408 compressed) at 0x00010000 in 1.0 seconds (effective 1287.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

Flashing always succeeds. This could be related to your OS drivers or your environment in general. I am not sure if we want to go deeper into this. Do you consider the `--no-stub` option to be a acceptable workaround for now?

dobairoland, 2024-02-19:
Works for me as well on Linux:

```
 esptool.py --chip esp32h2 -p /dev/ttyACM0 --before=default_reset --after=hard_reset write_flash -u --flash_mode dio --flash_freq 48m --flash_size 2MB 0x0 build/bootloader/bootloader.bin 0x10000 build/hello_world.bin 0x8000 build/partition_table/partition-table.bin
esptool.py vv4.8.dev1
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-H2 (revision v0.1)
Features: BLE, IEEE802.15.4
Crystal is 32MHz
MAC: 60:55:f9:ff:fe:f7:3e:51
BASE MAC: 60:55:f9:f7:3e:51
MAC_EXT: ff:fe
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x00035fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Wrote 32768 bytes at 0x00000000 in 0.3 seconds (924.8 kbit/s)...
Hash of data verified.
Wrote 163840 bytes at 0x00010000 in 1.0 seconds (1366.1 kbit/s)...
Hash of data verified.
Wrote 16384 bytes at 0x00008000 in 0.1 seconds (1616.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

dobairoland, 2024-02-19:
I'm guessing based on Python 3.9.0 that you use some older version of Linux distribution. Are you able to upgrade? Or try to flash through UART until you can upgrade.

radimkarnis, 2024-03-04:
I believe this is a local environment issue, as it cannot be reproduced on our side. I am closing this issue for this reason and inactivity.

closed 2024-03-04 by radimkarnis