espressif/esptool #958: esptool.py v4.7.0 does not work with ESP32-C3 SuperMini

DooMMasteR, 2024-02-23:
### Operating System

Ubuntu 23.10

### Esptool Version

v4.7.0

### Python Version

Python 3.11.6

### Chip Description

ESP32-C3

### Device Description

ESP32-C3 SuperMini

### Hardware Configuration

No external devices.

### How is Esptool Run

CLI

### Full Esptool Command Line that Was Run

<pre><font color="#2AA1B3">python3</font> <font color="#2AA1B3"><u style="text-decoration-style:single">esptool.py</u></font> <font color="#2AA1B3">-p</font> <font color="#2AA1B3"><u style="text-decoration-style:single">/dev/ttyACM0</u></font> <font color="#2AA1B3">-b</font> <font color="#2AA1B3">460800</font> <font color="#2AA1B3">--before</font> <font color="#2AA1B3">default_reset</font> <font color="#2AA1B3">--after</font> <font color="#2AA1B3">hard_reset</font> <font color="#2AA1B3">--chip</font> <font color="#2AA1B3">esp32c3</font>  <font color="#2AA1B3">write_flash</font> <font color="#2AA1B3">--flash_mode</font> <font color="#2AA1B3">dio</font> <font color="#2AA1B3">--flash_size</font> <font color="#2AA1B3">detect</font> <font color="#2AA1B3">0x0</font> <font color="#33C7DE"><u style="text-decoration-style:single">~</u></font><font color="#2AA1B3"><u style="text-decoration-style:single">/Downloads/ucware-air-factory.bin</u></font> </pre>

### Esptool Output

```shell
esptool.py v4.7.0
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-C3 (QFN32) (revision v0.4)
Features: WiFi, BLE
Crystal is 40MHz
MAC: dc:da:0c:8e:8b:3c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0xffff3f, SizeID=0xff), defaulting to 4MB
Flash will be erased from 0x00000000 to 0x000eefff...
Compressed 978576 bytes to 576728...

A fatal error occurred: Packet content transfer stopped (received 0 bytes)
```

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2024-02-23:
Hello @DooMMasteR,
this seems like a hardware issue to me:

Esptool is working and communicating with the chip (you can see the chip is detected, stub uploaded, baud rate changed, ...).
Only when the `write_flash` operation starts, you can see things start to fail since the flash memory chip can't be detected (`Warning: Could not auto-detect Flash size (FlashID=0xffff3f, SizeID=0xff)`). This implies a hardware issue.

Can you check your board for solder bridges or any other defects? Is there anything connected to your board (if so, try disconnecting everything)?

radimkarnis, 2024-02-23:
Thanks for checking.

Let's see if this is really about the flash chip or if there is an issue with the drivers:

- Can you try running esptool with a lower baud rate? 
- Can you try running other commands than `write_flash`? (you can try `flash_id, `get_security_info`, or `read_mac`). 
- Can you try with another host PC (or OS)?

DooMMasteR, 2024-02-23:
So at `9600` baud the same happens using write.
`flash_id` results in 
```
esptool.py v4.7.0
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-C3 (QFN32) (revision v0.4)
Features: WiFi, BLE
Crystal is 40MHz
MAC: dc:da:0c:8e:8b:3c
Uploading stub...
Running stub...
Stub running...
Manufacturer: 3f
Device: ffff
Detected flash size: Unknown
Hard resetting via RTS pin...
```
`get_security_info`
```
esptool.py v4.7.0
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-C3 (QFN32) (revision v0.4)
Features: WiFi, BLE
Crystal is 40MHz
MAC: dc:da:0c:8e:8b:3c
Uploading stub...
Running stub...
Stub running...

Security Information:
=====================
Flags: 0x00000000 (0b0)
Key Purposes: (0, 0, 0, 0, 0, 0, 12)
Chip ID: 5
API Version: 3
Secure Boot: Disabled
Flash Encryption: Disabled
SPI Boot Crypt Count (SPI_BOOT_CRYPT_CNT): 0x0
Hard resetting via RTS pin...
```
`read_mac`
```
esptool.py v4.7.0
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-C3 (QFN32) (revision v0.4)
Features: WiFi, BLE
Crystal is 40MHz
MAC: dc:da:0c:8e:8b:3c
Uploading stub...
Running stub...
Stub running...
MAC: dc:da:0c:8e:8b:3c
Hard resetting via RTS pin...
```

DooMMasteR, 2024-02-23:
This is the "early" boot output.

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x5 (DOWNLOAD(USB/UART0/1))
Saved PC:0x4004d190
waiting for download
```

DooMMasteR, 2024-02-23:
Tried it on an older machine running Windows too, same behaviour, it times out after a short moment...
here is the trace of it 
[esp32c3.txt](https://github.com/espressif/esptool/files/14384873/esp32c3.txt)

DooMMasteR, 2024-02-23:
Same with `--no-stub` 
[esp32c3.txt](https://github.com/espressif/esptool/files/14385131/esp32c3.txt)

outcasta4, 2024-02-24:
Serial port /dev/ttyACM0
Connecting...
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (QFN32) (revision v0.4)
Features: WiFi, BLE
Crystal is 40MHz
MAC: dc:da:0c:8e:8a:60
Uploading stub...Serial port /dev/ttyACM0
Connecting...
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (QFN32) (revision v0.4)
Features: WiFi, BLE
Crystal is 40MHz
MAC: dc:da:0c:8e:8a:60
Uploading stub...
Running stub...
Stub running...
MAC: dc:da:0c:8e:8a:60
Hard resetting via RTS pin...

Running stub...
Stub running...
MAC: dc:da:0c:8e:8a:60
Hard resetting via RTS pin...

Yo, I got the exact same issue,  same output same board on Windows 11 and Debian machine.

NekuSoul, 2024-02-25:
Having the same issues. I'm connecting the device, hold BOOT while pressing RST to disrupt the boot loop, then run the following command:

```
./esptool --port /dev/ttyACM0 write_flash 0x0 ~/Arduino/BareMinimum/BareMinimum.ino.esp32c3.bin
```
```
esptool.py v4.7.0
Serial port /dev/ttyACM0
Connecting...
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (QFN32) (revision v0.4)
Features: WiFi, BLE
Crystal is 40MHz
MAC: dc:da:0c:8e:c4:7c
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00037fff...
Compressed 226624 bytes to 127566...

A fatal error occurred: Packet content transfer stopped (received 8 bytes)
```

I've also tried flashing from Windows and using a Flipper Zero to no avail.

As an additional test I've flashed the device using the above command while also having the device connected to the Flipper Zero. Checking the UART Terminal gives me the following output:

```
+-----------------------+
|rst:0x1(POWERON),boot0x|
|5(DOWNLOAD(USB/UART    |
|0/1))                  |
|waiting for download   |
|OHAI                   |
|                       |
|p                      |
+-----------------------+
```

Device in question: https://www.amazon.de/dp/B0CPY6C7YL

radimkarnis, 2024-02-28:
@DooMMasteR @NekuSoul thanks for the investigation.

You can see the ESP works properly (e.g. we are getting security info output), but the flash chip is not working:

```
Manufacturer: 3f
Device: ffff
Detected flash size: Unknown
```

This is definitely not a software issue. I suspect a design flaw in the ultra-cheap C3 SuperMini devkit. Maybe the SPI flash pins are shorted or it's not getting enough power?

dobairoland, 2024-02-28:
The board got one review on Amazon.de. It states that one out of three bought modules had a short. So please check your boards and in case of issues contact the manufacturer.

DooMMasteR, 2024-02-28:
I have tested all 20 and they all have the issue, so I just returned them, there is no visible soldering flaw on any of them and the SPI and UART pins are all not shorted....
Mine were sourced on AliExpress though and since they have free returns, I guessed it was way easier just to dumpe the boards back on them.

radimkarnis, 2024-02-28:
Closing the issue. Hopefully, the manufacturer will solve the issue ASAP and this was just a small faulty batch.

closed 2024-02-28 by radimkarnis

dotsam, 2024-03-12:
I've also just received 5 of these boards from AliExpress, and judging from your photos and mine, the SoC that we received is in fact a plain ESP32-C3 without any on-board flash.

[image]

[image]

(From https://www.espressif.com/sites/default/files/documentation/esp32-c3_datasheet_en.pdf)

[image]

(From https://www.espressif.com/sites/default/files/documentation/espressif_chip_packaging_information_en.pdf)

radimkarnis, 2024-03-13:
@dotsam you are right, thanks for noticing! That explains it. I advise anyone having this issue to return the boards ASAP..

DooMMasteR, 2024-03-13:
Yeah I already did so.

NekuSoul, 2024-03-13:
@dotsam Good info. Checking the picture in the review from the store page I linked again, is indeed a model with embedded flash, which explains why theirs worked and ours not.

@radimkarnis Yeah, I already returned them, got some from a different brand and source and flashed them without issues.

xbmcnut, 2024-05-30:
Thanks for the detailed info. I too got burnt with five from AE that I can't flash with WLED or ESPHome. Requested refund and now await the Seeed Studio version 😡

yanxke, 2024-06-04:
Just got a batch with this issue as well.  What good are these chips without flash and why would anyone want to buy these?