espressif/esptool #961: wemos lolin32 Failed uploading: uploading error: exit status 2

Bri-iggy, 2024-03-03:
### Operating System

windows 11

### Esptool Version

esptool.py v4.6

### Python Version

Python 3.12.2

### Full Esptool Command Line that Was Run

--chip esp32 --port "COM3" --baud 115200  --before default_reset --after hard_reset write_flash  -z --flash_mode keep --flash_freq keep --flash_size keep 0x1000

### Esptool Output

```shell
--chip esp32 --port "COM3" --baud 115200  --before default_reset --after hard_reset write_flash  -z --flash_mode keep --flash_freq keep --flash_size keep 0x1000
esptool.py v4.6
Serial port COM3
Connecting......................................

A fatal error occurred: Failed to connect to ESP32: Wrong boot mode detected (0x13)! The chip needs to be in download mode.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Failed uploading: uploading error: exit status 2
```

### What is the Expected Behaviour?

It supposed to get the chip Id

### More Information

I tried on my friend's pc and it has the same error

dobairoland, 2024-03-03:
The command is for writing flash instead of getting the chip id (`chip_id`). Also it is incomplete. After 0x1000 the binary should follow.

But not the above is causing the error. This is probably a hardware or OS driver issue. The troubleshooting guide is linked in the message. We probably couldn't suggest anything else.

radimkarnis, 2024-03-15:
Closing this issue due to inactivity and incomplete information. Please go through the troubleshooting guide to see the recommended action.

closed 2024-03-15 by radimkarnis