espressif/esptool #962: Esp32 C3 Super MIni Flash failed

faizannazir, 2024-03-09:
### Operating System

Window 11

### Esptool Version

4.7.0

### Python Version

3.11.0

### Esptool Output

```shell
Enabling default SPI flash mode...
WARNING: Failed to communicate with the flash chip, read/write operations will fail. Try checking the chip connections or removing any other hardware connected to IOs.
Manufacturer: ff
Device: ffff
Detected flash size: Unknown

 esptool --no-stub --chip esp32c3 --port "COM9" --baud 921600  --before default_reset --after hard_reset write_flash --no-compress --flash_mode keep --flash_freq keep --flash_size keep 0x0 "C:\Users\hp 840 G5\AppData\Local\Temp\arduino\sketches\176DE3D0312A606C8CA421C07576CC1B/sketch_mar10a.ino.bootloader.bin" 0x8000 "C:\Users\hp 840 G5\AppData\Local\Temp\arduino\sketches\176DE3D0312A606C8CA421C07576CC1B/sketch_mar10a.ino.partitions.bin" 0xe000 "C:\Users\hp 840 G5\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14/tools/partitions/boot_app0.bin" 0x10000 "C:\Users\hp 840 G5\AppData\Local\Temp\arduino\sketches\176DE3D0312A606C8CA421C07576CC1B/sketch_mar10a.ino.bin"  
>> 
esptool.py v4.7.0
Serial port COM9
Connecting...
Chip is ESP32-C3 (QFN32) (revision v0.4)
Features: WiFi, BLE
Crystal is 40MHz
MAC: dc:da:0c:8e:a0:bc
Changing baud rate to 921600
Changed.
Enabling default SPI flash mode...
WARNING: Failed to communicate with the flash chip, read/write operations will fail. Try checking the chip connections or removing any other hardware connected to IOs.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00003fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00047fff...
Erasing flash...

A fatal error occurred: Serial data stream stopped: Possible serial noise or corruption.
```

faizannazir, 2024-03-09:
```
`esptool --no-stub -p COM9 flash_id                                                                              
esptool.py v4.7.0                                                                                                                                  
Serial port COM9                                                                                                                                   
Connecting...                                                                                                                                      
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (QFN32) (revision v0.4)
Features: WiFi, BLE
Crystal is 40MHz
MAC: dc:da:0c:8e:a0:bc
Enabling default SPI flash mode...
WARNING: Failed to communicate with the flash chip, read/write operations will fail. Try checking the chip connections or removing any other hardware connected to IOs.
Manufacturer: ff
Device: ffff
Detected flash size: Unknown
Hard resetting via RTS pin...`
```

faizannazir, 2024-03-09:
```
 esptool flash_id                                                                                                
esptool.py v4.7.0                                                                                                                                  
Found 3 serial ports                                                                                                                               
Serial port COM9                                                                                                                                   
Connecting...
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (QFN32) (revision v0.4)
Features: WiFi, BLE
Crystal is 40MHz
MAC: dc:da:0c:8e:a0:bc
Uploading stub...
Running stub...
Stub running...
Manufacturer: 3f
Device: ffff
Detected flash size: Unknown
Hard resetting via RTS pin...
```

radimkarnis, 2024-03-09:
Hi @faizannazir,
please see the discussion here: https://github.com/espressif/esptool/issues/958. These cheap devkits seem to be prone to failures. There's nothing that can be done on the esptool side.

closed 2024-03-09 by radimkarnis