espressif/esptool #963: USB-C vs MicroUSB upload on ESP32-C6

lbernstone, 2024-03-10:
### Operating System

Ubuntu 22.04

### Esptool Version

v4.7.0 (from master)

### Python Version

3.10.12

### Chip Description

ESP32-C6

### Device Description

ESP32-C6-DevKit-C

### Hardware Configuration

Micro-USB module connected on 12/13

### How is Esptool Run

terminal

### Full Esptool Command Line that Was Run

esptool.py chip_id

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

lbernstone, 2024-03-10:
esptool works fine when connecting through the USB-C port on board.  Trying to use a MicroUSB module (which matches my custom board), I get unexpected behavior.  In the breadboard config I could believe line noise, but on my board, these are close traces and have ESD protectors.
Here's the lead-in traces from esptool.py chip_id
USB-C
```
esptool.py v4.7.0
Found 1 serial ports
Serial port /dev/ttyACM0
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
[usb_c.log](https://github.com/espressif/esptool/files/14551513/usb_c.log)
[usb_a.log](https://github.com/espressif/esptool/files/14551515/usb_a.log)

    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.000 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010804000707122000000000c0
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010804000707122000000000c0
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Read 1 bytes: c0
TRACE +0.000 Read 27 bytes: 
    0108040007071220 00000000c0c00108 | ....... ........
    0400070712200000 0000c0           | ..... .....
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Read 14 bytes: c0010804000707122000000000c0
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010804000707122000000000c0
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010804000707122000000000c0
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010804000707122000000000c0
TRACE +0.000 Received full packet: 010804000707122000000000

Detecting chip type...TRACE +0.000 command op=0x14 data len=0 wait_response=1 timeout=3.000 data=
TRACE +0.000 Write 10 bytes: c00014000000000000c0
TRACE +0.000 Read 1 bytes: c0
TRACE +0.000 Read 33 bytes: 
    0114180000000000 0000000000000000 | ................
    0000000c0d000000 0000000000000000 | ................
    c0                                | .
TRACE +0.000 Received full packet: 
    0114180000000000 0000000000000000 | ................
    0000000c0d000000 0000000000000000 | ................
TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=00100040
TRACE +0.000 Write 14 bytes: c0000a04000000000000100040c0
TRACE +0.000 Read 1 bytes: c0
TRACE +0.000 Read 2 bytes: 010a
TRACE +0.000 Read 1 bytes: 04
TRACE +0.000 Read 10 bytes: 006f80e02c00000000c0
TRACE +0.000 Received full packet: 010a04006f80e02c00000000
TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=80f58740
TRACE +0.000 Write 14 bytes: c0000a04000000000080f58740c0
TRACE +0.000 Read 1 bytes: c0
TRACE +0.000 Read 2 bytes: 010a
TRACE +0.000 Read 1 bytes: 04
TRACE +0.000 Read 10 bytes: 000300000000000000c0
TRACE +0.000 Received full packet: 010a04000300000000000000
TRACE +0.000 command op=0x09 data len=16 wait_response=1 timeout=3.000 data=181c0b60a13ad850ffffffff00000000
TRACE +0.000 Write 26 bytes: 
    c000091000000000 00181c0b60a13ad8 | ............`.:.
    50ffffffff000000 00c0             | P.........
[... 161 lines cut ...]
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.562 No serial data received.
.TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.361 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010804000707122000000000c0
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +3.004 Serial data stream stopped: Possible serial noise or corruption.
.TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
```

dobairoland, 2024-03-11:
This issue looks to me a hardware one and not an esptool one.

> Micro-USB module connected on 12/13

And what about the reset circuitry needed for [automatic bootloader mode](https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html#automatic-bootloader)?

You can see "ESP-ROM:esp32c6-" in the trace output. This indicates that the reset into download mode didn't went well.

Try to connect a [terminal program](https://docs.espressif.com/projects/esptool/en/latest/esp32/troubleshooting.html#serial-terminal-programs) instead and test what you receive upon normal and download reset. 

HINT: Holding down the Boot button while connecting might help.

lbernstone, 2024-03-11:
Yep.  Hardware.  Needs to have USB-OTG set with 5K1 on ID pin.

closed 2024-03-11 by lbernstone