espressif/esptool #965: Flash with CH348 usb to serial chip

Roberto-spm, 2024-03-14:
### Operating System

Debian trixie

### Esptool Version

v4.2.1

### Python Version

python 3.11.5

### Chip Description

ESP32-wroom-32

### Device Description

I made a custom board with ESP32-WROOM-32 cpu

### How is Esptool Run

linux command line

### Full Esptool Command Line that Was Run

python3 "/home/spm/Arduino/hardware/espressif/esp32/tools/esptool/esptool.py" --chip esp32 --port "/dev/ttyCH9344USB4" --baud 921600  --before default_reset --after hard_reset write_flash  -z --flash_mode dio --flash_freq 80m --flash_size 4MB 0x1000 "/tmp/arduino/sketches/9FBFADF38D6EC1C1860ABE314506BB1A/Contalitri_1_2_5.ino.bootloader.bin" 0x8000 "/tmp/arduino/sketches/9FBFADF38D6EC1C1860ABE314506BB1A/Contalitri_1_2_5.ino.partitions.bin" 0xe000 "/home/spm/Arduino/hardware/espressif/esp32/tools/partitions/boot_app0.bin" 0x10000 "/tmp/arduino/sketches/9FBFADF38D6EC1C1860ABE314506BB1A/Contalitri_1_2_5.ino.bin"

### Esptool Output

```shell
esptool.py v4.2.1
Serial port /dev/ttyCH9344USB4
Connecting...
Failed to get PID of a device on /dev/ttyCH9344USB4, using standard reset sequence.

Traceback (most recent call last):
  File "/home/spm/Arduino/hardware/espressif/esp32/tools/esptool/esptool.py", line 34, in <module>
    esptool._main()
  File "/home/spm/Arduino/hardware/espressif/esp32/tools/esptool/esptool/__init__.py", line 1004, in _main
    main()
  File "/home/spm/Arduino/hardware/espressif/esp32/tools/esptool/esptool/__init__.py", line 646, in main
    esp = esp or get_default_connected_device(
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/spm/Arduino/hardware/espressif/esp32/tools/esptool/esptool/__init__.py", line 895, in get_default_connected_device
    _esp.connect(before, connect_attempts)
  File "/home/spm/Arduino/hardware/espressif/esp32/tools/esptool/esptool/loader.py", line 631, in connect
    last_error = self._connect_attempt(
                 ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/spm/Arduino/hardware/espressif/esp32/tools/esptool/esptool/loader.py", line 555, in _connect_attempt
    self.bootloader_reset(usb_jtag_serial, extra_delay)
  File "/home/spm/Arduino/hardware/espressif/esp32/tools/esptool/esptool/loader.py", line 530, in bootloader_reset
    self._setDTR(False)  # IO0=HIGH
    ^^^^^^^^^^^^^^^^^^^
  File "/home/spm/Arduino/hardware/espressif/esp32/tools/esptool/esptool/loader.py", line 449, in _setDTR
    self._port.setDTR(state)
  File "/usr/lib/python3/dist-packages/serial/serialutil.py", line 603, in setDTR
    self.dtr = value
    ^^^^^^^^
  File "/usr/lib/python3/dist-packages/serial/serialutil.py", line 473, in dtr
    self._update_dtr_state()
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 715, in _update_dtr_state
    fcntl.ioctl(self.fd, TIOCMBIC, TIOCM_DTR_str)
OSError: [Errno 22] Invalid argument
```

### More Information

I made a board with 6 serial ports and usb input, with CH348 chip.
The usb to serial board I made works: the serial communication between pc and esp32 is clear in both directions.
The esp32 in my board can be flashed without problem, using a commercial serial to usb adapter and the same esptool command (with /dev/ttyUSB0 device)

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2024-03-14:
Hello @Roberto-spm,
it seems like your `CH348` drivers are having trouble with setting the `DTR` and `RTS` control lines used to reset the chip into the bootloader mode.

Can you please try flashing without the reset sequence? You need to set `--before no_reset` and switch the ESP32 into the bootloader mode manually before flashing.

Roberto-spm, 2024-03-14:
Hello Radim,

thank you, this solved perfectly my problem, now it works!
I disabled the DTR in the chip, by the way I set the bootloader mode via software with 2 mosfets

Thanks again, best regards
Roberto

radimkarnis, 2024-03-14:
Nice! I am glad this worked!

closed 2024-03-14 by radimkarnis