espressif/esptool #967: Flashing an ESP chip results in "A fatal error occurred: The chip stopped responding." after 3% of writing

CoryFoy, 2024-03-20:
### Operating System

macOS 14.4

### Esptool Version

esptool.py v4.7.0

### Python Version

3.11.6

### Chip Description

ESP32-D0WD-V3 (revision v3.1)

### Device Description

Station Edition G1

### Hardware Configuration

No

### How is Esptool Run

Terminal / Command Line (though I tried the Web Flasher too)

### Full Esptool Command Line that Was Run

esptool.py -p /dev/cu.usbserial-558C0013971 write_flash 0x00 firmware-station-g1-2.3.1.4fa7f5a.bin

### Esptool Output

```shell
esptool.py v4.7.0
Loaded custom configuration from /Users/foyc/Downloads/firmware/2.3.1/esptool.cfg
Serial port /dev/cu.usbserial-558C0013971
Connecting...........
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting......
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 08:3a:8d:92:7c:30
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x0020afff...
Compressed 2140544 bytes to 1254288...
Writing at 0x0001b22c... (3 %)Traceback (most recent call last):
  File "/opt/homebrew/lib/python3.11/site-packages/esptool/__init__.py", line 1139, in _main
    main()
  File "/opt/homebrew/lib/python3.11/site-packages/esptool/__init__.py", line 923, in main
    operation_func(esp, args)
  File "/opt/homebrew/lib/python3.11/site-packages/esptool/cmds.py", line 598, in write_flash
    esp.flash_defl_block(block, seq, timeout=timeout)
  File "/opt/homebrew/lib/python3.11/site-packages/esptool/loader.py", line 131, in inner
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.11/site-packages/esptool/loader.py", line 1077, in flash_defl_block
    self.check_command(
  File "/opt/homebrew/lib/python3.11/site-packages/esptool/loader.py", line 467, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.11/site-packages/esptool/loader.py", line 436, in command
    p = self.read()
        ^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.11/site-packages/esptool/loader.py", line 369, in read
    return next(self._slip_reader)
           ^^^^^^^^^^^^^^^^^^^^^^^
StopIteration

A fatal error occurred: The chip stopped responding.
```

### More Information

I was able to get it to flash via the Web Flasher yesterday. However settings weren't persisting when using the Meshtastic app, so I figured I would try reflashing, and now I can't get it to flash at all. It almost always fails at the 3% mark

Other things I've tried:

- Using a custom esptool.cfg to increase timeout to 10 seconds
- Trying a different firmware version
- Running `esptool.py erase_flash`, disconnecting the board, and then trying the write_flash
- Lowering the baud rate to 115200

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

CoryFoy, 2024-03-21:
Update - I've been digging into some other troubleshooting steps. I added in the `--no-stub` option and it looks like it was able to write the flash, but the device isn't starting up yet:

```
2.3.1 % esptool.py -p /dev/cu.usbserial-558C0013971 --no-stub write_flash 0x00 firmware-station-g1-2.3.1.4fa7f5a.bin 
esptool.py v4.7.0
Loaded custom configuration from /Users/foyc/Downloads/firmware/2.3.1/esptool.cfg
Serial port /dev/cu.usbserial-558C0013971
Connecting.......
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.....
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 08:3a:8d:92:7c:30
Enabling default SPI flash mode...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x0020afff...
Erasing flash...
Took 7.25s to erase flash block
Wrote 2141184 bytes at 0x00000000 in 229.5 seconds (74.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

Here's my `esptool.cfg` file:

```
2.3.1 % cat esptool.cfg 
[esptool]
timeout = 30
max_timeout = 240
erase_write_timeout_per_mb = 40
mem_end_rom_timeout = 1.2
serial_write_timeout = 20
```

radimkarnis, 2024-03-22:
Hi @CoryFoy,

>  but the device isn't starting up yet

this will be an issue with the pre-built binary. Can you see any output from the ESP? Is it stuck in a boot loop? You've managed to get a successful flash with esptool (`Hash of data verified.` means all went well). 

If the app is not starting after a successful flash, it's time to start to look at the binary itself. Is there maybe a way for you to build it yourself? FLashing pre-built binaries from the internet can be both unreliable and also dangerous.

CoryFoy, 2024-03-22:
Hi @radimkarnis Thanks so much for the reply! I'm fairly new to embedded programming, so I appreciate the pointers. I did pull a Serial monitor of the device, and you're right - it appears to be stuck in a boot loop. I am flashing with a Meshtastic release built for this device (Station G1) which had previously worked - but I can definitely look more into the binary itself. 

Is there significance to the fact that I can only load it with `--no-stub`? Does that seem to point to some sort of a chip corruption or issue? Or is it not really all that important?

Here's the output from the serial monitor just for completeness sake. But sounds like this is likely not an esptool issue after all.

```
rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
ets Jul 29 2019 12:21:46
```

CoryFoy, 2024-03-25:
Closing since we suspect this is actually a chip issue, not a software one.

closed 2024-03-25 by CoryFoy